The-OpenROAD-Project/OpenROAD #2320: ppl: hungarian_solver_.solve called with zero length matrix

antonblanchard, 2022-09-29:
I have another test case that is hitting issues in PPL. This time in `HungarianAlgorithm::solve`:

```
/usr/include/c++/12/bits/stl_vector.h:1123: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = std::vector<int>; _Alloc = std::allocator<std::vector<int> >; reference = std::vector<int>&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.

 3# raise in /lib64/libc.so.6
 4# abort in /lib64/libc.so.6
 5# __gnu_debug::_Safe_sequence_base::_M_get_mutex() in /lib64/libstdc++.so.6
 6# HungarianAlgorithm::solve(std::vector<std::vector<int, std::allocator<int> >, std::allocator<std::vector<int, std::allocator<int> > > >&, std::vector<int, std::allocator<int> >&) in openroad
 7# ppl::IOPlacer::findPinAssignment(std::vector<ppl::Section, std::allocator<ppl::Section> >&) in openroad
 8# ppl::IOPlacer::run(bool) in openroad
```

If it doesn't reproduce, this patch should trip it:

```
--- a/src/ppl/src/HungarianMatching.cpp
+++ b/src/ppl/src/HungarianMatching.cpp
@@ -63,6 +63,10 @@ HungarianMatching::HungarianMatching(Section& section,
 void HungarianMatching::findAssignment()
 {
   createMatrix();
+  if (hungarian_matrix_.size() == 0) {
+    printf("hungarian_matrix_.size() 0\n");
+    exit(1);
+  }
   hungarian_solver_.solve(hungarian_matrix_, assignment_);
 }

```

Test case: [io_placement_multiplier_asap7_base_2022-09-28_14-32-2.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9669860/io_placement_multiplier_asap7_base_2022-09-28_14-32-2.tar.gz)

antonblanchard, 2022-09-29:
Unsure if this is correct, but this avoids it:

```
--- a/src/ppl/src/HungarianMatching.cpp
+++ b/src/ppl/src/HungarianMatching.cpp
@@ -63,7 +63,8 @@ HungarianMatching::HungarianMatching(Section& section,
 void HungarianMatching::findAssignment()
 {
   createMatrix();
-  hungarian_solver_.solve(hungarian_matrix_, assignment_);
+  if (hungarian_matrix_.size() != 0)
+    hungarian_solver_.solve(hungarian_matrix_, assignment_);
 }
 
 void HungarianMatching::createMatrix()
```

antonblanchard, 2022-09-29:
@eder-matheus FYI

eder-matheus, 2022-09-29:
> Unsure if this is correct, but this avoids it:
>
> ```
> --- a/src/ppl/src/HungarianMatching.cpp
> +++ b/src/ppl/src/HungarianMatching.cpp
> @@ -63,7 +63,8 @@ HungarianMatching::HungarianMatching(Section& [...]

Thanks, @antonblanchard! Your fix is correct. We perform this check in the `findAssignmentForGroups` function, but I probably missed the `findAssignment` at that time. The fix is in this PR: https://github.com/The-OpenROAD-Project/OpenROAD/pull/2322

closed 2022-09-29 by eder-matheus