The-OpenROAD-Project/OpenROAD #2326: LVS check is incorrect

Xies3, 2022-10-02:
[6_final_lvs.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9691486/6_final_lvs.zip)
I run the drc and lvs check where the lvs check is conducted by netgen after running the Makefile in OpenROAD-flow-scripts/flow. There are no errors found in drc while the result of lvs showed that there's a mismatch of circuit 1 and circuit 2. Could you please help me with this problem. Thank you very much. I attached the cdl, gds, makefile, an lvs report.

maliberty, 2022-10-02:
I am not a netgen user and have no idea how to debug problems related to other tools.  If you can point out something that is specifically wrong with the OR output I can take a look at that.  This is not a complete test case and can't be debugged in any case.

Xies3, 2022-10-02:
Yes, so there are still unconnected areas in the generated cdl file.

发自 Outlook<http://aka.ms/weboutlook>

maliberty, 2022-10-02:
That is not enough of a description to take any action.

Xies3, 2022-10-03:
Yes. In the 6_final lvs report, some nets are mismatching and I think that is because some names of instances are not in the library. So is that openroad used some instances that were not in the library and added them to the gds? Thank you very much.

maliberty, 2022-10-03:
What instances?  If you don't give a specific problem don't expect any action.

Also you have not provided a complete test case allowing me to reproduce the result.

vijayank88, 2022-10-03:
@Xies3 
Its good to try this design with `OpenLane flow` https://github.com/The-OpenROAD-Project/OpenLane as it had all signoff checks tools integrated.
If still LVS failures there, file github issue.

maliberty, 2022-10-03:
If the same problem happens in OL the same data will be needed - a specific problem and a complete test case.  I am not clear this is a tool integration issue.  It not clear whether this is a yosys, OR, KLayout, netgen, or user error problem yet.

If you are using ORFS then "make final_report_issue" should package a test case.

maliberty, 2022-10-05:
Please close this if no further action is needed.

saicharan0112, 2022-10-06:
@maliberty This is related to OpenFASOC/temp-sense generator. With the latest version of openroad, the drc check for the generated gds file, the following message is generated 

```
Extracting sky130_fd_sc_hd__decap_4 into sky130_fd_sc_hd__decap_4.ext:
freeMagic called with NULL argument.
```

Also this is reflected in the lvs check where there is a device mismatch for this exact cell.

Please note that this is not observed from the version [2.0_5006_g503c27e77](https://anaconda.org/LiteX-Hub/openroad/2.0_5006_g503c27e77/download/linux-64/openroad-2.0_5006_g503c27e77-20220921_210118.tar.bz2) and earlier, where Magic and Netgen are the latest versions. 

Since the issue is clearly something to do with the openroad version change, do you have any idea for this behaviour with the recent openroad changes (was there any bugfix or something)?

saicharan0112, 2022-10-06:
I have also mentioned this in the skywater-pdk slack and here is the conversation thread - https://open-source-silicon.slack.com/archives/C016YSAP6TZ/p1665075690355409

msaligane, 2022-10-06:
Looks like it is a Magic issue. @saicharan0112 Can you open a github issue in Magic?

closed 2022-10-06 by msaligane

saicharan0112, 2022-10-08:
@maliberty As promised, here is the conclusion of this issue. The issue is with the update made to pdn file in https://github.com/The-OpenROAD-Project/OpenROAD/pull/2289 where power connections are done in a different way.