The-OpenROAD-Project/OpenROAD #2339: Resizer unable to optimize fanout for GF180 PDK

vijayank88, 2022-10-06:
Testing `aes` design with GF180 PDK through `OpenLane` flow.
Using `OpenROAD 4174c3ad802d2ac1d04d387d2c4b883903f6647e`

`resizer.log`
```
===========================================================================
report_checks -path_delay max (Setup)
============================================================================
Startpoint: reset_n (input port clocked by clk)
Endpoint: _38075_ (recovery check against rising-edge clock clk)
Path Group: **async_default**
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                  0.15    0.00    0.00   clock clk (rise edge)
                          0.00    0.00   clock network delay (ideal)
                         10.00   10.00 ^ input external delay
                  1.15    0.64   10.64 ^ reset_n (in)
     1    0.05                           reset_n (net)
                  1.15    0.00   10.64 ^ input10/I (gf180mcu_fd_sc_mcu7t5v0__buf_20)
                  0.83    0.72   11.36 ^ input10/Z (gf180mcu_fd_sc_mcu7t5v0__buf_20)
    86    0.72                           net10 (net)
                  0.87    0.09   11.45 ^ load_slew88/I (gf180mcu_fd_sc_mcu7t5v0__buf_20)
                  1.56    1.10   12.55 ^ load_slew88/Z (gf180mcu_fd_sc_mcu7t5v0__buf_20)
   179    1.36                           net88 (net)
                  1.58    0.10   12.65 ^ load_slew87/I (gf180mcu_fd_sc_mcu7t5v0__buf_20)
                  2.09    1.39   14.04 ^ load_slew87/Z (gf180mcu_fd_sc_mcu7t5v0__buf_20)
   236    1.87                           net87 (net)
                  2.12    0.14   14.17 ^ load_slew85/I (gf180mcu_fd_sc_mcu7t5v0__buf_20)
                  2.09    1.49   15.66 ^ load_slew85/Z (gf180mcu_fd_sc_mcu7t5v0__buf_20)
   236    1.80                           net85 (net)
                  2.11    0.12   15.77 ^ load_slew84/I (gf180mcu_fd_sc_mcu7t5v0__buf_20)
                  1.62    1.19   16.96 ^ load_slew84/Z (gf180mcu_fd_sc_mcu7t5v0__buf_20)
   185    1.40                           net84 (net)
                  1.63    0.04   17.00 ^ load_slew83/I (gf180mcu_fd_sc_mcu7t5v0__buf_20)
                  1.77    1.23   18.23 ^ load_slew83/Z (gf180mcu_fd_sc_mcu7t5v0__buf_20)
   210    1.55                           net83 (net)
                  1.90    0.28   18.51 ^ _38075_/RN (gf180mcu_fd_sc_mcu7t5v0__dffrnq_1)
                                 18.51   data arrival time

                  0.15   50.00   50.00   clock clk (rise edge)
                          0.00   50.00   clock network delay (ideal)
                         -0.25   49.75   clock uncertainty
                          0.00   49.75   clock reconvergence pessimism
                                 49.75 ^ _38075_/CLK (gf180mcu_fd_sc_mcu7t5v0__dffrnq_1)
                          0.00   49.75   library recovery time
                                 49.75   data required time
-----------------------------------------------------------------------------
                                 49.75   data required time
                                -18.51   data arrival time
-----------------------------------------------------------------------------
                                 31.24   slack (MET)
````
Test case: 
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9724859/issue_reproducible.zip)

maliberty, 2022-10-06:
Please run this with the latest OR and package a test case from that: incompatible database schema revision

maliberty, 2022-10-06:
I don't see any max_fanout specified in gf180

antonblanchard, 2022-10-06:
The SDC file has a fanout limit:

```
set_max_fanout 8.0000 [current_design]
```
but the liberty file is missing:

```
  default_fanout_load : 1.0000000000;
```

Same problem as we had on sky130.

maliberty, 2022-10-06:
@proppy @QuantamHD PDK issue...

QuantamHD, 2022-10-06:
Looking now.

maliberty, 2022-10-06:
@louiic see the note from Anton

QuantamHD, 2022-10-06:
@maliberty @antonblanchard I don't see a `default_fanout_load` in the NDA'd liberty. So we'll need to add it.

QuantamHD, 2022-10-06:
What is a sensible default?

antonblanchard, 2022-10-06:
> What is a sensible default?

I think 1, that's what we changed the sky130 PDK to use in https://github.com/RTimothyEdwards/open_pdks/commit/86f9b0c385d32b2d65fd789427f2ea3f0b7331d3

louiic, 2022-10-06:
Ethan,

I tried setting default_max_fanout in .LIB … did not help.

Louii

antonblanchard, 2022-10-06:
> Ethan, I tried setting default_max_fanout in .LIB … did not help.

Works for me. Are you sure your SDC has a `set_max_fanout` statement and your liberty file has `default_fanout_load : 1`?

baseline:

```
Startpoint: reset_n (input port clocked by clk)
Endpoint: _38075_ (recovery check against rising-edge clock clk)
Path Group: **async_default**
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                  0.15    0.00    0.00   clock clk (rise edge)
                          0.00    0.00   clock network delay (ideal)
                         10.00   10.00 ^ input external delay
                  0.19    0.07   10.07 ^ reset_n (in)
     1    0.00                           reset_n (net)
                  0.19    0.00   10.07 ^ input53/I (gf180mcu_fd_sc_mcu7t5v0__buf_2)
                  0.57    0.55   10.62 ^ input53/Z (gf180mcu_fd_sc_mcu7t5v0__buf_2)
     1    0.05                           net98 (net)
                  0.57    0.00   10.62 ^ input10/I (gf180mcu_fd_sc_mcu7t5v0__buf_20)
                  0.84    0.68   11.30 ^ input10/Z (gf180mcu_fd_sc_mcu7t5v0__buf_20)
    86    0.72                           net10 (net)
                  0.87    0.09   11.39 ^ repeater88/I (gf180mcu_fd_sc_mcu7t5v0__buf_20)
                  1.56    1.10   12.49 ^ repeater88/Z (gf180mcu_fd_sc_mcu7t5v0__buf_20)
   179    1.36                           net88 (net)
                  1.58    0.10   12.59 ^ repeater87/I (gf180mcu_fd_sc_mcu7t5v0__buf_20)
                  2.09    1.39   13.98 ^ repeater87/Z (gf180mcu_fd_sc_mcu7t5v0__buf_20)
   236    1.87                           net87 (net)
                  2.12    0.14   14.11 ^ repeater85/I (gf180mcu_fd_sc_mcu7t5v0__buf_20)
                  2.09    1.49   15.60 ^ repeater85/Z (gf180mcu_fd_sc_mcu7t5v0__buf_20)
   236    1.80                           net85 (net)
                  2.11    0.12   15.72 ^ repeater84/I (gf180mcu_fd_sc_mcu7t5v0__buf_20)
                  1.62    1.19   16.91 ^ repeater84/Z (gf180mcu_fd_sc_mcu7t5v0__buf_20)
   185    1.40                           net84 (net)
                  1.63    0.04   16.94 ^ repeater83/I (gf180mcu_fd_sc_mcu7t5v0__buf_20)
                  1.77    1.23   18.17 ^ repeater83/Z (gf180mcu_fd_sc_mcu7t5v0__buf_20)
   210    1.55                           net83 (net)
                  1.90    0.28   18.45 ^ _38075_/RN (gf180mcu_fd_sc_mcu7t5v0__dffrnq_1)
                                 18.45   data arrival time

                  0.15   50.00   50.00   clock clk (rise edge)
                          0.00   50.00   clock network delay (ideal)
                         -0.25   49.75   clock uncertainty
                          0.00   49.75   clock reconvergence pessimism
                                 49.75 ^ _38075_/CLK (gf180mcu_fd_sc_mcu7t5v0__dffrnq_1)
                          0.00   49.75   library recovery time
                                 49.75   data required time
-----------------------------------------------------------------------------
                                 49.75   data required time
                                -18.45   data arrival time
-----------------------------------------------------------------------------
                                 31.30   slack (MET)
```

With the fix:

```
Startpoint: reset_n (input port clocked by clk)
Endpoint: _37688_ (recovery check against rising-edge clock clk)
Path Group: **async_default**
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                  0.15    0.00    0.00   clock clk (rise edge)
                          0.00    0.00   clock network delay (ideal)
                         10.00   10.00 ^ input external delay
[... 75 lines cut ...]
                  0.62    0.64   23.86 ^ fanout124/Z (gf180mcu_fd_sc_mcu7t5v0__buf_1)
     4    0.02                           net169 (net)
                  0.62    0.00   23.86 ^ fanout120/I (gf180mcu_fd_sc_mcu7t5v0__buf_2)
                  0.71    0.70   24.56 ^ fanout120/Z (gf180mcu_fd_sc_mcu7t5v0__buf_2)
     8    0.06                           net165 (net)
                  0.71    0.00   24.56 ^ fanout119/I (gf180mcu_fd_sc_mcu7t5v0__buf_2)
                  0.73    0.72   25.28 ^ fanout119/Z (gf180mcu_fd_sc_mcu7t5v0__buf_2)
     8    0.06                           net164 (net)
                  0.73    0.00   25.28 ^ _37688_/RN (gf180mcu_fd_sc_mcu7t5v0__dffrnq_1)
                                 25.28   data arrival time

                  0.15   50.00   50.00   clock clk (rise edge)
                          0.00   50.00   clock network delay (ideal)
                         -0.25   49.75   clock uncertainty
                          0.00   49.75   clock reconvergence pessimism
                                 49.75 ^ _37688_/CLK (gf180mcu_fd_sc_mcu7t5v0__dffrnq_1)
                          0.33   50.08   library recovery time
                                 50.08   data required time
-----------------------------------------------------------------------------
                                 50.08   data required time
                                -25.28   data arrival time
-----------------------------------------------------------------------------
```

Are you sure you are setting

maliberty, 2022-10-06:
@antonblanchard  What happens to the area/power with this change?

louiic, 2022-10-06:
Anton,

Did all these … did not help

.LIB

  default_max_fanout : 1 ;
  default_fanout_load : 1 ;

.SDC

set_max_fanout 20 [current_design]

Louii

antonblanchard, 2022-10-07:
> @antonblanchard What happens to the area/power with this change? Timing is met either way

I was investigating why the design specified a max fanout, but OpenROAD ignored it. No idea what it does for area/performance.

@vijayank88 what is the issue here? As @maliberty mentions the design makes timing.

vijayank88, 2022-10-07:
with gf180 there are slew violations post routing. So just back traced to placement and found fanout issue. so raised this.
`max slew violation = 132`
`max cap violation = 32`

vijayank88, 2022-10-07:
@antonblanchard 
I have updated liberty file as follows:
```
slew_upper_threshold_pct_fall : 70 ; 
  slew_upper_threshold_pct_rise : 70 ; 
  default_max_fanout : 1.0000000000 ;
  default_fanout_load : 1.0000000000 ;
  nom_process : 1 ; 
```
Still having same fanout issue. Can you share snippet where its got updated?

antonblanchard, 2022-10-07:
@vijayank88 can you be more specific when you talk about "fanout issue"? Do you mean you have slew and capacitance violations when measured after global placement, or do you mean that the SDC fanout constraint is not being obeyed?

I see 0 slew and capacitance violations using your original test case, but I did have to convert your old odb by writing out a def on an old checkout of openroad and reading the def and writing the odb on a more recent one. Perhaps something critical in the odb was lost in the process.

The version of OpenROAD you are using looks to be pretty ancient, can you please recreate on a recent check out?

vijayank88, 2022-10-07:
@antonblanchard 
Issue is sdc fanout constraint not obeyed. 
Latest test case: 
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9731376/issue_reproducible.zip)

antonblanchard, 2022-10-07:
> @antonblanchard Issue is sdc fanout constraint not obeyed.

I took your most recent zip, made the following change:

```
--- ./tmp/synthesis/resizer_gf180mcu_fd_sc_mcu7t5v0__tt_025C_3v30.lib~	2022-10-07 20:23:36.805561983 +1100
+++ ./tmp/synthesis/resizer_gf180mcu_fd_sc_mcu7t5v0__tt_025C_3v30.lib	2022-10-07 20:23:58.070617591 +1100
@@ -23,6 +23,7 @@
   capacitive_load_unit(1, pf);
   pulling_resistance_unit : 1ohm ; 
   leakage_power_unit : 1uW ; 
+  default_fanout_load: 1;
   input_threshold_pct_fall : 50 ; 
   input_threshold_pct_rise : 50 ; 
   output_threshold_pct_fall : 50 ; 
```

And I see it work as expected:

```Startpoint: reset_n (input port clocked by clk)
Endpoint: _38616_ (recovery check against rising-edge clock clk)
Path Group: **async_default**
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                  0.15    0.00    0.00   clock clk (rise edge)
                          0.00    0.00   clock network delay (ideal)
                         10.00   10.00 ^ input external delay
                  0.16    0.05   10.05 ^ reset_n (in)
     1    0.00                           reset_n (net)
                  0.16    0.00   10.05 ^ input10/I (gf180mcu_fd_sc_mcu7t5v0__clkbuf_1)
                  0.25    0.34   10.40 ^ input10/Z (gf180mcu_fd_sc_mcu7t5v0__clkbuf_1)
     1    0.01                           net10 (net)
                  0.25    0.00   10.40 ^ fanout546/I (gf180mcu_fd_sc_mcu7t5v0__clkbuf_2)
                  0.45    0.56   10.96 ^ fanout546/Z (gf180mcu_fd_sc_mcu7t5v0__clkbuf_2)
     5    0.03                           net546 (net)
                  0.45    0.01   10.97 ^ fanout388/I (gf180mcu_fd_sc_mcu7t5v0__buf_1)
                  0.69    0.67   11.64 ^ fanout388/Z (gf180mcu_fd_sc_mcu7t5v0__buf_1)
     4    0.03                           net388 (net)
                  0.69    0.00   11.64 ^ fanout387/I (gf180mcu_fd_sc_mcu7t5v0__clkbuf_2)
                  0.45    0.65   12.29 ^ fanout387/Z (gf180mcu_fd_sc_mcu7t5v0__clkbuf_2)
     5    0.03                           net387 (net)
                  0.45    0.00   12.29 ^ fanout349/I (gf180mcu_fd_sc_mcu7t5v0__clkbuf_1)
                  0.50    0.56   12.85 ^ fanout349/Z (gf180mcu_fd_sc_mcu7t5v0__clkbuf_1)
     4    0.02                           net349 (net)
                  0.50    0.00   12.85 ^ fanout348/I (gf180mcu_fd_sc_mcu7t5v0__clkbuf_1)
                  0.52    0.58   13.43 ^ fanout348/Z (gf180mcu_fd_sc_mcu7t5v0__clkbuf_1)
     4    0.02                           net348 (net)
                  0.52    0.00   13.43 ^ fanout347/I (gf180mcu_fd_sc_mcu7t5v0__buf_1)
                  0.63    0.65   14.07 ^ fanout347/Z (gf180mcu_fd_sc_mcu7t5v0__buf_1)
     4    0.02                           net347 (net)
                  0.63    0.00   14.07 ^ fanout345/I (gf180mcu_fd_sc_mcu7t5v0__buf_1)
                  0.75    0.73   14.80 ^ fanout345/Z (gf180mcu_fd_sc_mcu7t5v0__buf_1)
     5    0.03                           net345 (net)
                  0.75    0.00   14.80 ^ fanout343/I (gf180mcu_fd_sc_mcu7t5v0__buf_1)
                  0.78    0.74   15.55 ^ fanout343/Z (gf180mcu_fd_sc_mcu7t5v0__buf_1)
     4    0.03                           net343 (net)
                  0.78    0.00   15.55 ^ _38616_/RN (gf180mcu_fd_sc_mcu7t5v0__dffrnq_1)
                                 15.55   data arrival time

                  0.15   50.00   50.00   clock clk (rise edge)
                          0.00   50.00   clock network delay (ideal)
                         -0.25   49.75   clock uncertainty
                          0.00   49.75   clock reconvergence pessimism
                                 49.75 ^ _38616_/CLK (gf180mcu_fd_sc_mcu7t5v0__dffrnq_1)
                          0.32   50.07   library recovery time
                                 50.07   data required time
-----------------------------------------------------------------------------
                                 50.07   data required time
                                -15.55   data arrival time
-----------------------------------------------------------------------------
                                 34.53   slack (MET)
```

vijayank88, 2022-10-07:
@antonblanchard 
Thanks. By updating `default_fanout_load : 1.0000000000 ;` into `liberty` files resolved fanout issue.
Can you please once verify is any other parameters to be updated in `.lib` files of GF180 PDK?
We can request @proppy to update this in PDK.

jjcherry56, 2022-10-07:
@vijayank88 @louiic @QuantamHD all confused default_max_fanout with default_fanout_load.
Contrary to common sense, fanout is defined by the liberty pin attribute fanout_load, which defaults to
the library attribute default_fanout_load, NOT the connections in the netlist.
If default_fanout_load is 0, the pin is not counted in the fanout of the driver and there can never be fanout violations. We went through this same nonsense with sky130.
As a result of that, opensta warns about this when reading the liberty, but of course no one actually looks at the warnings.

@louiic @vijayank88 it would never make any sense to set default_max_fanout to 1 in the liberty.
That would constrain all gates to only have one load.

QuantamHD, 2022-10-07:
@jjcherry56 What modification to the liberty file would you suggest?

jjcherry56, 2022-10-07:
The only change required is fixing default_fanout_load to 1

antonblanchard, 2022-10-11:
I created https://github.com/google/gf180mcu-pdk/issues/81

QuantamHD, 2022-11-23:
Can we validate that this is fixed with the upstream changes?

vijayank88, 2022-11-23:
@QuantamHD With latest PDK this issue got resolved.

closed 2022-11-23 by vijayank88