The-OpenROAD-Project/OpenROAD #2343: Par tool tests fail in Ubuntu 22

habibayassin, 2022-10-07:
Running `./test/regression par` in Ubuntu 22.04 with gcc version 11.2.0 causes the following tests to fail. Meanwhile, they pass on Ubuntu 20 with gcc version 9.4.0 and on Centos 7 with gcc version 4.8.5. 

```
partition (tcl) *FAIL* 
change_eval_method (tcl) *FAIL*
verilog_write (tcl) *FAIL*
report_netlist_partitions (tcl) *FAIL*
cluster (tcl) *FAIL*
```

diffs file
```
*** ~/OpenROAD/src/par/test/partition.ok	2022-10-04 08:42:59.098020854 +0000
--- ~/OpenROAD/src/par/test/results/partition-tcl.log	2022-10-07 21:24:00.706236225 +0000
***************
*** 8,11 ****
  [INFO PAR-0060] [MLPart] Run completed. Partition ID = 0. Total runs = 1.
  Writing partition id's to DB.
  Writing done.
! No differences found.
--- 8,13 ----
  [INFO PAR-0060] [MLPart] Run completed. Partition ID = 0. Total runs = 1.
  Writing partition id's to DB.
  Writing done.
! Differences found at line 1.
! _397_ 0
! _397_ 1
*** ~/OpenROAD/src/par/test/change_eval_method.ok	2022-10-04 08:42:59.098020854 +0000
--- ~/OpenROAD/src/par/test/results/change_eval_method-tcl.log	2022-10-07 21:24:01.322295143 +0000
***************
*** 20,35 ****
  [INFO PAR-0059] [MLPart] Partitioned graph for seed 1914837114.
  [INFO PAR-0060] [MLPart] Run completed. Partition ID = 1. Total runs = 2.
  [INFO PAR-0006] Partitioning Results for ID = 1 and Tool = mlpart.
! [INFO PAR-0007] Best results used seed 1914837114.
! [INFO PAR-0008] Number of Hyperedge Cuts = 53.
! [INFO PAR-0009] Number of Terminals = 116.
! [INFO PAR-0010] Cluster Size SD = 63.215405559088204.
! [INFO PAR-0011] Cluster Area SD = 756.9479423976262.
! [INFO PAR-0012] Total Hop Weight = 5300.
  [INFO PAR-0006] Partitioning Results for ID = 0 and Tool = mlpart.
! [INFO PAR-0007] Best results used seed -1731515372.
! [INFO PAR-0008] Number of Hyperedge Cuts = 52.
! [INFO PAR-0009] Number of Terminals = 113.
! [INFO PAR-0010] Cluster Size SD = 62.31121488143206.
! [INFO PAR-0011] Cluster Area SD = 804.9718550980525.
! [INFO PAR-0012] Total Hop Weight = 5200.
--- 20,35 ----
  [INFO PAR-0059] [MLPart] Partitioned graph for seed 1914837114.
  [INFO PAR-0060] [MLPart] Run completed. Partition ID = 1. Total runs = 2.
  [INFO PAR-0006] Partitioning Results for ID = 1 and Tool = mlpart.
! [INFO PAR-0007] Best results used seed -289663928.
! [INFO PAR-0008] Number of Hyperedge Cuts = 54.
! [INFO PAR-0009] Number of Terminals = 117.
! [INFO PAR-0010] Cluster Size SD = 62.72708745031926.
! [INFO PAR-0011] Cluster Area SD = 753.8910315821512.
! [INFO PAR-0012] Total Hop Weight = 5400.
  [INFO PAR-0006] Partitioning Results for ID = 0 and Tool = mlpart.
! [INFO PAR-0007] Best results used seed 1608637542.
! [INFO PAR-0008] Number of Hyperedge Cuts = 51.
! [INFO PAR-0009] Number of Terminals = 110.
! [INFO PAR-0010] Cluster Size SD = 61.82384248815339.
! [INFO PAR-0011] Cluster Area SD = 760.8601629077448.
! [INFO PAR-0012] Total Hop Weight = 5100.
*** ~/OpenROAD/src/par/test/verilog_write.ok	2022-10-04 08:42:59.098020854 +0000
--- ~/OpenROAD/src/par/test/results/verilog_write-tcl.log	2022-10-07 21:24:01.846345262 +0000
***************
*** 9,12 ****
  Writing partition id's to DB.
  Writing done.
  Writing partition to verilog.
! No differences found.
--- 9,14 ----
  Writing partition id's to DB.
  Writing done.
  Writing partition to verilog.
! Differences found at line 19.
!  wire partition__104_;
!  wire partition__136_;
*** ~/OpenROAD/src/par/test/report_netlist_partitions.ok	2022-10-04 08:42:59.098020854 +0000
--- ~/OpenROAD/src/par/test/results/report_netlist_partitions-tcl.log	2022-10-07 21:24:02.366394998 +0000
***************
*** 7,14 ****
  [INFO PAR-0059] [MLPart] Partitioned graph for seed 1608637542.
  [INFO PAR-0060] [MLPart] Run completed. Partition ID = 0. Total runs = 1.
  [INFO PAR-0019] The netlist has 4 partitions.
! [INFO PAR-0020] Partition 0 has 129 vertices.
! [INFO PAR-0020] Partition 1 has 11 vertices.
! [INFO PAR-0020] Partition 2 has 95 vertices.
! [INFO PAR-0020] Partition 3 has 180 vertices.
  [INFO PAR-0021] The total number of vertices is 415.
--- 7,14 ----
  [INFO PAR-0059] [MLPart] Partitioned graph for seed 1608637542.
  [INFO PAR-0060] [MLPart] Run completed. Partition ID = 0. Total runs = 1.
  [INFO PAR-0019] The netlist has 4 partitions.
! [INFO PAR-0020] Partition 0 has 118 vertices.
! [INFO PAR-0020] Partition 1 has 194 vertices.
! [INFO PAR-0020] Partition 2 has 93 vertices.
! [INFO PAR-0020] Partition 3 has 10 vertices.
  [INFO PAR-0021] The total number of vertices is 415.
*** ~/OpenROAD/src/par/test/cluster.ok	2022-10-04 08:42:59.098020854 +0000
--- ~/OpenROAD/src/par/test/results/cluster-tcl.log	2022-10-07 21:24:03.390492940 +0000
***************
*** 7,10 ****
  [INFO PAR-0064] [MLPart] Run completed. Cluster ID = 0.
  Writing cluster id's to DB.
  Writing done.
! No differences found.
--- 7,12 ----
  [INFO PAR-0064] [MLPart] Run completed. Cluster ID = 0.
  Writing cluster id's to DB.
  Writing done.
! Differences found at line 1.
! _397_ 68
! _397_ 69
```

@maliberty

maliberty, 2022-10-08:
This is going to be painful.  Where is there an ubuntu 22 machine that I can try this on?

QuantamHD, 2022-10-08:
You should be able to create one in Google Cloud

habibayassin, 2022-10-08:
> This is going to be painful. Where is there an ubuntu 22 machine that I can try this on?

there is a machine on GCP named "test-vm" I created recently that is ubuntu 22 based

maliberty, 2022-10-11:
MLpart is hard to debug legacy code and after several hours I don't see a real path to convergence.  We are planning to replace it with TritonPart so I don't think it is worth the time to sort this out.  I suggest we just disable this test and move on.

vijayank88, 2023-06-22:
@habibayassin 
Can you test again with latest build and update the status?

vijayank88, 2023-06-22:
regression unit test seems updated. Nothing fails for `par` with latest master.
```
./test/regression par
------------------------------------------------------
par
read_part (tcl) pass
partition_gcd (tcl) pass
partition_gcd (py) pass
------------------------------------------------------
Passed 2
See /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/src/par/test/results for log files
------------------------------------------------------
Passed all tool regressions.
```

closed 2023-06-22 by vijayank88