The-OpenROAD-Project/OpenROAD #2346: global routing failed

vijayank88, 2022-10-08:
OpenLane CI failures for latest OpenROAD updates: https://github.com/The-OpenROAD-Project/OpenLane/pull/1303

OpenROAD Commit: `OpenROAD v2.0-5170-gd919ca2d4 `
Test case:
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9738869/issue_reproducible.zip)

Logs:
```
[INFO GRT-0015] Inserted 5 diodes.
[INFO GRT-0009] rerouting 139 nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 11
[INFO GRT-0006] Repairing antennas, iteration 2.
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0012] Found 1 antenna violations.
[INFO GRT-0015] Inserted 1 diodes.
[INFO GRT-0009] rerouting 12 nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 11
[ERROR GRT-0228] Horizontal edge usage exceeds the maximum allowed.
Error: groute.tcl, 41 GRT-0228
while evaluating source run.tcl
```

vijayank88, 2022-10-08:
As per log if I change the `DIODE_INSERTION_STRATEGY` flow works fine and completes global routing stage.

maliberty, 2022-10-11:
I'm testing a fix now.

maliberty, 2022-10-11:
Apparently my fix takes care of the one design I looked at but there are still others that have issues...

vijayank88, 2022-10-11:
@maliberty 
Thanks. Let me know if you need test case for other issues.

maliberty, 2022-10-11:
Thanks but I extracted another myself

maliberty, 2022-10-11:
@luis201420 you can ignore the reproducer here and use the one I sent you for y_huff in slack

maliberty, 2022-12-15:
@vijayank88 is this still relevant?

vijayank88, 2022-12-15:
resolved.

closed 2022-12-15 by vijayank88