The-OpenROAD-Project/OpenROAD #2349: pdn : Secondary power net does get routed to instances in voltage domain

lulu9312, 2022-10-10:
There is a secondary power net VREG which is defined for a voltage domain. There are some instances that are placed in the voltage domain. The VREG net does not get routed to the instances even though their pins are defined as POWER in their respective lef files.
 Here are some screenshots for the same :
[image]

[image]

**To reproduce the case:**
Download 
[reproduce_case_routing.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9746197/reproduce_case_routing.tar.gz)

maliberty, 2022-10-10:
@arlpetergadfort thoughts?

arlpetergadfort, 2022-10-10:
@maliberty PDNgen doesn't route to instances (Except for blocks and pads). It's not clear what the expectation is here. This seems similar to the question about the point-to-point routing.

arlpetergadfort, 2022-10-10:
There is for the VDD and VSS, the secondary nets, do not get rails.

rovinski, 2022-10-11:
Yes, this sounds like the same feature request as #2209.

maliberty, 2025-08-25:
duplicate

closed 2025-08-25 by maliberty