The-OpenROAD-Project/OpenROAD #2350: DFT insertion

Blebowski, 2022-10-10:
Hello,

please take this issue as question only.

I would be curious what is the status of OpenRoad when it comes to DFT, is there a plan to eventually support
things like scan chain insertion, scan compression, ATPG(SAF, TF), Fault grading, etc?

I have searched for some open-source solutions, but it seems to me there is nothing practically usable.

maliberty, 2022-10-10:
There is currently no plan but we would be glad to have submissions.  The only OSS tool I know about is https://github.com/AUCOHL/Fault with the open  OL issue https://github.com/The-OpenROAD-Project/OpenLane/issues/966 to integrate it.

QuantamHD, 2022-10-10:
We link in the program [ABC](https://github.com/berkeley-abc/abc) which has commands for ATPG, so there are some tools that you could use to integrate.

proppy, 2023-06-22:
I think @fgaray has done work recently on this.

fgaray, 2023-06-22:
Hi @Blebowski ,

As @proppy said, recently I started working on scan chain insertion, you can try some commands like preview_dft or insert_dft. Check the readme here: https://github.com/The-OpenROAD-Project/OpenROAD/tree/master/src/dft

Just be aware that this is in a early state and we don't support any optimizations or scan compression, just basic scan chain creation with the option of selecting the clock mixing.