The-OpenROAD-Project/OpenROAD #2354: rsz: max wire length repair creates slew violations

antonblanchard, 2022-10-10:
While running some global placement experiements I noticed `repair_design` didn't fix all slew violations. To reproduce you need the following change to OpenROAD, which just reduces the percentage of nets that are optimised for timing from 10% to 1% (I wasn't easily able to create a reproducer without it unfortunately):

```
diff --git a/src/rsz/src/Resizer.cc b/src/rsz/src/Resizer.cc
index f1d7d8e87..2efe7273d 100644
--- a/src/rsz/src/Resizer.cc
+++ b/src/rsz/src/Resizer.cc
@@ -934,7 +934,7 @@ Resizer::findResizeSlacks1()
   }
 
   // Find the nets with the worst slack.
-  double worst_percent = .1;
+  double worst_percent = .01;
   //  sort(nets.begin(), nets.end(). [&](const Net *net1,
   sort(nets, [this](const Net *net1,
                  const Net *net2)
```

`repair_design` completes without error:

```
[INFO RSZ-0027] Inserted 19 input buffers.
[INFO RSZ-0028] Inserted 26 output buffers.
[INFO RSZ-0058] Using max wire length 162um.
[INFO RSZ-0034] Found 5889 slew violations.
[INFO RSZ-0036] Found 19 capacitance violations.
[INFO RSZ-0037] Found 18 long wires.
[INFO RSZ-0038] Inserted 455 buffers in 5890 nets.
[INFO RSZ-0039] Resized 6199 instances.
```

But there are slew violations:

```
max slew

Pin                                    Limit    Slew   Slack
------------------------------------------------------------
_077131_/A                            320.00  332.36  -12.36 (VIOLATED)
_077262_/A                            320.00  332.14  -12.14 (VIOLATED)
_078465_/A2                           320.00  331.92  -11.92 (VIOLATED)
_077651_/A2                           320.00  331.56  -11.56 (VIOLATED)
_077769_/A2                           320.00  331.19  -11.19 (VIOLATED)
_077991_/A2                           320.00  331.12  -11.12 (VIOLATED)
_077794_/A1                           320.00  331.05  -11.05 (VIOLATED)
_077034_/A                            320.00  330.69  -10.69 (VIOLATED)
_078402_/A1                           320.00  330.49  -10.49 (VIOLATED)
_077041_/A                            320.00  329.44   -9.44 (VIOLATED)
_077145_/A1                           320.00  329.29   -9.29 (VIOLATED)
_078419_/A1                           320.00  329.07   -9.07 (VIOLATED)
_078005_/A2                           320.00  328.23   -8.23 (VIOLATED)
_077424_/A                            320.00  328.23   -8.23 (VIOLATED)
_077027_/A                            320.00  328.14   -8.14 (VIOLATED)
_077294_/A                            320.00  327.60   -7.60 (VIOLATED)
_076882_/A                            320.00  327.48   -7.48 (VIOLATED)
_076804_/B                            320.00  326.16   -6.16 (VIOLATED)
_077668_/A1                           320.00  323.80   -3.80 (VIOLATED)
_077363_/A                            320.00  323.76   -3.76 (VIOLATED)
_077116_/B                            320.00  323.51   -3.51 (VIOLATED)
_077788_/A                            320.00  320.38   -0.38 (VIOLATED)
_077278_/A                            320.00  320.32   -0.32 (VIOLATED)
_077717_/A2                           320.00  320.25   -0.25 (VIOLATED)
_076790_/B                            320.00  320.16   -0.16 (VIOLATED)
```

All of these are on the same net:

```
    - net2587 ( max_length2587 Y ) ( _077241_ A2 ) ( _076802_ A ) ( _076890_ A2 ) ( _077668_ A1 ) ( _077363_ A ) ( _077294_ A )
      ( _076882_ A ) ( _077041_ A ) ( _077145_ A1 ) ( _078465_ A2 ) ( _077262_ A ) ( _077131_ A ) ( _077651_ A2 ) ( _077769_ A2 )
      ( _077991_ A2 ) ( _077794_ A1 ) ( _077034_ A ) ( _078402_ A1 ) ( _078419_ A1 ) ( _078005_ A2 ) ( _077424_ A ) ( _077027_ A )
      ( _076804_ B ) ( _077116_ B ) ( _076790_ B ) ( _107777_ B ) ( _077788_ A ) ( _077278_ A ) ( _077717_ A2 ) ( _077985_ A2 )
      ( _077357_ A ) ( _077288_ A ) ( _078326_ A ) ( _076896_ A ) ( _077061_ A ) ( _077018_ A1 ) ( _077245_ A ) ( _078343_ A1 )
      ( _107775_ B ) ( _107774_ B ) ( _077411_ B ) ( _077409_ B ) ( _107779_ B ) ( _077412_ A1 ) ( _077390_ B ) ( _076864_ B )
      ( _077393_ B ) ( _078494_ A1 ) + USE SIGNAL ;
```

Since this net is driven by `max_length2587` I presume max wire repair was responsible for the slew violations.

Test case: [repair_design_slew.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9749690/repair_design_slew.tar.gz)

jjcherry56, 2022-10-11:
I cannot reproduce this.  I get different results from you using OR
commit 64a40a83a Merge pull request #2351 from
The-OpenROAD-Project-staging/gui-fanout on macos and linux (and those
results do not agree with each other).  This is not surprising, I see
this all the time and you did not state what OR commit shows the
problem.  If you write the def or db after global placement it may be
possible to reproduce it because the variation may in the gpl results.

antonblanchard, 2022-10-11:
@jjcherry56 You're right, I should have just saved the def post global placement. I retested on 93f69d5c5922 with an updated test case and it reproduced:

[repair_design_slew-2.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9759506/repair_design_slew-2.tar.gz)

jjcherry56, 2022-10-11:
Thanks, I can reproduce it now

jjcherry56, 2022-11-01:
fixed in OR e589eac58

closed 2022-11-01 by jjcherry56