The-OpenROAD-Project/OpenROAD #2364: Post routing timing not met with STA

vijayank88, 2022-10-13:
Testing this design with GF180 PDK with OpenLane flow.

OpenROAD Commit: 79313e90ded2667486b33a87effb23bc56d41bf1

Description:

The problem is design meets timing till global routing stage, but post routing its not met timing.

Global routing log:
```
===========================================================================
 report_tns
============================================================================
tns -28.20
tns_report_end
wns_report

===========================================================================
 report_wns
============================================================================
wns -1.06
wns_report_end
worst_slack

===========================================================================
 report_worst_slack -max (Setup)
============================================================================
worst slack -1.06

===========================================================================
 report_worst_slack -min (Hold)
============================================================================
worst slack 0.18
worst_slack_end
clock_skew

===========================================================================
 report_clock_skew
============================================================================
Clock clock
Latency      CRPR       Skew
_58809_/CLK ^
  12.52
_57348_/CLK ^
   3.84      0.00       8.69

clock_skew_end
==========================================================
```
Post routing Timing report:
```
===========================================================================
max slew violation count 130
max fanout violation count 3420
max cap violation count 39
============================================================================
check_slew_end
tns_report

===========================================================================
 report_tns
============================================================================
tns -29404.88
tns_report_end
wns_report

===========================================================================
 report_wns
============================================================================
wns -33.46
wns_report_end
worst_slack

===========================================================================
 report_worst_slack -max (Setup)
============================================================================
worst slack -33.46

===========================================================================
 report_worst_slack -min (Hold)
============================================================================
worst slack 0.20
worst_slack_end
clock_skew

===========================================================================
 report_clock_skew
============================================================================
Clock clock
Latency      CRPR       Skew
_58809_/CLK ^
  22.80
_57348_/CLK ^
   6.12      0.00      16.69

clock_skew_end
power_report
==============================================
```
Attached global route test case and post routing timing report test case.

antonblanchard, 2022-10-13:
Openlane doesn't have any RC parasitic estimates for either sky130 or gf180 - the values in the TLEF are not good enough. Without reasonable estimates global routing and detailed routing STA are going to diverge.

ORFS has scripts to calculate them from a series of real designs.

tspyrou, 2022-10-13:
@vijayank88 can you take the setrc.tcl file from openroad-flow-scripts/flows/platforms/sky130 that matches the platform for this testcase and source it in this run? If the timing is much better then this needs to become an openlane issue. Thanks @antonblanchard !

vijayank88, 2022-10-14:
@antonblanchard 
[set_rc.tcl](https://github.com/The-OpenROAD-Project/OpenLane/blob/3ee8cf0da0b5292557e3f3ae398194d65970a6d2/script/openroad/common/set_rc.tcl) sourced in OpenLane flow.
But I'm not seen `::env(VIAS_RC)` `::env(LAYERS_RC)` configuration anywhere defined.
Do you aware where it's been used?

vijayank88, 2022-10-14:
@tspyrou 
I've copied [setRC.tcl](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/sky130hd/setRC.tcl) from ORFS flow and ran. But not seen any improvement wrt timing report.
Changed only  layer name `met` to `Metal`. Is RC values need any update?

tspyrou, 2022-10-20:
@vijayank88 in the timing report after placement based timing optimization, does openroad think that the timing is met? If no, this is an openroad issue, if yes this is a setRC.tcl setup issue. Can you let me know?

vijayank88, 2022-10-20:
@tspyrou 
Placement stage timing met and resizer log:
```
===========================================================================
 report_worst_slack -max (Setup)
============================================================================
worst slack 1.72

===========================================================================
 report_worst_slack -min (Hold)
============================================================================
worst slack 0.53
worst_slack_end
clock_skew
```
So, its a `setRC.tcl` setup issue.

vijayank88, 2022-10-28:
@tspyrou 
Shall I transfer this issue to OpenLane directly?
Because some of MPW CI designs as well facing same issue. 
Placement/CTS/Global routing stage slack met. But during signoff stage facing setup/hold violations and flow get failed.