The-OpenROAD-Project/OpenROAD #2377: grt: drt pin_access contributes to RC parasitic errors

antonblanchard, 2022-10-16:
I'm running tests with `pin_access` called before `global_routing`. I notice a number of short nets have significantly higher resistances. In this case (`_03345_`) grt estimates 88 ohms but after detailed routing it is 41 ohms. Capacitance is also off by an order of magnitude.

Looking at the debug:

```
[DEBUG GRT-0001] net _03345_
(33.21, 64.53)  1 -> (33.21, 64.53)  2 l=0.00
[DEBUG GRT-0001] _03345_:0 -> _25665_/B (33.44, 64.59) 0.30u layer=2 r=0.009 via_res=0.017 c=0.043
[DEBUG GRT-0001] pin _25665_/B -> to grid 0u layer=2 r=0.009 via_res=0.017 c=0.043
[DEBUG GRT-0001] _03345_:0 -> _25664_/Y (32.20, 64.56) 1.03u layer=2 r=0.030 via_res=0.017 c=0.151
[DEBUG GRT-0001] pin _25664_/Y -> to grid 1u layer=2 r=0.030 via_res=0.017 c=0.151
 Pi model c2=8.68e-17 rpi=72.1 c1=2.29e-15 max_r=73.6
 Pi model c2=8.68e-17 rpi=72.1 c1=2.29e-15 max_r=73.6
```

The yellow dots show the points grt is using. It seems it always selects the bottom left position (I guess the first valid position?):

[image]

Would it be possible to choose a better access point so grt can make better estimates (eg the green point)? I presume drt is doing that pretty early on.

antonblanchard, 2022-10-17:
The issue is with `_25664_/Y`, and if I'm reading this right, pin access did choose a location on the right of the cell:

```
 gcclean2via _25664_ A VIA23 189 81 MX
 gcclean2via _25664_ C VIA12 549 117 MX
 gcclean2via _25664_ B VIA12 729 117 MX
 gcclean2via _25664_ Y VIA12 1053 45 MX
```

It looks like the issue is GRT is using pin access, but the GRT parasitics are not.

maliberty, 2022-10-17:
I think you are putting to much focus on very small nets that have very little impact on timing.  The percentage error sounds impressive but the absolute values are trivial.  I'll look at this if you want but I suspect it has little to do with the aps.  You'll need to provide a test case.

antonblanchard, 2022-10-17:
> I think you are putting to much focus on very small nets that have very little impact on timing. The percentage error sounds impressive but the absolute values are trivial. I'll look at this if you want but I suspect it has little to do with the aps. You'll need to provide a test case.

That's fair. At a high level I'm trying to reduce the differences I see in STA between grt and drt. I am still concerned that these small nets are contributing to errors when calculating our RC parasitic estimates (ie ORFS `util/correlateRC.py`), but I'm yet to prove if it is.

antonblanchard, 2022-10-21:
Fix has been merged, closing

closed 2022-10-21 by antonblanchard