The-OpenROAD-Project/OpenROAD #2382: Having trouble reading in lef files

gbrown40, 2022-10-19:
When I read in lefs from the open source Ariane RISC-V design, I am getting issues like:

[WARNING ODB-0176] error: undefined layer (metal1) referenced
[WARNING ODB-0186] macro XOR2_X2 references unknown site FreePDK45_38x28_10R_NP_162NW_34O

Looking at the lef files, layers that cannot be found when reading in do exist, so I'm not sure why they are causing problems. These issues cause the lef to be read in only partially, leaving out some of the design. I am wondering if there is anything I can do to remedy these reading errors.

QuantamHD, 2022-10-19:
Could you provide a link to the tech lef you are using?

maliberty, 2022-10-19:
Please package this as a standalone test case (lefs & tcl script) and upload it.

openroadie, 2022-10-24:
@maliberty I can take a look once the test case is available.

gbrown40, 2022-10-24:
Sorry for the late reply, where should I upload the files?

gbrown40, 2022-10-27:
Here are the lef files I am trying to read in:
[lefs.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9882416/lefs.zip)

vijayank88, 2022-10-28:
@gbrown40 
May I know which OpenROAD version? 
`openroad --version`

vijayank88, 2022-10-29:
I'm not used python API. But you're using  2 months old commit. https://github.com/The-OpenROAD-Project/OpenROAD/search?q=49fdec96d&type=commits . Update OpenROAD and test it.

maliberty, 2024-02-29:
No activity - please reopen if still relevant and a test case can be provided.

closed 2024-02-29 by maliberty