The-OpenROAD-Project/OpenROAD #2383: Standard cells of type buffer are not connected to power

lulu9312, 2022-10-19:
Buffer cells does not connect to power rails.

**Here's the screenshot :**
[image]

I saw a similar issue while running gcd design as well.

lulu9312, 2022-10-19:
I tried uploading file to reproduce the issue but I don't know for some reason it's not uploading.

maliberty, 2022-10-19:
Only certain file types are allowed by GH so perhaps that is your issue.  Do you have a global_connect statement to connect these instances?

lulu9312, 2022-10-20:
Yes. I have used global_connect statement.
Here's the file to reproduce this case:
[detail_route_ldo_sky130hvl_base_2022-10-20_10-22.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9826400/detail_route_ldo_sky130hvl_base_2022-10-20_10-22.tar.gz)

maliberty, 2022-10-21:
It appears that you need to run your global connect statements again as the instance in question must have been created afterwards.

@arlpetergadfort does you recent change in odb provide a way to do this from the db or do the commands need to be run again?

arlpetergadfort, 2022-10-21:
@maliberty if the flow is fully in ODB, then yes, the rules are saves in the database and they can be rerun at anytime with `global_connect`.

maliberty, 2022-10-24:
@lulu9312 what flow are you using?  ORFS? OL? other?

lulu9312, 2022-10-25:
@maliberty I am using ORFS flow. I have added global_connect statement before grobal route, now the cells are connected to power.

[image]

closed 2022-10-25 by maliberty