The-OpenROAD-Project/OpenROAD #2386: FMTLib OpenROAD Compile

QuantamHD, 2022-10-19:
**Describe the bug**
New versions of FMTLib no longer allow formatting of user defined types. 

**Expected Behavior**
OpenROAD should compile with newer versions of fmtlib. We should add a CI check to make sure that OpenROAD compiles with new versions of fmtlib so that we don't introduce more cases like this.

**Environment**

Debian Testing
```
YOUR SURVEY HERE
```
i
**To Reproduce**

i
**Logs**
i
**Screenshots**
i
**Additional context**
i

maliberty, 2022-10-19:
I think fmt has been horrible broken by their new changes.  See https://github.com/fmtlib/fmt/issues/3088 (and https://github.com/fmtlib/fmt/issues/3081).

ekliptik, 2022-11-15:
I got it to build with new versions through the provided workaround. I added `add_compile_definitions(FMT_DEPRECATED_OSTREAM)` to CMakeLists.txt of fmtlib, built and `make install`ed it, then I also built spdgen and OpenROAD from source.

maliberty, 2022-11-15:
Is it necessary to recompile fmtlib in order to use FMT_DEPRECATED_OSTREAM?  Can you just add it to OR?

ekliptik, 2022-11-15:
Yes, it is necessary. It's behind an `#ifdef` directive, so it needs to be set at fmtlib compilation. It only applies to new versions of fmtlib, which CentOS and Debian might not have yet, I don't know how packaging there really works. A patched fmtlib fork could be maintained alongside OpenROAD, used as a submodule, and CMake somehow instructed to ignore local fmtlib instances and always use the forked one to build OpenROAD. Or we can ask the maintainers to expose the flag in their CMakeLists so an unpatched version can be built with the flag passed to CMake.

fgaray, 2023-12-28:
Confirmed that this was fixed by #4351, I can compile in debian testing but keep in mind that currently it is shipping with fmt [9.1.0](https://packages.debian.org/search?keywords=libfmt&searchon=names&suite=bookworm&section=all) so you may want to manually compile version >=10 or wait until it is available in the repos.

I think you may want to close this @maliberty or @QuantamHD

closed 2023-12-28 by maliberty