The-OpenROAD-Project/OpenROAD #2395: grt: read_guides doesn't clear existing guides from database

antonblanchard, 2022-10-21:
**Describe the bug**
`read_guides` partially appends to the previous guides, such that RC parasitic estimates change.

**Expected Behavior**
I would expect either the previous guides to be completely cleared, or the old guides and new guides should be successfully merged.

**Environment**
```
Git commit: a01312143a07839e13ed723264d8e34b87b4d750
kernel: Linux 5.19.14-200.fc36.x86_64
os: Fedora Linux 36 (Workstation Edition)
cmake version 3.22.2
gcc (GCC) 12.2.1 20220819 (Red Hat 12.2.1-2)
clang version 14.0.5 (Fedora 14.0.5-1.fc36)
```

**To Reproduce**

With the attached test case, `doit_0.tcl` uses the data directly in the database after calling `global_route`:

```
openroad doit_0.tcl | grep -A 20 'net final_adder.o.*11'
```

```
[DEBUG GRT-0001] net final_adder.o\[11\]
(5.67, 1.35)  1 -> (5.67, 1.35)  2 l=0.00
(5.67, 1.35)  2 -> (21.87, 1.35)  2 l=16.20
(21.87, 1.35)  2 -> (21.87, 1.35)  3 l=0.00
(21.87, 1.35)  3 -> (21.87, 5.13)  3 l=3.78
(21.87, 5.13)  3 -> (21.87, 5.13)  2 l=0.00
(21.87, 5.13)  2 -> (22.41, 5.13)  2 l=0.54
(22.41, 5.13)  2 -> (22.41, 5.13)  1 l=0.00
[DEBUG GRT-0001] final_adder.o\[11\]:0 -> final_adder.o\[11\]:1 16.20u layer=2 r=0.392 c=2.175
[DEBUG GRT-0001] final_adder.o\[11\]:1 -> final_adder.o\[11\]:2 via 2-3 r=0.017
[DEBUG GRT-0001] final_adder.o\[11\]:2 -> final_adder.o\[11\]:3 3.78u layer=3 r=0.092 c=0.488
[DEBUG GRT-0001] final_adder.o\[11\]:3 -> final_adder.o\[11\]:4 via 3-2 r=0.017
[DEBUG GRT-0001] final_adder.o\[11\]:4 -> final_adder.o\[11\]:5 0.54u layer=2 r=0.013 c=0.073
[DEBUG GRT-0001] final_adder.o\[11\]:5 -> final_adder.U$$845/Y (22.61, 5.16) 0.23u layer=2 r=0.006 via_res=0.017 c=0.031
[DEBUG GRT-0001] pin final_adder.U$$845/Y -> to grid 0u layer=2 r=0.006 via_res=0.017 c=0.031
[DEBUG GRT-0001] final_adder.o\[11\]:0 -> _2303_/D (5.74, 1.38) 0.10u layer=2 r=0.002 via_res=0.017 c=0.013
[DEBUG GRT-0001] pin _2303_/D -> to grid 0u layer=2 r=0.002 via_res=0.017 c=0.013
 Pi model c2=6.5e-16 rpi=364 c1=2.75e-15 max_r=574
 Pi model c2=6.5e-16 rpi=364 c1=2.75e-15 max_r=574
```

If we add a call to `read_guides`, we should get the data, but instead we have a number of new nodes, and the resistance has dropped from 364 ohms to 332 ohms, presumably because we have some parallel paths:

```
openroad doit_1.tcl | grep -A 30 'net final_adder.o.*11'
```

```
[DEBUG GRT-0001] net final_adder.o\[11\]
(5.67, 1.35)  1 -> (5.67, 1.35)  2 l=0.00
(5.67, 1.35)  2 -> (21.87, 1.35)  2 l=16.20
(21.87, 1.35)  2 -> (21.87, 1.35)  3 l=0.00
(21.87, 1.35)  3 -> (21.87, 5.13)  3 l=3.78
(21.87, 5.13)  3 -> (21.87, 5.13)  2 l=0.00
(21.87, 5.13)  2 -> (22.41, 5.13)  2 l=0.54
(22.41, 5.13)  2 -> (22.41, 5.13)  1 l=0.00
(5.67, 1.35)  1 -> (5.67, 1.35)  1 l=0.00
(5.67, 1.35)  2 -> (21.87, 1.35)  2 l=16.20
(21.87, 1.35)  3 -> (21.87, 5.13)  3 l=3.78
(21.87, 5.13)  2 -> (22.41, 5.13)  2 l=0.54
(22.41, 5.13)  1 -> (22.41, 5.13)  1 l=0.00
[DEBUG GRT-0001] final_adder.o\[11\]:0 -> final_adder.o\[11\]:1 16.20u layer=2 r=0.392 c=2.175
[DEBUG GRT-0001] final_adder.o\[11\]:1 -> final_adder.o\[11\]:2 via 2-3 r=0.017
[DEBUG GRT-0001] final_adder.o\[11\]:2 -> final_adder.o\[11\]:3 3.78u layer=3 r=0.092 c=0.488
[DEBUG GRT-0001] final_adder.o\[11\]:3 -> final_adder.o\[11\]:4 via 3-2 r=0.017
[DEBUG GRT-0001] final_adder.o\[11\]:4 -> final_adder.o\[11\]:5 0.54u layer=2 r=0.013 c=0.073
[DEBUG GRT-0001] final_adder.o\[11\]:0 -> final_adder.o\[11\]:1 16.20u layer=2 r=0.392 c=2.175
[DEBUG GRT-0001] final_adder.o\[11\]:2 -> final_adder.o\[11\]:3 3.78u layer=3 r=0.092 c=0.488
[DEBUG GRT-0001] final_adder.o\[11\]:4 -> final_adder.o\[11\]:5 0.54u layer=2 r=0.013 c=0.073
[DEBUG GRT-0001] final_adder.o\[11\]:5 -> final_adder.U$$845/Y (22.61, 5.16) 0.23u layer=2 r=0.006 via_res=0.017 c=0.031
[DEBUG GRT-0001] pin final_adder.U$$845/Y -> to grid 0u layer=2 r=0.006 via_res=0.017 c=0.031
[DEBUG GRT-0001] final_adder.o\[11\]:0 -> _2303_/D (5.74, 1.38) 0.10u layer=2 r=0.002 via_res=0.017 c=0.013
[DEBUG GRT-0001] pin _2303_/D -> to grid 0u layer=2 r=0.002 via_res=0.017 c=0.013
 Pi model c2=1.16e-15 rpi=332 c1=4.98e-15 max_r=574
 Pi model c2=1.16e-15 rpi=331 c1=4.97e-15 max_r=574
```

If we call `read_guides` twice, we die:

```
# openroad doit_2.tcl 
...
Stack trace:
 0# 0x0000000000C3D57E in openroad
 1# 0x00007FA837A3EA70 in /lib64/libc.so.6
 2# grt::FastRouteCore::releaseNetResources(int) in openroad
 3# grt::FastRouteCore::clearNetRoute(int) in openroad
 4# grt::FastRouteCore::addNet(odb::dbNet*, int, bool, int, int, int, int, float, std::vector<int, std::allocator<int> >*) in openroad
 5# grt::GlobalRouter::makeFastrouteNet(grt::Net*) in openroad
 6# grt::GlobalRouter::initNets(std::vector<grt::Net*, std::allocator<grt::Net*> >&) in openroad
 7# grt::GlobalRouter::readGuides(char const*) in openroad
```

Test case: [read_guides_issue.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9835524/read_guides_issue.tar.gz)

eder-matheus, 2022-10-21:
@antonblanchard Could you check this PR: https://github.com/The-OpenROAD-Project/OpenROAD/pull/2397? I may still see some differences in the debug prints on how the topology was constructed, but they should be equivalent, and the parasitic values should be the same.

Also, you should not see the crash again.

closed 2022-10-21 by eder-matheus

antonblanchard, 2022-10-21:
Thanks @eder-matheus. This may be a different issue, but I still see a significant difference in the RC parasitic estimates between using the guides in the db, and `read_guides`:

With the same test case and using the guides from the db, things looks reasonable. The parasitic resistance estimate is 364 ohms:

```
# openroad doit_0.tcl |grep -A 20 'net final_adder.o.*11'
[DEBUG GRT-0001] net final_adder.o\[11\]
(5.67, 1.35)  1 -> (5.67, 1.35)  2 l=0.00
(5.67, 1.35)  2 -> (21.87, 1.35)  2 l=16.20
(21.87, 1.35)  2 -> (21.87, 1.35)  3 l=0.00
(21.87, 1.35)  3 -> (21.87, 5.13)  3 l=3.78
(21.87, 5.13)  3 -> (21.87, 5.13)  2 l=0.00
(21.87, 5.13)  2 -> (22.41, 5.13)  2 l=0.54
(22.41, 5.13)  2 -> (22.41, 5.13)  1 l=0.00
[DEBUG GRT-0001] final_adder.o\[11\]:0 -> final_adder.o\[11\]:1 16.20u layer=2 r=0.392 c=2.175
[DEBUG GRT-0001] final_adder.o\[11\]:1 -> final_adder.o\[11\]:2 via 2-3 r=0.017
[DEBUG GRT-0001] final_adder.o\[11\]:2 -> final_adder.o\[11\]:3 3.78u layer=3 r=0.092 c=0.488
[DEBUG GRT-0001] final_adder.o\[11\]:3 -> final_adder.o\[11\]:4 via 3-2 r=0.017
[DEBUG GRT-0001] final_adder.o\[11\]:4 -> final_adder.o\[11\]:5 0.54u layer=2 r=0.013 c=0.073
[DEBUG GRT-0001] final_adder.o\[11\]:5 -> final_adder.U$$845/Y (22.61, 5.16) 0.23u layer=2 r=0.006 via_res=0.017 c=0.031
[DEBUG GRT-0001] pin final_adder.U$$845/Y -> to grid 0u layer=2 r=0.006 via_res=0.017 c=0.031
[DEBUG GRT-0001] final_adder.o\[11\]:0 -> _2303_/D (5.74, 1.38) 0.10u layer=2 r=0.002 via_res=0.017 c=0.013
[DEBUG GRT-0001] pin _2303_/D -> to grid 0u layer=2 r=0.002 via_res=0.017 c=0.013
 Pi model c2=6.5e-16 rpi=364 c1=2.75e-15 max_r=574
 Pi model c2=6.5e-16 rpi=364 c1=2.75e-15 max_r=574
```

Running the test where we read use `read_guides`, the parasitic resistance estimate is way off (25.5 ohm), and it looks like we are missing some nodes:

```
# openroad doit_1.tcl |grep -A 20 'net final_adder.o.*11'
[DEBUG GRT-0001] net final_adder.o\[11\]
(5.67, 1.35)  1 -> (5.67, 1.35)  1 l=0.00
(5.67, 1.35)  2 -> (21.87, 1.35)  2 l=16.20
(21.87, 1.35)  3 -> (21.87, 5.13)  3 l=3.78
(21.87, 5.13)  2 -> (22.41, 5.13)  2 l=0.54
(22.41, 5.13)  1 -> (22.41, 5.13)  1 l=0.00
[DEBUG GRT-0001] final_adder.o\[11\]:0 -> final_adder.o\[11\]:1 16.20u layer=2 r=0.392 c=2.175
[DEBUG GRT-0001] final_adder.o\[11\]:2 -> final_adder.o\[11\]:3 3.78u layer=3 r=0.092 c=0.488
[DEBUG GRT-0001] final_adder.o\[11\]:4 -> final_adder.o\[11\]:5 0.54u layer=2 r=0.013 c=0.073
[DEBUG GRT-0001] final_adder.o\[11\]:5 -> final_adder.U$$845/Y (22.61, 5.16) 0.23u layer=2 r=0.006 via_res=0.017 c=0.031
[DEBUG GRT-0001] pin final_adder.U$$845/Y -> to grid 0u layer=2 r=0.006 via_res=0.017 c=0.031
[DEBUG GRT-0001] final_adder.o\[11\]:0 -> _2303_/D (5.74, 1.38) 0.10u layer=2 r=0.002 via_res=0.017 c=0.013
[DEBUG GRT-0001] pin _2303_/D -> to grid 0u layer=2 r=0.002 via_res=0.017 c=0.013
 Pi model c2=1.64e-17 rpi=25.5 c1=8.69e-17 max_r=35.8
```

antonblanchard, 2022-10-21:
Thinking about it some more, it almost certainly is an unrelated issue. I'm happy to open a new issue if you prefer.

eder-matheus, 2022-10-21:
Thanks for the feedback, @antonblanchard. I think a new issue would be better to avoid confusion with two different issues here, if you don't mind.