The-OpenROAD-Project/OpenROAD #2399: grt: RC parasitic estimates wrong with read_guides

antonblanchard, 2022-10-22:
**Describe the bug**
There are issues with RC parasitic estimates when using `read_guides`.

**Expected Behavior**
RC parasitic estimates should be the same, whether the data was sourced from the odb or `read_guides`

**Environment**
```
Git commit: 53c8d2fa59bcde425bdf566f48cad543b999c6a2
kernel: Linux 5.19.14-200.fc36.x86_64
os: Fedora Linux 36 (Workstation Edition)
cmake version 3.22.2
gcc (GCC) 12.2.1 20220819 (Red Hat 12.2.1-2)
clang version 14.0.5 (Fedora 14.0.5-1.fc36)
```

**To Reproduce**

The following test case shows the issue:

```
# openroad parasitics.tcl
...
[DEBUG GRT-0001] net net_M3
(14.85, 9.99)  1 -> (14.85, 9.99)  1 l=0.00
(14.85, 9.99)  2 -> (14.85, 9.99)  2 l=0.00
(14.85, 9.99)  3 -> (14.85, 89.91)  3 l=79.92
(14.85, 89.91)  2 -> (14.85, 89.91)  2 l=0.00
(14.85, 89.91)  1 -> (14.85, 89.91)  1 l=0.00
[DEBUG GRT-0001] net_M3:0 -> net_M3:0 via 1-1 r=0.017
[DEBUG GRT-0001] net_M3:1 -> net_M3:1 via 2-2 r=0.017
[DEBUG GRT-0001] net_M3:2 -> net_M3:3 79.92u layer=3 r=2.625 c=11.761
[DEBUG GRT-0001] net_M3:4 -> net_M3:4 via 2-2 r=0.017
[DEBUG GRT-0001] net_M3:5 -> net_M3:5 via 1-1 r=0.017
[DEBUG GRT-0001] net_M3:4 -> inv_M3_2/A (14.60, 89.67) 0.49u layer=2 r=0.015 via_res=0.017 c=0.072
[DEBUG GRT-0001] pin inv_M3_2/A -> to grid 0u layer=2 r=0.015 via_res=0.017 c=0.072
[DEBUG GRT-0001] net_M3:1 -> inv_M3_1/Y (14.67, 9.75) 0.42u layer=2 r=0.012 via_res=0.017 c=0.061
[DEBUG GRT-0001] pin inv_M3_1/Y -> to grid 0u layer=2 r=0.012 via_res=0.017 c=0.061
 Pi model c2=3.06e-17 rpi=29.5 c1=3.06e-17 max_r=29.5
```

The connectivity appears wrong. An example of what is expected (parasitic estimates calculated directly after global routing, ie not using `read_guides`:

```
[DEBUG GRT-0001] net net_M3
(14.85, 9.99)  1 -> (14.85, 9.99)  2 l=0.00
(14.85, 9.99)  2 -> (14.85, 9.99)  3 l=0.00
(14.85, 9.99)  3 -> (14.85, 89.91)  3 l=79.92
(14.85, 89.91)  3 -> (14.85, 89.91)  2 l=0.00
(14.85, 89.91)  2 -> (14.85, 89.91)  1 l=0.00
[DEBUG GRT-0001] net_M3:0 -> net_M3:1 via 2-3 r=0.017
[DEBUG GRT-0001] net_M3:1 -> net_M3:2 79.92u layer=3 r=2.625 c=11.761
[DEBUG GRT-0001] net_M3:2 -> net_M3:3 via 3-2 r=0.017
[DEBUG GRT-0001] net_M3:3 -> inv_M3_2/A (14.60, 89.67) 0.49u layer=2 r=0.015 via_res=0.017 c=0.072
[DEBUG GRT-0001] pin inv_M3_2/A -> to grid 0u layer=2 r=0.015 via_res=0.017 c=0.072
[DEBUG GRT-0001] net_M3:0 -> inv_M3_1/Y (14.67, 9.75) 0.42u layer=2 r=0.012 via_res=0.017 c=0.061
[DEBUG GRT-0001] pin inv_M3_1/Y -> to grid 0u layer=2 r=0.012 via_res=0.017 c=0.061
 Pi model c2=5.56e-15 rpi=2.5e+03 c1=7.02e-15 max_r=2.72e+03
 Pi model c2=5.56e-15 rpi=2.5e+03 c1=7.02e-15 max_r=2.72e+03
```

I'm guessing it has something to do with setting seg.final_layer.

Test case:  [read_guides_issue-2.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9843502/read_guides_issue-2.tar.gz)

eder-matheus, 2022-11-22:
@antonblanchard sorry for the delay on it, but now I think I've fixed this issue. Could you try this branch: https://github.com/The-OpenROAD-Project/OpenROAD/pull/2539?

Note that you'll have to generate the guide file again, since the core of the fix is on how we write them.

closed 2022-11-29 by eder-matheus

antonblanchard, 2022-11-30:
Thanks @eder-matheus this fixes the issue I was hitting.