The-OpenROAD-Project/OpenROAD #2419: `openroad -python` API breaks

donn, 2022-10-26:
**Describe the bug**
There are a number of API breaks when it comes to `openroad -python` as of late: First, `-exit` is now required as otherwise OpenROAD will drop into a command prompt, which was previously not the case (but overall easy to work around); more importantly, `openroad -exit -python -m` to load a Python module no longer works *at all*. This impedes [DFFRAM](https://github.com/AUCOHL/DFFRAM)'s placeram module.

The behavior used to be that whenever the interpreter saw `-python` it appeared to leave processing the rest of the arguments to the Python executable.

**Expected Behavior**
`openroad -exit -python -m` should work.

**Environment**
```
Git commit: f7df40058721c701313c67009df1e25ba41365cf
kernel: Linux 5.15.65-1-MANJARO
os: Manjaro Linux 
cmake version 3.24.1
gcc (GCC) 12.2.0
clang version 14.0.6
```

**To Reproduce**
```sh
mkdir -p my_module
echo "print('hello, world!')" > my_module/__main__.py
openroad -exit -python -m my_module
```

**Logs**
```
OpenROAD v2.0-5406-gf7df40058 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[WARNING ORD-0039] .openroad ignored with -python
Usage: openroad [-help] [-version] [-no_init] [-exit] [-gui] [-threads count|max] [-log file_name] [-metrics file_name] cmd_file
  -help                 show help and exit
  -version              show version and exit
  -no_init              do not read .openroad init file
  -threads count|max    use count threads
  -no_splash            do not show the license splash at startup
  -exit                 exit after reading cmd_file
  -gui                  start in gui mode
  -python               start with python interpreter [limited to db operations]
  -log <file_name>      write a log in <file_name>
  -metrics <file_name>  write metrics in <file_name> in JSON format
  cmd_file              source cmd_file
Python 3.10.6 (main, Aug  3 2022, 17:39:45) [GCC 12.1.1 20220730] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
```

**Screenshots**

N/A

**Additional context**

The new `[WARNING ORD-0039] .openroad ignored with -python` is thoughtful, but mostly irrelevant to my use-cases. Is there a way to suppress it?

maliberty, 2022-10-26:
@proppy would you take a look?  It seems related to your changes.

rovinski, 2022-10-26:
I wonder if this was a supported feature or unintended consequence. It obviously seems useful.

I think a question arises about whether `-python` is supposed to be an "ordered" or "unordered" flag (I can't find proper taxonomy for this name convention). Is `-python` _supposed_ to change the parsing of all arguments after it, or is that unintentional? Do we expect `openroad -exit -python` to work but `openroad -python -exit` to not work?

I think either:
1. Passing args after `-python` needs to be made an intended behavior and documented / put into CI
2. `-python` needs to be an "unordered" flag, in which case we need an additional argument such as `-python_args "STRING"` to enable explicit passing of arguments to the python shell.

All that aside, does `openroad -exit -python script.py` not load and run `script.py` on the current version?

proppy, 2022-10-27:
> There are a number of API breaks when it comes to openroad -python as of late: First, -exit is now required as otherwise OpenROAD will drop into a command prompt, which was previously not the case (but overall easy to work around)

Yes spotted this too, see https://github.com/The-OpenROAD-Project/OpenROAD/issues/2287 and https://github.com/The-OpenROAD-Project/OpenROAD/pull/2288 as an attempt to fix it.
 
> more importantly, openroad -exit -python -m to load a Python module no longer works at all. This impedes [DFFRAM](https://github.com/AUCOHL/DFFRAM)'s placeram module.

does calling `placeram.py` as a script workaround the issue?

> I wonder if this was a supported feature or unintended consequence. It obviously seems useful.

My understanding from the chatter on #2287 was that it was never quite documented/supported. If that's a behavior the project cares about we should make sure it's captured by non-regression tests.

>  Is -python supposed to change the parsing of all arguments after it, or is that unintentional?

Based on @donn report, it seems that the expected behavior was that:
```
whenever the interpreter saw -python it appeared to leave processing the rest of the arguments to the Python executable.
```

> Passing args after -python needs to be made an intended behavior and documented / put into CI

+1

donn, 2022-10-28:
@proppy 
> does calling placeram.py as a script workaround the issue?

Nope. PlaceRAM uses relative imports, so a module is required. I guess another workaround would be to create a standalone file that imports the module but that is… jank

proppy, 2022-10-31:
Looking at this it seems that we can easily restore the previous behavior by simply calling `Py_Main` with the original `argc` and `argv` but that would effectively means that we ignore the `-exit` flag, since `Py_Main` will exit the interpreter and effectively loose its context.

So before "fixing" this regression I think we need to agree on which usecases we want to support:
A. users should be able to call `openroad -python <args>` with the same arguments as the regular python interpreter and expect a similar behavior (no explicit `-exit` flag needed)
B. users should be able to call `openroad -python script.py` and drop into an interactive REPL on a python interpreter context after `script.py` execution.
C. users should be able to call `openroad -exit -python script.py A B C` and have `script.py` be called with `sys.argv == ['script.py', 'A', 'B', 'C']`

If all are desirable we need to do more fancy argument processing and handle the case where remaining arguments include a valid python script or generic extra arguments.

maliberty, 2022-10-31:
If the user doesn't give -exit then a REPL should remain open.  (A) doesn't make that clear so I'm not sure what "similar behavior" means.

proppy, 2022-10-31:
>  I'm not sure what "similar behavior" means.

Similar behavior as the regular python command line interpreter (same flags, same behavior wrt to exit criteria and status).

> If the user doesn't give -exit then a REPL should remain open.

Even if a file parameter is specified? If that's the case do we care about the interpreter context being the same for the file execution, and the interactive REPL?

maliberty, 2022-10-31:
It happens all the time that I want to run a script and leave the tool open to inspect the results.  How would I do that with your proposal where it exits without -exit being specified.

proppy, 2022-10-31:
> It happens all the time that I want to run a script and leave the tool open to inspect the results. 

Then we need find a way to satisfy what @donn is after (forwarding the flag verbatim to the python interpreter) and what you're after (sharing the interpreter state between a script execution and an interactive session) without changing (too much?) the cli calling convention. 
PS: I'm curious how the cli managed to accomodate both use cases before the refactor in #1812 (https://github.com/The-OpenROAD-Project/OpenROAD/commit/47e8b5f57e2f680172f99c0433c01ed91fcf7fe7).

> How would I do that with your proposal where it exits without -exit being specified.

If we forward the argument verbatim to `Py_Main` this can work by passing `openroad -python -i script.py` or doing `PYTHONINSPECT=1 openroad -python script.py`, see https://docs.python.org/3/using/cmdline.html#cmdoption-i. As it would make `Py_Main` drop into an interactive loop: see https://github.com/python/cpython/blob/main/Modules/main.c#L544.

Bonus point: this could potentially clear up some of the confusing semantic around the ordering of the `-python` and `-exit` flag pointed by @rovinski in https://github.com/The-OpenROAD-Project/OpenROAD/issues/2419#issuecomment-1292641530, as we can simply say: "everything after `-python` is the same as https://docs.python.org/3/using/cmdline.html#command-line"

Since the code path that @donn's care about for executing a module https://github.com/python/cpython/blob/main/Modules/main.c#L604 is only reachable thru the public API's `Py_Main` function, the only alternative I can think of that would also preserve the `-exit` behavior that @maliberty cares about, would be map the `-exit` flag to the `inspect` bit of the interpreter config https://docs.python.org/3/c-api/init_config.html#c.PyConfig.inspect.

maliberty, 2022-10-31:
Bash handles this scenario with -- 
```
       --        A  --  signals the end of options and disables further option
                 processing.  Any arguments after the -- are treated as  file-
                 names and arguments.  An argument of - is equivalent to --.
```
So we could have

openroad -python -exit -- -m module

The other alternative is to say everything after -python is passed to Py_Main directly.  Either is ok so long as it is clearly documented.

proppy, 2022-10-31:
> The other alternative is to say everything after -python is passed to Py_Main directly. Either is ok so long as it is clearly documented.

But would you prefer it to be implemented with the current `-exit` behavior? or rely on the built-in handling of `-i` and `PYTHONINSPECT` by `Py_Main`?

maliberty, 2022-10-31:
I don't think it makes sense to have both -i & -exit.  Switching from -exit to -i would be nice for compatibility with python but would be a pain in all other regards.  

I also don't understand the python decision that -i implies "The PYTHONSTARTUP file is not read."  That potentially makes -i change the results which I find undesirable.  We separate -no_init from -exit which seems better to me.

In short I prefer -exit unless you have strong feelings.

PR #2479 (2022-11-11): src/Main: simplify python flag handling
redo of #2438

- set interpreter config according to command line flags
- forward arguments to python interpreter
- only init python when flag is passed (remove signal save/restore)

Fixes #2419

# Tests

I'd love to add something to the test suite, but https://wiki.tcl-lang.org/page/Expect doesn't seems to be supported by the `test/regression` (and would be needed in order to test proper interaction with the REPL).

```
OpenROAD 🍡 cat foo.py 
import odb
print(odb)
import sys
a = sys.argv
print(a)
```

```
OpenROAD 🍧 build/src/openroad -python foo.py 
Op [... cut]
closed 2022-11-11 by maliberty