The-OpenROAD-Project/OpenROAD #2432: Get GCD antenna fixing to work in sky130hd in a flow.tcl based unit test // It works in hs but not hd.

tspyrou, 2022-10-28:
Its best to work on these one at a time.

tspyrou, 2022-10-28:
@jjcherry56 fyi

eder-matheus, 2022-10-29:
@tspyrou @jjcherry56 I've worked with gcd this week, and I think I found the problem. Here's the scenario:

* Before `global_route` and `repair_antennas` commands, we run `pin_access` to use the results in grt.
* The `repair_antennas` command is appropriately detecting the violations based on the global routing results and inserting the diodes as expected.
* After `repair_antennas`, the detailed router is not being updated with the new APs of the moved cells, nor with the APs for the inserted diodes.
* Running `pin_access` again after `repair_antennas` fixes part of the issue: the moved cells are updated. However, the detailed router data structures are not updated with the diodes (the standard cells and the connection with the nets).

Example: In the figure below, the ANTENNA_5 cell is connected to the net highlighted in yellow. But the detailed router isn't seeing this connection. In summary, I'm trying to make drt more incremental so it's internal structures are updated when we have these db updates.
[image]

jjcherry56, 2022-10-29:
I suspect drt is severely lacking when it comes to incremental changes and
it will be very error prone to try and add incrementality. Is there an API
to remove the drt data structures (which stupidly copy the entire db) and
just start from scratch as a workaround/short term solution?

eder-matheus, 2022-10-31:
I've added a clear function that runs before pin_access, which is now working. I'll update my table with the comparison of antenna violations, and if the numbers look good, I'll create a PR.

eder-matheus, 2022-10-31:
@jjcherry56 @tspyrou Here is the updated table after the fix on drt:
[image]

- We can see a large reduction in the number of DRT antenna violations now, but it is still far from optimal. I'm investigating some designs to understand why there are still violations in there
  - In the case of gcd_sky130hd, the difference between the PAR value with the allowed ratio is pretty small, and it makes sense that the GRT routing doesn't trigger the violation in this pin. Below is the detailed violation: 
[image]
- Other designs like aes_sky130hd have violations with a large difference between the calculated PAR and the allowed ratio, like this one:
[image]

For this specific net, I see that only one diode was inserted, but it is far from the pin with the violation. I wonder if this is the only cause for the violation still there.
[image]

I'll keep investigating the reminiscent violations.

jjcherry56, 2022-11-01:
I am not an antenna expert but I don't believe the proximity of the diode to the load is an issue. The diode just provides a path to the substrate if too much charge accumulates on the wire during fabrication.

I would focus on aes because it is the smallest design that has significant residual violations.
I am going guess that the problem is "almost" violating nets that grt/drt differentces push the ratio over the edge.
That would argue for repairing within a margin like repair_design/repair_timing commands.

eder-matheus, 2022-11-29:
@jjcherry56 I've added a margin parameter to the `repair_antenna`, but it doesn't solve all the violations. It only solves a few of them. Differences between the global route and the detailed route regarding the layer assignment of the wires cause the remaining violations. For example:

In the net below, drt has an antenna violation on the long `met3` wire.
[image]

The violation is not detected on the grt because it has a break on this long wire near the violated pin, going to the layer above.
[image]

But in the final drt, we don't have this layer change to `met4`, so the violation exists only there.
[image]

Not sure what are the next steps on it. I think we could update drt to follow the guides for some nets, but this could affect DRCs and other metrics.

maliberty, 2022-11-29:
I think we'll need a post-drt repair as well.

eder-matheus, 2022-11-29:
> I think we'll need a post-drt repair as well.

We could insert diodes post-drt and reroute the modified nets. But I believe we'll need support for incremental drt. AFAIK we can't route a set of nets if other nets are already routed.

maliberty, 2022-11-29:
We will have to deal with incremental drt.  Any post-drt optimization will need it too.

QuantamHD, 2022-11-29:
@eder-matheus are you running your tests on the latest Master branch? I made some fixes related to antenna fixing that might affect this test

eder-matheus, 2022-11-29:
> @eder-matheus are you running your tests on the latest Master branch? I made some fixes related to antenna fixing that might affect this test

@QuantamHD Yes, I'm using the latest master branch, though I don't think they affect directly my tests with sky130 designs.

vijayank88, 2023-06-22:
@eder-matheus 
I've ran gcd design with latest ORFS commit. Not find any antenna violations.
```
==========================================================================
check_antennas
--------------------------------------------------------------------------
[WARNING ANT-0011] -report_violating_nets is deprecated.
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO ANT-0002] Found 0 net violations.
[INFO ANT-0001] Found 0 pin violations.
```
In OpenLane based flow :
```
[INFO GRT-0014] Routed nets: 412
[INFO GRT-0006] Repairing antennas, iteration 1.
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0012] Found 3 antenna violations.
[INFO GRT-0015] Inserted 4 diodes.
[INFO GRT-0009] rerouting 279 nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 12
[INFO GRT-0006] Repairing antennas, iteration 2.
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0012] Found 0 antenna violations.
```
Which unit test it's referring?

closed 2024-12-18 by eder-matheus