The-OpenROAD-Project/OpenROAD #2464: Unable to verify the installation of OpenRoad using clone and build method

tingho0823, 2022-11-03:
Hi ,
I am using Ubuntu 22.04 in WSL (Windows Subsystem for Linux). I am trying to verify my installation of OpenROAD using clone and build and getting error by following the website.
[https://openroad.readthedocs.io/en/latest/user/BuildLocally.html](url)

The error showed that the command for openroad -help is not found.

[image]

Thanks for helping.

rovinski, 2022-11-03:
Does `OpenROAD-flow-scripts/tools/install/OpenROAD/bin` exist? If not, then the binary was not installed properly.
Or if you have Docker installed, it might be inside the Docker container.

vvbandeira, 2022-11-04:
@tingho0823 if the binary does not exists as Austin mentioned, please upload the following:
```
./etc/DependencyInstaller.sh -dev |& tee dependency.log
./etc/Env.sh |& tee env.log
./build_openroad.sh --local |& tee build.log
```

tingho0823, 2022-11-04:
Hi @vvbandeira , the  OpenROAD-flow-scripts/tools/install/OpenROAD/bin  does not exist like Austi mentioned.Here is my dependency.log ,env.log, build.log
[build.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/9939864/build.log)
[env.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/9939866/env.log)

[dependency.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/9939865/dependency.log)

tingho0823, 2022-11-04:
I found out that i only can run the `etc/DependencyInstaller.sh` in `\OpenROAD-flow-scripts/tools/OpenROAD`. here is the dependency.log

[dependency.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/9940366/dependency.log)

tingho0823, 2022-11-05:
I try to run `./build_openroad.sh` again but it failed at 98%
[image]

rovinski, 2022-11-05:
Your log shows several
```
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
```
Is it possible you are running out of memory? By default the build script uses all available cores to build in parallel. You can reduce the number of parallel processes with `./build_openroad.sh --threads N` where `N` is the number of threads. If you have e.g. an 8-core machine you might try 4 threads instead.

tingho0823, 2022-11-05:
Thanks @rovinski  and @vvbandeira for helping! Much appreciated. I am able to use OpenRoad now.

closed 2022-11-05 by tingho0823