The-OpenROAD-Project/OpenROAD #2470: Feature: Allow compressed files (`.gz`) for various file types.

hpretl, 2022-11-05:
**Describe the bug**

This is no bug but a feature request.

Since more and more PDKs and PDK variants get available disk space becomes an issue, also for large designs.  According to the OpenROAD documentation (Nov 04, 2022) some filetypes are already supported:

> If CMake finds libz, OpenSTA can read Verilog, SDF, SPF, and SPEF files compressed with gzip.

**Expected Behavior**

Allow `gzip` compression on all potentially large file types, as `.lib`, `.lef`, `.def`, `.gds`, etc., across the whole OpenROAD toolchain.

**Environment**

n/a

**To Reproduce**

n/a

**Logs**

n/a

**Screenshots**

n/a

**Additional context**

n/a

maliberty, 2022-11-14:
We already support compress .lib but could do so for other files.  Note that gds isn't read by OR but by KLayout (or Magic in OL).

maliberty, 2022-11-14:
gzip lef & def are also already supported.  Is there anything to do here?

hpretl, 2022-11-25:
I ran a quick test with OpenLane and an error was reported, but most likely this due to the OL scripts. So, can we say that OpenROAD is ready, but OpenLane needs to use this functionality?

maliberty, 2022-11-25:
I believe so.  What error do you see in OL?

maliberty, 2024-09-10:
Nothing here to address as it seems to be an OL issue.

closed 2024-09-10 by maliberty