The-OpenROAD-Project/OpenROAD #2471: GDS file is missing aftre make Design_Cofig

tingho0823, 2022-11-06:
Hi ,
I am using Ubuntu 22.04 in WSL (Windows Subsystem for Linux).I build OpenRoad locally.
Now  i try to  `make DESIGN_CONFIG=./designs/sky130hd/ibex/config.mk` but the final GDS file is missing. 
I suspect that I installed **KLayout** in wrong path. Below show the output after  I `make Design_CONFIG` and the path of KLayout installed.

[image]

tingho0823, 2022-11-06:
**Latest Update**
I try to reinstall  Klayout in  `/OpenROAD-flow-scripts` and run `make DESIGN_CONFIG=./designs/sky130hd/ibex/config.mk` and the output is shown i got signal number 11 error

[image]

rovinski, 2022-11-06:
Please use the issue template so that we can help you most efficiently.
Please attach your flow log.

vijayank88, 2022-11-07:
@tingho0823 
Make sure using latest commits from master.
Can you check where klayout installed `which klayout`

Do run the flow again and check:
```
source setup_env.sh
cd flow
make clean_all
make
```

maliberty, 2022-12-15:
No activity; please reopen if still relevant.

closed 2022-12-15 by maliberty