The-OpenROAD-Project/OpenROAD #2482: `extract_parasitics` crash

kareefardi, 2022-11-09:
**Describe the bug**

crash after running `extract_parasitics`.

**Expected Behavior**

proper exit with an error message or a warning.

**Environment**

```
Git commit: ff442dd5425fdd179418de62187f52620d0939a9
kernel: Linux 5.14.0-1054-oem
os: Ubuntu 20.04.3 LTS (Focal Fossa)
cmake version 3.14.0
gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
clang version 10.0.0-4ubuntu1
```

**To Reproduce**

attached is a test case with the relevant files (.def, .lef, extraction rules file and flow.tcl)
[test-case.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/9970650/test-case.tar.gz)

**Logs**

stack trace:
```
 0# 0x000056441E5467B8 in openroad
 1# 0x00007F311DB350C0 in /lib/x86_64-linux-gnu/libc.so.6
 2# odb::dbWireEncoder::initPath(odb::dbTechLayer*, unsigned char, odb::dbTechLayerRule*) in openroad
 3# odb::dbWireEncoder::newPathShort(int, odb::dbTechLayer*, odb::dbWireType, odb::dbTechLayerRule*) in openroad
 4# odb::tmg_conn::addToWire(int, int, int, bool, bool) in openroad
 5# odb::tmg_conn::treeReorder(bool, bool) in openroad
 6# odb::tmg_conn::analyzeNet(odb::dbNet*, bool, bool, bool, int, int, bool) in openroad
 7# odb::orderWires(odb::dbBlock*, bool, bool, int, int) in openroad
 8# rcx::Ext::extract(rcx::Ext::ExtractOptions) in openroad
 9# extract(char const*, int, double, float, int, int, int, char const*, bool, bool) in openroad
10# 0x000056441EC48913 in openroad
11# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
12# 0x00007F311F822924 in /lib/x86_64-linux-gnu/libtcl8.6.so
13# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
14# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
15# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
16# ord::tclAppInit(Tcl_Interp*) in openroad
17# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
18# main in openroad
19# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
20# _start in openroad
```

errors in the log
```
Error 0: Internal Error: tmg_conn::detachTilePins: tilepin inside iterm. should not happen.
Error 0: Internal Error: tmg_conn::detachTilePins: tilepin inside iterm. should not happen.
Error 0: Internal Error: tmg_conn::detachTilePins: tilepin inside iterm. should not happen.
Error 0: Internal Error: tmg_conn::detachTilePins: tilepin inside iterm. should not happen.
Error 0: Internal Error: tmg_conn::detachTilePins: tilepin inside iterm. should not happen.
Error 0: Internal Error: tmg_conn::detachTilePins: tilepin inside iterm. should not happen.
Notice 0: error, net 13, shorts to another term
Notice 0: wire point (2061355 210895)
Notice 0:  flash_io0_ieb_core
```
It is not totally clear if errors like the one above are the cause of the crash. Also, from the screenshot bellow I don't understand what is the problem with `flash_io0_ieb_core`

**Screenshots**

[image]

**Additional context**

maliberty, 2022-11-09:
I see you have NDRs like:

  - met2_width_280
     + LAYER met2 WIDTH 280 ;

which is invalid.  In the DEF manual it states "Every nondefault rule must specify a width for every layer."  This is violated in your design.  How were these NDRs generated?

kareefardi, 2022-11-09:
They were generated by magic and I believe that they could be fixed. Is that the cause of the crash?

maliberty, 2022-11-09:
Yes.  We can make the DEF reader reject your input but that's about all.

RTimothyEdwards, 2022-11-09:
@maliberty :  NDR errors are all mine, as I implemented the generation and parsing of non-default rules in magic fairly recently.  I read through the DEF spec but I did not see such a rule requiring all layers in each non-default rule (although, like pretty much the entire spec, the language is ambiguous. . . That could be read to mean only that every LAYER statement must have an associated WIDTH, which is how I would read the spec).

maliberty, 2022-11-09:
@RTimothyEdwards its not a good spec.  I am basing it on the LAYER statement in NONDEFAULTRULES
[image]

and also the second sentence in
[image]

RTimothyEdwards, 2022-11-09:
@maliberty :  The first quote matches my understanding and implementation of the nondefault rule.  The second sentence appears to refer only to vias since it's in the middle of a paragraph about vias, and (in my reading) indicates that every via layer must have a specified width.

maliberty, 2022-11-09:
I don't believe it only applies to vias.

kareefardi, 2022-11-10:
An updated def ([chip_io.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9983211/chip_io.zip)) doesn't crash and produces a spef ([chip_io.nom.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/9983222/chip_io.nom.zip))
However the resultant spef has a problem where nets have empty `*CONN` section. such as line `12143`
```
*D_NET *699 0.0160395
*CONN
*CAP
1 *699:117 6.98441e-06
```
I am not entirely sure if these are due to these warnings during extraction such as these.
```
[WARNING RCX-0114] Net 699 clock_pad__OE_N does not start from an iterm or a bterm.
```
I also don't fully understand what unique about nets like `clock_pad__OE_N` to produce such a warning

maliberty, 2022-11-10:
The shapes in this net are not connected to the pin shapes.  In short you still have a bad DEF:
[image]

You can see pin shapes at the top and left that are highlighted but have no shapes reaching them.

RTimothyEdwards, 2022-11-10:
@maliberty :  Can you please give me a coordinate where the net and pin are disconnected in the above screenshot?

maliberty, 2022-11-10:
@RTimothyEdwards  There isn't a single connected pin so pick any from the DEF provided

RTimothyEdwards, 2022-11-10:
@maliberty :  Okay, I see it now.  I seem to have broken something related to wire extensions while trying to fix something else.

tspyrou, 2022-11-14:
It looks like this is not an openroad issue.

kareefardi, 2022-11-14:
The cause of the crash isn't an openroad issue. The crash itself is an openroad issue

maliberty, 2022-11-14:
We can add an error in DEF parsing.

PR #2492 (2022-11-14): odb: error if NDR doesn't contain all layers
Fixes #2482
commit 1050b8e2 (2022-11-14): odb: error if NDR doesn't contain all layers

Fixes #2482
closed 2022-11-14 by maliberty