The-OpenROAD-Project/OpenROAD #2488: Grammar railroad diagram

mingodad, 2022-11-12:
Using this tool https://www.bottlecaps.de/convert/ to convert https://github.com/The-OpenROAD-Project/OpenROAD/blob/429a272dae6e384dcdc7ebc99bdc4ecb0f66f8a1/src/odb/src/lef/lef/lef.y and https://raw.githubusercontent.com/The-OpenROAD-Project/OpenROAD/7f985d5e0a07ebb08ccff8c4453abec95d747a26/src/odb/src/def/def/def.y into an EBNF understood by https://www.bottlecaps.de/rr/ui and manually replacing the tokens (search & replace `K_*` by `/*K_*/*`) we can have a navigable railroad diagram.

Copy and paste one of the `EBNF` shown bellow on https://www.bottlecaps.de/rr/ui on the tab `Edit Grammar` then click the tab `View Diagram`:
```
/* converted on Sat Nov 12, 2022, 11:19 (UTC+01) by bison-to-w3c v0.61 which is Copyright (c) 2011-2022 by Gunther Rademacher <grd@gmx.net> */
//From https://github.com/The-OpenROAD-Project/OpenROAD/blob/429a272dae6e384dcdc7ebc99bdc4ecb0f66f8a1/src/odb/src/lef/lef/lef.y
lef_file ::= error? rule* extension_opt end_library
version  ::= /*K_*/"VERSION" T_STRING ';'
int_number
         ::= NUMBER
dividerchar
         ::= /*K_*/"DIVIDERCHAR" QSTRING ';'
busbitchars
         ::= /*K_*/"BUSBITCHARS" QSTRING ';'
end_library
         ::= ( /*K_*/"END" /*K_*/"LIBRARY" )?
rule     ::= version
           | busbitchars
           | case_sensitivity
           | units_section
           | layer_rule
           | via
           | viarule
           | viarule_generate
           | dividerchar
           | wireextension
           | msg_statement
           | spacing_rule
           | dielectric
           | minfeature
           | irdrop
           | site
           | macro
           | array
           | def_statement
           | nondefault_rule
           | prop_def_section
           | universalnoisemargin
           | edgeratethreshold1
           | edgeratescalefactor
           | edgeratethreshold2
           | noisetable
           | correctiontable
           | input_antenna
           | output_antenna
           | inout_antenna
           | antenna_input
           | antenna_inout
           | antenna_output
           | manufacturing
           | fixedmask
           | useminspacing
           | clearancemeasure
           | maxstack_via
           | create_file_statement
case_sensitivity
         ::= /*K_*/"NAMESCASESENSITIVE" ( /*K_*/"ON" | /*K_*/"OFF" ) ';'
wireextension
         ::= /*K_*/"NOWIREEXTENSIONATPIN" ( /*K_*/"ON" | /*K_*/"OFF" ) ';'
fixedmask
         ::= /*K_*/"FIXEDMASK" ';'
manufacturing
         ::= /*K_*/"MANUFACTURINGGRID" int_number ';'
useminspacing
         ::= /*K_*/"USEMINSPACING" spacing_type spacing_value ';'
clearancemeasure
         ::= /*K_*/"CLEARANCEMEASURE" clearance_type ';'
clearance_type
         ::= /*K_*/"MAXXY"
           | /*K_*/"EUCLIDEAN"
[... 1283 lines cut ...]
         ::= nondefault_start nondefault_def+ nondefault_end
nondefault_start
         ::= /*K_*/"NONDEFAULTRULES" NUMBER ';'
nondefault_end
         ::= /*K_*/"END" /*K_*/"NONDEFAULTRULES"
nondefault_def
         ::= '-' T_STRING nondefault_option* ';'
nondefault_option
         ::= '+' ( /*K_*/"HARDSPACING" | /*K_*/"LAYER" T_STRING /*K_*/"WIDTH" NUMBER nondefault_layer_option* | ( /*K_*/"VIA" | /*K_*/"VIARULE" ) T_STRING | /*K_*/"MINCUTS" T_STRING NUMBER )
           | nondefault_prop_opt
nondefault_layer_option
         ::= ( /*K_*/"DIAGWIDTH" | /*K_*/"SPACING" | /*K_*/"WIREEXT" ) NUMBER
nondefault_prop_opt
         ::= '+' /*K_*/"PROPERTY" nondefault_prop*
nondefault_prop
         ::= T_STRING ( NUMBER | QSTRING | T_STRING )
styles_section
         ::= styles_start styles_rule* styles_end
styles_start
         ::= /*K_*/"STYLES" NUMBER ';'
styles_end
         ::= /*K_*/"END" /*K_*/"STYLES"
styles_rule
         ::= '-' /*K_*/"STYLE" NUMBER firstPt nextPt otherPts ';'
```

maliberty, 2022-11-12:
That's clever.  What do you propose we do with it?

There are places where it would be nice to inline the rules in the diagram, e.g.
[image]
[image]
[image]

mingodad, 2022-11-12:
I hope it'll help document the language and also debug/develop the grammar, you can download the railroad diagram and add to the documentation.

mingodad, 2022-11-12:
Also notice that the railroad diagram is navigable so we can click on `[spacing_type]` to jump to it's definition, if the definition is not singular it always appear separated.

maliberty, 2022-11-12:
If you compare the diagrams on https://www.json.org/json-en.html they are more readable.  In general the [reference manual](https://www.ispd.cc/contests/18/lefdefref.pdf) gives a fuller definition and description.

maliberty, 2026-03-12:
Its yours

sparsh-karna, 2026-03-12:
Hi @maliberty, great question.

The two tools are available on Maven Central:
- `de.bottlecaps.ebnf-convert:ebnf-convert-lib:0.73`
- `de.bottlecaps.rr:rr-lib:2.6`

They could potentially be managed via `rules_jvm_external` and 
invoked through a `genrule` in BUILD. Happy to implement this 
if that's the preferred direction, just waiting on your and 
@hzeller @QuantamHD's input before going down that path.

In the meantime, here's a preview of the generated diagrams 
from the current Python pipeline (docs are still in a rough 
state, structure will be improved before the PR):

https://github.com/sparsh-karna/OpenROAD/blob/odb/grammar-railroad-diagrams/src/odb/doc/DEF_Grammar.md
https://github.com/sparsh-karna/OpenROAD/tree/odb/grammar-railroad-diagrams/src/odb/doc/images

sparsh-karna, 2026-03-13:
Fixed, added a neutral background to the SVGs so the 
diagram lines are readable regardless of theme. 
Updated diagrams are pushed to the branch:
https://github.com/sparsh-karna/OpenROAD/tree/odb/grammar-railroad-diagrams/src/odb/doc/images