The-OpenROAD-Project/OpenROAD #2497: PDNGEN Array Spacing Violation

QuantamHD, 2022-11-15:
It looks like PDNGen may not be complying with a particular array spacing rule in GF180MCU.

https://github.com/efabless/caravel_user_project/issues/185

Specifically https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu7t5v0/blob/d3798681284ea9b96bfd57bfda20948ac86767e4/tech/gf180mcu_5LM_1TM_9K_7t_tech.lef#L111

maliberty, 2022-11-15:
@arlpetergadfort thoughts?

arlpetergadfort, 2022-11-15:
@maliberty @QuantamHD looks real, not sure why its missing this (it might be honoring the LEF85 ones and missing the non-LEF58, with a testcase I would be able to replicate and fix the issue.

maliberty, 2022-11-15:
@arlpetergadfort this is related to gf180 caravel and I've asked for a test case already from efabless

maliberty, 2022-11-15:
@kareefardi please attached the test case to this issue.

arlpetergadfort, 2022-11-15:
@maliberty Looking around the LEF parser, I cannot see where the array spacing data is stored? There is clearly handling for it in the lef parser itself, but I can't see where it gets added to ODB.

QuantamHD, 2022-11-15:
@arlpetergadfort It looks like it's set in odb here.   

https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/odb/src/lefin/ArraySpacingParser.cpp#L64

arlpetergadfort, 2022-11-15:
@QuantamHD those are only the LEF58_ARRAYSPACING rules as far as I can tell. (it's only triggered on the LEF58_ARRAYSPACING keyword).

arlpetergadfort, 2022-11-15:
This is where that keyword is added to the lefiLayer, but that data is never access again: https://github.com/The-OpenROAD-Project/OpenROAD/blob/c653563d31c25f39cc23f84c3d487aea7ca30136/src/odb/src/lef/lef/lefiLayer.cpp#L3750

maliberty, 2022-11-15:
It looks like we don't store it currently for the base rule but do for the property.  That just means somebody has to add the  code to query the lef parser and build up the odb structure.  @arlpetergadfort is this something you want to handle or do you prefer I do it?

QuantamHD, 2022-11-16:
@proppy Could you provide a test case?

kareefardi, 2022-11-16:
here is a test case [issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/10020940/issue_reproducible.tar.gz)

arlpetergadfort, 2022-11-16:
@kareefardi how was this issue generated? The ODB does not have the array spacing rules in it, which leads me to think it was not run with the head of OpenROAD or now run from the beginning. Could you please update OpenROAD and rerun from the start to ensure the tech lef if read in with the new version.

kareefardi, 2022-11-16:
Yes it is using an older version of openroad. I will generate a new one based on the current head.

kareefardi, 2022-11-16:
Here is another reproducible. this time there is a change in vias inserted looks like vias are using 0.36um spacing. [issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/10026155/issue_reproducible.tar.gz)

maliberty, 2022-11-16:
@kareefardi is the 0.36 acceptable (ie is this resolved)?

QuantamHD, 2022-11-17:
@kareefardi Is there still a DRC violation?

kareefardi, 2022-11-17:
It is DRC clean now

closed 2022-11-17 by maliberty