The-OpenROAD-Project/OpenROAD #2503: Question: Resquesting clarification and possibly new feature on the ability to support multi via requirements on clock and "special" data nets

azwefabless, 2022-11-15:
I have been unable to determine if dual (or triple) via can be implemented in SKY130 for clock and other designated special nets as the default connectivity instead of single via.  The purpose is 3 fold: 

1) Reduce clock net timing variation across corners by reducing the portion of the clock impedance that is consumed by via by either 2x or 4x.  This reduces the probability of hold time violations.

2) Reduce the variability of low impedance data nets with high capacitive load across corners. (Compiled memory word line drivers as an example)

3)Increase the current carrying capability of via connections in very high drive and high capacitive load nets like resetb or porb

Is there a flow for this that works today? Thanks in advance , A

maliberty, 2022-11-15:
Do you mean a double cut via or two different vias?  I'm guessing the former.

We have some support for NDRs but it hasn't undergone a lot of usage so far.  If you want to try out there is a create_ndr command but I would expect some bumps.

azwefabless, 2022-11-15:
Yes the former is what I'm looking for but for only a specific subset of routes operating using NDR. 
As an example. Route Key analog nets group 1 using NDR1 (For example a fully shielded input reference voltage) , Clock group 2 using NDR2, Reset and POR using NDR3 and rest of nets using Default rules. I know this sounds like a lot but is consistent with my prior use of commercial tools and has significant advantages for functionality. Other use would be to declare WL nets in a compiled memory and use the tool to increase vias on WL driver to reduce X corner variation in the memory. (Metal resistance changes 10s of % but VIA changes >3X min to max on many layers.

maliberty, 2022-11-15:
I agree that NDRs can bring big advantages which is why we have worked on them.  The person working on NDRs left the project which is why the work stopped progressing.  I would be glad to pick it up again if you want to help test it out.

azwefabless, 2022-11-15:
I might not get to test it myself Mat but I am sure that somebody at efabless can test it and we do want the functionality so I'll commit to that and get you a name to add to this ticket in a day or two.

azwefabless, 2022-11-30:
Person to work with Kareem and he is being added to ticket. Could I also request permission to assign in the future?

maliberty, 2024-09-03:
Stale

closed 2024-09-03 by maliberty