The-OpenROAD-Project/OpenROAD #2514: build_openroad --local --latest fails with cmake error the 2nd time around

stefanottili, 2022-11-17:
### Describe the bug

I'm building OpenRoad-flow-scripts with build_openroad --local --latest on a M1 MacBook Air.

On Nov 15 I had to "brew install or-tools" to be able to build. 
A fresh build after build_openroad --clean succeeded.

Todays build_openroad --local --latest failed with a CMake Error, missing cbc, see log below.

### Expected Behavior

be able to run build_openroad --local --latest incrementally without having to --clean first.

### Environment

```shell
Git commit: 63206cb894bbb07a2e388cdfda31a46095bf737a
kernel: Darwin 21.6.0
os: 	macOS 	12.6.1
cmake version 3.24.3
Apple clang version 13.1.6 (clang-1316.0.21.2.5)
```

### To Reproduce

./build_openroad.sh --local --latest on MacBook twice, fails the 2nd time

### Relevant log output

```shell
CMake Error at /opt/homebrew/lib/cmake/ortools/ortoolsTargets.cmake:61 (set_target_properties):
  The link interface of target "ortools::ortools" contains:

    PkgConfig::Cbc

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /opt/homebrew/lib/cmake/ortools/ortoolsConfig.cmake:82 (include)
  src/mpl2/CMakeLists.txt:42 (find_package)
```

vvbandeira, 2022-11-20:
> On Nov 15 I had to "brew install or-tools" to be able to build.

We are installing or-tools by downloading the upstream tar distribution, see [here](https://github.com/The-OpenROAD-Project/OpenROAD/blob/174739b179897aaf160580061053a213e5a5b07c/etc/DependencyInstaller.sh#L118-L128). This will pin the version to something we know is working for our application. We are also assuming the location of `or-tools` to be where our installer sets, see [here](https://github.com/The-OpenROAD-Project/OpenROAD/blob/174739b179897aaf160580061053a213e5a5b07c/src/CMakeLists.txt#L36-L37) and [here](https://github.com/The-OpenROAD-Project/OpenROAD/blob/174739b179897aaf160580061053a213e5a5b07c/src/mpl2/CMakeLists.txt#L36-L40).

You can try to workaround by exporting the following variables:
```
export CMAKE_INSTALL_RPATH="/opt/homebrew/or-tools/lib"
export CMAKE_PREFIX_PATH="/opt/homebrew/or-tools/lib/cmake"
```

Or using the `DependencyInstaller.sh` script to install in the OpenROAD's default path.

Note: there is no guarantee that OpenROAD will always work with the latest `or-tools`, I would suggest to run `brew pin or-tools` to avoid updating by accident.

stefanottili, 2022-11-20:
a) the DependencyInstaller.sh want's to install or-tools in /opt but I don't have root access to this machine
b) openroad compile after a build_openroad.sh --clean cleaning OpenRoad or the first time after a clone
So why would it fail the cmake configure the 2nd time around ?
c) the issue is not that cmake wouldn't find or-tools, the log indicates that or-tools cmake has an issue finding cbc, even though both are installed via homebrew.

I'm aware that Mac and howebrew are not primary build targets.
I also have to admit zero knowledge of how cmake works, so any hint on how to debug cmake and get this to work with or-tools and cbc from homebrew is welcome.

stefanottili, 2022-11-20:
I saw the "cmake-or-tools" commit merge but now OpenRoad fails in a different way before it get's to or-tools.

./build_openroad.sh --local --latest
[INFO FLW-0002] Updating git submodules.
Submodule path 'tools/OpenROAD': checked out 'd95c9111c39beea98c34ef9089f722057cdca58c'
[INFO FLW-0004] Updating OpenROAD app to the HEAD  of origin/master.
Previous HEAD position was d95c9111c Merge pull request #2477 from The-OpenROAD-Project-staging/odb-lefbloat
HEAD is now at 655c839d2 Merge pull request #2527 from vvbandeira/cmake-or-tools
From https://github.com/The-OpenROAD-Project/OpenROAD
 * branch                master     -> FETCH_HEAD
...
-- GUI is enabled
-- Found Boost: /opt/homebrew/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization 
CMake Error at /opt/homebrew/Cellar/cmake/3.25.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find OpenMP_CXX (missing: OpenMP_CXX_FLAGS OpenMP_CXX_LIB_NAMES)
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.25.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.25.0/share/cmake/Modules/FindOpenMP.cmake:580 (find_package_handle_standard_args)
  src/drt/CMakeLists.txt:48 (find_package)
-- Configuring incomplete, errors occurred!

maliberty, 2022-11-20:
Nothing has changed with regard to openmp

stefanottili, 2022-11-21:
I don't know why it's failing with FindOpenMP after a fresh pull today.

I've been compiling OR for month with only 'export LEMON_DIR'  pointing to the lonely dependency that wasn't using homebrew packages.

I would hope that, if there is a need to pin a brew or-tools dependency, this will be done in the same was as it's currently done for spdr and fmt.

Please don't introduce a dependency that requires root access when there is a brew package available.

maliberty, 2022-11-21:
I don't know what "Please don't introduce a dependency that requires root access when there is a brew package available." means.  I'm not a mac users but it is just a ordinary package with no special requirements.

rovinski, 2022-11-21:
@stefanottili tangential, but you should be able to use:
```
brew install The-OpenROAD-Project/lemon-graph/lemon-graph
```
We host a homebrew formula for this reason.

vvbandeira, 2022-11-22:
> I saw the "cmake-or-tools" commit merge but now OpenRoad fails in a different way before it get's to or-tools.
> 
> ./build_openroad.sh --local --latest [INFO FLW-0002] Updating git submodules. Submodule path 'tools/OpenROAD': checked out 'd95c9111c39beea98c34ef9089f722057cdca58c' [INFO FLW-0004] Updating OpenROAD app to the HEAD of origin/master. Previous HEAD position was [d95c911](https://github.com/The-OpenROAD-Project/OpenROAD/commit/d95c9111c39beea98c34ef9089f722057cdca58c) Merge pull request #2477 from The-OpenROAD-Project-staging/odb-lefbloat HEAD is now at [655c839](https://github.com/The-OpenROAD-Project/OpenROAD/commit/655c839d206bae189124392364553baad7e665f2) Merge pull request #2527 from vvbandeira/cmake-or-tools From https://github.com/The-OpenROAD-Project/OpenROAD
> 
> * branch                master     -> FETCH_HEAD
>   ...
>   -- GUI is enabled
>   -- Found Boost: /opt/homebrew/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization
>   CMake Error at /opt/homebrew/Cellar/cmake/3.25.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
>   Could NOT find OpenMP_CXX (missing: OpenMP_CXX_FLAGS OpenMP_CXX_LIB_NAMES)
>   Call Stack (most recent call first):
>   /opt/homebrew/Cellar/cmake/3.25.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
>   /opt/homebrew/Cellar/cmake/3.25.0/share/cmake/Modules/FindOpenMP.cmake:580 (find_package_handle_standard_args)
>   src/drt/CMakeLists.txt:48 (find_package)
>   -- Configuring incomplete, errors occurred!

@stefanottili did you update your `xcode` since you first posted the issue? could you add here the output of:
```
cd tools/OpenROAD
./etc/Env.sh
```

stefanottili, 2022-11-23:
MacBook Air M1, macOS Monterey 12.6.1

a) Command Line Tools are 13.1.6 because 14.0 didn't compile abc, didn't try with 14.1 (yet)

b) ./tools/OpenROAD/etc/DependencyInstaller.sh requires root privileges to install or-tools in /opt/or-tools.
There is a homebrew or-tools package, why not use that ?
Every other package requirement is using homebrew too.
spdr and fmt are pinned to specific versions.

c) I've removed a local installation of lemon and used
brew install The-OpenROAD-Project/lemon-graph/lemon-graph

d) Nov 22 build attempt

OpenRoad-flow-scripts % git pull
Already up to date.
build_openroad.sh --local --latest

... 
builds yosys
...
[INFO FLW-0018] Compiling OpenROAD.
-- The CXX compiler identification is AppleClang 13.1.6.13160021
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-5810-g40f3313c1
-- System name: Darwin
-- Compiler: AppleClang 13.1.6.13160021
-- Build type: RELEASE
-- Install prefix: /Users/stefanthiede/OpenROAD-flow-scripts/tools/install/OpenROAD
-- The C compiler identification is AppleClang 13.1.6.13160021
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
...
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- TCL library: /opt/homebrew/opt/tcl-tk/lib/libtcl8.6.dylib
-- TCL header: /opt/homebrew/opt/tcl-tk/include/tcl.h
-- Found SWIG: /opt/homebrew/bin/swig (found suitable version "4.1.0", minimum required is "3.0")  
CMake Warning at src/CMakeLists.txt:162 (message):
  Using SWIG >= 4.1.0 -flatstaticmethod flag for python
-- Found Boost: /opt/homebrew/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- boost: 1.80.0
-- Found Python3: /opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/include/python3.11 (found version "3.11.0") found components: Development Development.Module Development.Embed 
-- Found ZLIB: /Library/Developer/CommandLineTools/SDKs/MacOSX12.3.sdk/usr/lib/libz.tbd (found version "1.2.11") 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- spdlog: 1.9.2
CMake Warning at src/CMakeLists.txt:224 (message):
  spdlog: SPDLOG_FMT_EXTERNAL=ON
-- Found BISON: /opt/homebrew/opt/bison/bin/bison (found version "3.8.2") 
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- STA version: 2.3.2
-- STA git sha: 895ab26f6c1b83671a32ff2678b4f6f866f4221e
-- System name: Darwin
-- Compiler: AppleClang 13.1.6.13160021
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- Install prefix: /Users/stefanthiede/OpenROAD-flow-scripts/tools/install/OpenROAD
-- Found FLEX: /opt/homebrew/opt/flex/bin/flex (found version "2.6.4") 
-- TCL library: /opt/homebrew/opt/tcl-tk/lib/libtcl8.6.dylib
-- TCL header: /opt/homebrew/opt/tcl-tk/include/tcl.h
-- SSTA: 0
-- STA executable: /Users/stefanthiede/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/sta
-- GPU is not enabled
-- Found Protobuf: /opt/homebrew/lib/libprotobuf.dylib (found version "3.21.9") 
-- Found re2: /opt/homebrew/lib/cmake/re2/re2Config.cmake (found version "9.0.0") 
-- Found Eigen3: /opt/homebrew/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- GUI is enabled
-- Found Boost: /opt/homebrew/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization 
CMake Error at /opt/homebrew/Cellar/cmake/3.25.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find OpenMP_CXX (missing: OpenMP_CXX_FLAGS OpenMP_CXX_LIB_NAMES)
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.25.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.25.0/share/cmake/Modules/FindOpenMP.cmake:580 (find_package_handle_standard_args)
  src/drt/CMakeLists.txt:48 (find_package)
-- Configuring incomplete, errors occurred!
See also "/Users/stefanthiede/OpenROAD-flow-scripts/tools/OpenROAD/build/CMakeFiles/CMakeOutput.log".
See also "/Users/stefanthiede/OpenROAD-flow-scripts/tools/OpenROAD/build/CMakeFiles/CMakeError.log".

e) env 

cd tools/OpenRoad
OpenROAD % ./etc/Env.sh 
Git commit: 40f3313c1c6b44144938de07064b940d08fd5e06
kernel: Darwin 21.6.0
os: 	macOS 	12.6.1
cmake version 3.25.0
Apple clang version 13.1.6 (clang-1316.0.21.2.5)

stefanottili, 2022-11-23:
./build_openroad.sh --local --latest
[INFO FLW-0002] Updating git submodules.
Submodule path 'tools/OpenROAD': checked out 'd95c9111c39beea98c34ef9089f722057cdca58c'
[INFO FLW-0004] Updating OpenROAD app to the HEAD  of origin/master.
Previous HEAD position was d95c9111c Merge pull request #2477 from The-OpenROAD-Project-staging/odb-lefbloat
HEAD is now at 40f3313c1 Merge pull request #2538 from The-OpenROAD-Project-staging/mpl2_opt
From https://github.com/The-OpenROAD-Project/OpenROAD
 * branch                master     -> FETCH_HEAD
Already up to date.
[INFO FLW-0001] Using local build method. This will create binaries at 'tools/install' unless overwritten.
[INFO FLW-0017] Compiling Yosys.
$BREW_PREFIX is [/opt/homebrew/opt]
mkdir -p /Users/stefanthiede/OpenROAD-flow-scripts/tools/install/yosys/bin
cp yosys yosys-config yosys-abc yosys-filterlib yosys-smtbmc /Users/stefanthiede/OpenROAD-flow-scripts/tools/install/yosys/bin
strip -S /Users/stefanthiede/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys
strip /Users/stefanthiede/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-abc
strip /Users/stefanthiede/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-filterlib
mkdir -p /Users/stefanthiede/OpenROAD-flow-scripts/tools/install/yosys/share/yosys
cp -r share/. /Users/stefanthiede/OpenROAD-flow-scripts/tools/install/yosys/share/yosys/.
[INFO FLW-0018] Compiling OpenROAD.
-- OpenROAD version: v2.0-5810-g40f3313c1
-- System name: Darwin
-- Compiler: AppleClang 13.1.6.13160021
-- Build type: RELEASE
-- Install prefix: /Users/stefanthiede/OpenROAD-flow-scripts/tools/install/OpenROAD
-- TCL library: /opt/homebrew/opt/tcl-tk/lib/libtcl8.6.dylib
-- TCL header: /opt/homebrew/opt/tcl-tk/include/tcl.h
CMake Warning at src/CMakeLists.txt:162 (message):
  Using SWIG >= 4.1.0 -flatstaticmethod flag for python

-- Found Boost: /opt/homebrew/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- boost: 1.80.0
-- spdlog: 1.9.2
CMake Warning at src/CMakeLists.txt:224 (message):
  spdlog: SPDLOG_FMT_EXTERNAL=ON

-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- STA version: 2.3.2
-- STA git sha: 895ab26f6c1b83671a32ff2678b4f6f866f4221e
-- System name: Darwin
-- Compiler: AppleClang 13.1.6.13160021
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- Install prefix: /Users/stefanthiede/OpenROAD-flow-scripts/tools/install/OpenROAD
-- TCL library: /opt/homebrew/opt/tcl-tk/lib/libtcl8.6.dylib
-- TCL header: /opt/homebrew/opt/tcl-tk/include/tcl.h
-- SSTA: 0
-- STA executable: /Users/stefanthiede/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/sta
-- GPU is not enabled
-- GUI is enabled
-- Found Boost: /opt/homebrew/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization 
CMake Error at /opt/homebrew/Cellar/cmake/3.25.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find OpenMP_CXX (missing: OpenMP_CXX_FLAGS OpenMP_CXX_LIB_NAMES)
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.25.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.25.0/share/cmake/Modules/FindOpenMP.cmake:580 (find_package_handle_standard_args)
  src/drt/CMakeLists.txt:48 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/stefanthiede/OpenROAD-flow-scripts/tools/OpenROAD/build/CMakeFiles/CMakeOutput.log".
See also "/Users/stefanthiede/OpenROAD-flow-scripts/tools/OpenROAD/build/CMakeFiles/CMakeError.log".

vvbandeira, 2022-11-23:
@stefanottili I had the same problem on my mac and was able to fix with:
```
brew link --force libomp
```

stefanottili, 2022-11-24:
Yay, many thanks !
After another round of ./build_openroad --clean, this did the trick.
Back to being able to compile OpenRoad on Mac M1.
Now this only needs to be reflected in DependencyInstaller.sh, right ?

stefanottili, 2022-11-25:
well, today's ./build_openroad --local --latest craps out with the same cmake

CMake Error at /opt/homebrew/lib/cmake/ortools/ortoolsTargets.cmake:61 (set_target_properties):
  The link interface of target "ortools::ortools" contains:
    PkgConfig::Cbc
  but the target was not found.  Possible reasons include:

brew link --force libomp doesn't help.
I have to ./build_openroad --clean and blow the OpenRoad build directory away in order to be able to compile it.

vvbandeira, 2022-11-28:
@stefanottili do you have any changes to the source code? I am not able to reproduce your issue. Are you using any env like conda, venv or similar?

stefanottili, 2022-11-28:
This still happens when using
./build_openroad.sh --local --latest
or a completely fresh git clone ... 

The only other variable I can think of is what brew packages are installed.
I'll consider blowing all 3.5GB of brew packages away and reinstalling the dependencies.

But current ./tools/OpenROAD/etc/DependencyInstaller.sh want's to install or-tools in /opt
which requires root access, whereas I have it installed via brew.

==> or-tools: stable 9.4 (bottled), HEAD
Google's Operations Research tools
https://developers.google.com/optimization/
/opt/homebrew/Cellar/or-tools/9.4 (457 files, 20.4MB) *
  Poured from bottle on 2022-11-20 at 09:21:32
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/or-tools.rb

118 _installOrTools() {
119     os=$1
120     version=$2
121     arch=$3
122     orToolsVersionBig=9.4
123     orToolsVersionSmall=${orToolsVersionBig}.1874
124     orToolsFile=or-tools_${arch}_${os}-${version}_cpp_v${orToolsVersionSmall}.tar.gz
125     wget https://github.com/google/or-tools/releases/download/v${orToolsVersionBig}/${orToolsFile}
126     mkdir -p /opt/or-tools
127     tar --strip 1 --dir /opt/or-tools -xf ${orToolsFile}
128 }

.zshrc

export PATH=$PATH:/Users/user/OpenROAD-flow-scripts/tools/install/OpenROAD/bin

export PATH=/opt/homebrew/opt/bison/bin:$PATH
export PATH=/opt/homebrew/opt/flex/bin:$PATH
export PATH=/opt/homebrew/opt/ruby@3.1/bin:$PATH
export PATH=/opt/homebrew/opt/qt@5/bin:$PATH
export PATH=/opt/homebrew/bin:$PATH
export PATH=.:$PATH

vvbandeira, 2022-11-29:
@stefanottili I was able to reproduce your problem, looks like the version installed from `brew` is different from the one the installer script installs. Could you try to install using the installer script? If you do not want to use `sudo`, you can try to use another path on lines 126 and 127.

`[A]` is the brew install

```
[A]=/usr/local/opt/or-tools
[B]=/opt/or-tools
Usage:   <Enter>/'o'=open,'s'=sync,'q'=quit
Options: 'u'=update,'x'=set excludes,'i'=set ignore,'a'=set args, 'h'=hex mode, 'w'=wrap mode
Diff Args: -r --brief

    Files [A]/bin/fzn-ortools and [B]/bin/fzn-ortools differ
==> Files [A]/include/ortools/bop/bop_parameters.pb.h and [B]/include/ortools/bop/bop_parameters.pb.h differ
    Files [A]/include/ortools/constraint_solver/assignment.pb.h and [B]/include/ortools/constraint_solver/assignment.pb.h differ
    Files [A]/include/ortools/constraint_solver/demon_profiler.pb.h and [B]/include/ortools/constraint_solver/demon_profiler.pb.h differ
    Files [A]/include/ortools/constraint_solver/routing_enums.pb.h and [B]/include/ortools/constraint_solver/routing_enums.pb.h differ
    Files [A]/include/ortools/constraint_solver/routing_parameters.pb.h and [B]/include/ortools/constraint_solver/routing_parameters.pb.h differ
    Files [A]/include/ortools/constraint_solver/search_limit.pb.h and [B]/include/ortools/constraint_solver/search_limit.pb.h differ
    Files [A]/include/ortools/constraint_solver/search_stats.pb.h and [B]/include/ortools/constraint_solver/search_stats.pb.h differ
    Files [A]/include/ortools/constraint_solver/solver_parameters.pb.h and [B]/include/ortools/constraint_solver/solver_parameters.pb.h differ
    Files [A]/include/ortools/glop/parameters.pb.h and [B]/include/ortools/glop/parameters.pb.h differ
    Files [A]/include/ortools/graph/flow_problem.pb.h and [B]/include/ortools/graph/flow_problem.pb.h differ
    Files [A]/include/ortools/linear_solver/linear_solver.pb.h and [B]/include/ortools/linear_solver/linear_solver.pb.h differ
    Files [A]/include/ortools/packing/multiple_dimensions_bin_packing.pb.h and [B]/include/ortools/packing/multiple_dimensions_bin_packing.pb.h differ
    Files [A]/include/ortools/packing/vector_bin_packing.pb.h and [B]/include/ortools/packing/vector_bin_packing.pb.h differ
    Files [A]/include/ortools/pdlp/solve_log.pb.h and [B]/include/ortools/pdlp/solve_log.pb.h differ
    Files [A]/include/ortools/pdlp/solvers.pb.h and [B]/include/ortools/pdlp/solvers.pb.h differ
    Files [A]/include/ortools/sat/boolean_problem.pb.h and [B]/include/ortools/sat/boolean_problem.pb.h differ
    Files [A]/include/ortools/sat/cp_model.pb.h and [B]/include/ortools/sat/cp_model.pb.h differ
    Files [A]/include/ortools/sat/cp_model_service.pb.h and [B]/include/ortools/sat/cp_model_service.pb.h differ
    Files [A]/include/ortools/sat/sat_parameters.pb.h and [B]/include/ortools/sat/sat_parameters.pb.h differ
    Files [A]/include/ortools/scheduling/course_scheduling.pb.h and [B]/include/ortools/scheduling/course_scheduling.pb.h differ
    Files [A]/include/ortools/scheduling/jobshop_scheduling.pb.h and [B]/include/ortools/scheduling/jobshop_scheduling.pb.h differ
    Files [A]/include/ortools/scheduling/rcpsp.pb.h and [B]/include/ortools/scheduling/rcpsp.pb.h differ
    Files [A]/include/ortools/util/optional_boolean.pb.h and [B]/include/ortools/util/optional_boolean.pb.h differ
    Files [A]/lib/cmake/ortools/ortoolsConfig.cmake and [B]/lib/cmake/ortools/ortoolsConfig.cmake differ
    Files [A]/lib/cmake/ortools/ortoolsConfigVersion.cmake and [B]/lib/cmake/ortools/ortoolsConfigVersion.cmake differ
    Files [A]/lib/cmake/ortools/ortoolsTargets-release.cmake and [B]/lib/cmake/ortools/ortoolsTargets-release.cmake differ
    Files [A]/lib/cmake/ortools/ortoolsTargets.cmake and [B]/lib/cmake/ortools/ortoolsTargets.cmake differ
    Files [A]/lib/libortools.9.4.dylib and [B]/lib/libortools.9.4.dylib differ
    Files [A]/lib/libortools.9.dylib and [B]/lib/libortools.9.dylib differ
    Files [A]/lib/libortools.dylib and [B]/lib/libortools.dylib differ
    Files [A]/lib/libortools_flatzinc.9.4.dylib and [B]/lib/libortools_flatzinc.9.4.dylib differ
    Files [A]/lib/libortools_flatzinc.9.dylib and [B]/lib/libortools_flatzinc.9.dylib differ
    Files [A]/lib/libortools_flatzinc.dylib and [B]/lib/libortools_flatzinc.dylib differ
    Files [A]/README.md and [B]/README.md differ
    Files [A]/share/minizinc/solvers/ortools.msc and [B]/share/minizinc/solvers/ortools.msc differ
    Only in [A]/share: or-tools
    Only in [A]: .brew
    Only in [A]: INSTALL_RECEIPT.json
    Only in [A]: LICENSE
    Only in [B]/bin: protoc
    Only in [B]/bin: protoc-3.19.4.0
    Only in [B]/bin: scip
    Only in [B]/include/ortools/algorithms: csharp
    Only in [B]/include/ortools/algorithms: java
    Only in [B]/include/ortools/algorithms: python
    Only in [B]/include/ortools/algorithms: samples
    Only in [B]/include/ortools/constraint_solver: csharp
    Only in [B]/include/ortools/constraint_solver: docs
    Only in [B]/include/ortools/constraint_solver: python
    Only in [B]/include/ortools/constraint_solver: samples
    Only in [B]/include/ortools/flatzinc: mznlib
    Only in [B]/include/ortools/glop: samples
    Only in [B]/include/ortools/graph: csharp
    Only in [B]/include/ortools/graph: java
    Only in [B]/include/ortools/graph: python
    Only in [B]/include/ortools/graph: samples
    Only in [B]/include/ortools/gscip: gscip.pb.h
[... cut ...]
    Only in [B]/lib: libCoinUtils.a
    Only in [B]/lib: libOsi.a
    Only in [B]/lib: libOsiCbc.a
    Only in [B]/lib: libOsiClp.a
    Only in [B]/lib: libprotobuf-lite.a
    Only in [B]/lib: libprotobuf.a
    Only in [B]/lib: libprotoc.a
    Only in [B]/lib: libre2.a
    Only in [B]/lib: libscip.a
    Only in [B]/lib: libz.a
    Only in [B]/lib: pkgconfig
    Only in [B]/share: eigen3
    Only in [B]/share: man
    Only in [B]/share: pkgconfig
    Only in [B]: examples
    Only in [B]: Makefile
```

stefanottili, 2022-11-29:
Thanks for trying to get to the bottom of this.

I've changed tools/OpenROAD/etc/DependencyInstaller.sh 133,134 or-tools location to /opt/homebrew/or-tools,
ran the installer, looks good, files untarred.
```
./build_openroad.sh --local --latest
[INFO FLW-0002] Updating git submodules.
error: Your local changes to the following files would be overwritten by checkout:
	etc/DependencyInstaller.sh
Please commit your changes or stash them before you switch branches.
Aborting
fatal: Unable to checkout '40f3313c1c6b44144938de07064b940d08fd5e06' in submodule path 'tools/OpenROAD'
```
reversed the changes to be able to proceed and now cmake finds the ortools Config.cmake but fails with
```
./build_openroad.sh --local --latest
....
CMake Error at src/mpl2/CMakeLists.txt:42 (find_package):
  Found package configuration file:

    /opt/homebrew/or-tools/lib/cmake/ortools/ortoolsConfig.cmake

  but it set ortools_FOUND to FALSE so package "ortools" is considered to be
  NOT FOUND.  Reason given by package:

  The following imported targets are referenced, but are missing:
  Coin::CbcSolver Coin::OsiCbc Coin::ClpSolver Coin::OsiClp
```
As a cmake novice, I dunno why it would find cmake/ortools/ortoolsConfig.cmake the rest of what it needs.

stefanottili, 2022-11-29:
Is there a reason why the version specific or-tools are not installed the same way on MacOS as lemon-graph is ?

stefanottili, 2022-12-01:
I still get
```
The following imported targets are referenced, but are missing:
  Coin::CbcSolver Coin::OsiCbc Coin::ClpSolver Coin::OsiClp
```

stefanottili, 2022-12-09:
using a stand alone build of the or-tools and 
export CMAKE_PREFIX_PATH="/Users/user/or-tools_cpp_MacOsX-12.5_v9.4.1874"
gets OR to compile on Mac M1 again.

closed 2022-12-09 by stefanottili