The-OpenROAD-Project/OpenROAD #2516: unable to repair antennas

kareefardi, 2022-11-17:
### Describe the bug

openroad reports antenna violations but it is unable to repair them

### Expected Behavior

repair antenna violations

### OpenROAD Environment

```shell
Git commit: 7f00621cb612fd94e15b35790afe744c89d433a7
kernel: Linux 5.14.0-1054-oem
os: Ubuntu 20.04.3 LTS (Focal Fossa)
cmake version 3.14.0
gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
clang version 10.0.0-4ubuntu1
```

### OpenLane Environment

```shell
Kernel: Linux v5.14.0-1054-oem
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.11 (OK)
OpenLane Git Version: 1da2eec3d9b5a3612cb026cf9060cb7954e4c24a
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

1da2eec3 2022-11-16T01:37:27+02:00 [BOT] Update netgen (#1452) - Openlane Bot -  (HEAD -> master, tag: 2022.11.16, origin/master, origin/HEAD)
77fd531e 2022-11-15T13:32:13+02:00 Revert #1485 (#1492) - Mohamed Gaber -  ()
dbb24870 2022-11-15T13:05:26+02:00 [BOT] Update PDK (#1455) - Openlane Bot -  ()
---
Git Remotes

donn	git@github.com:donn/openlane.git (fetch)
donn	git@github.com:donn/openlane.git (push)
fork	git@github.com:kareefardi/OpenLane.git (fetch)
fork	git@github.com:kareefardi/OpenLane.git (push)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (push)
```

### To Reproduce

`./run.sh` or `./run.tcl` [issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/10032235/issue_reproducible.tar.gz)

### Relevant log output

```shell
[INFO GRT-0018] Total wirelength: 1570186 um
[INFO GRT-0014] Routed nets: 12882
[INFO GRT-0006] Repairing antennas, iteration 1.
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0012] Found 115 antenna violations.
[WARNING GRT-0243] Unable to repair antennas on net with diodes.
[INFO GRT-0015] Inserted 0 diodes.
[INFO GRT-0009] rerouting 9219 nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 15
[INFO GRT-0006] Repairing antennas, iteration 2.
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0012] Found 128 antenna violations.
[WARNING GRT-0243] Unable to repair antennas on net with diodes.
[INFO GRT-0015] Inserted 0 diodes.
[INFO GRT-0009] rerouting 39 nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 11
[INFO GRT-0006] Repairing antennas, iteration 3.
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0012] Found 132 antenna violations.
[WARNING GRT-0243] Unable to repair antennas on net with diodes.
[INFO GRT-0015] Inserted 0 diodes.
```

### Additional Context

this is for `gf180mcuC`

QuantamHD, 2022-11-18:
Apparently the issue is related to one of the antenna fixing strategies of breaking up a long wire by moving down a layer. This  apparently isn't working and sometimes the router does the wrong thing by going up a layer instead of down. This is my vague understanding from talking to Andy at Efabless

azwefabless, 2022-11-18:
Request from Google: Could Efablesss provide a little more feedback on the actual issue observed as a failure to fix. What is it that happens to the antenna violating nets that causes them not to be fixed?

kareefardi, 2022-11-18:
This is something different. This is about openroad failing not repairing antenna violations at all.

QuantamHD, 2022-11-18:
This looks like the issue comes down to GF180MCU being simpler than we expect.

In SKY130 we have antenna rules that look like this.
```
  ANTENNAMODEL OXIDE1 ;
  ANTENNADIFFSIDEAREARATIO PWL ( ( 0 400 ) ( 0.0125 400 ) ( 0.0225 2609 ) ( 22.5 11600 ) ) ;

```

However, in GF180MCU we have rules that look like.
```
    ANTENNAMODEL OXIDE1 ;
    ANTENNADIFFSIDEAREARATIO 400 ;
    ANTENNAGATEPLUSDIFF 2 ;
```

There is only a single value instead of a linearly interpolated list of values. 

```c++
bool AntennaChecker::antennaRatioDiffDependent(dbTechLayer* layer)
{
  if (layer->hasDefaultAntennaRule()) {
    const dbTechLayerAntennaRule* antenna_rule = layer->getDefaultAntennaRule();
    dbTechLayerAntennaRule::pwl_pair diffPAR = antenna_rule->getDiffPAR();
    dbTechLayerAntennaRule::pwl_pair diffPSR = antenna_rule->getDiffPSR();
    return diffPAR.indices.size() > 1 || diffPSR.indices.size() > 1;
  }
  return false;
}
```

The following code asserts that there must be at least two values for the rule to be diff dependent which is incorrect. When I changed the code to  

```c++
bool AntennaChecker::antennaRatioDiffDependent(dbTechLayer* layer)
{
  if (layer->hasDefaultAntennaRule()) {
    const dbTechLayerAntennaRule* antenna_rule = layer->getDefaultAntennaRule();
    dbTechLayerAntennaRule::pwl_pair diffPAR = antenna_rule->getDiffPAR();
    dbTechLayerAntennaRule::pwl_pair diffPSR = antenna_rule->getDiffPSR();
    return diffPAR.indices.size() >= 1 || diffPSR.indices.size() >= 1;
  }
  return false;
}
```

OpenROAD fixed nearly all of the antennas in this design.

closed 2022-11-19 by maliberty