The-OpenROAD-Project/OpenROAD #2521: OR cannot find libortools.so.9 - (installation problem)

mi-and-core, 2022-11-18:
Describe the bug
openroad does not find libortools.so.9

Expected Behavior
openroad should find libortools.so.9

OpenROAD Environment
openroad v2.0-5770-g9e3721536
os: Ubuntu 20.04

To Reproduce
run openroad

Relevant output
[INFO][FLOW] Using platform directory ./platforms/gf180
/home/regenmacher/Schreibtisch/ORFS/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad: error while loading shared libraries: libortools.so.9: cannot open shared object file: No such file or directory

Additional Context:
OR use the precompiled or-tools, installed in /opt/or-tools
The libortools.so.9 exists in /opt/or-tools/lib,
but is searched in /opt/or-tools/lib64

Solution: rename

mi-and-core, 2022-11-18:
@vijayank88
The problem is that after unpacking or-tools_amd64_ubuntu-20.04_cpp_v9.4.1874 in _installOrTools() ( ./etc/DependencyInstaller.sh) we have the following :
 
/opt/or-tools/bin/
/opt/or-tools/examples/
/opt/or-tools/include/
/opt/or-tools/lib/
/opt/or-tools/share/

but no
/opt/or-tools/lib64/

while  ./src/CMakeLists.txt says :

```
# The location used by etc/DependencyInstaller.sh
list(APPEND CMAKE_INSTALL_RPATH "/opt/or-tools/lib64")
```

so here is a mismatch between /lib and /lib64

note: ./etc/DependencyInstaller.sh untar the binary-release, but dont build googles or-tools from source

vijayank88, 2022-11-18:
@vvbandeira fyi

vijayank88, 2022-11-19:
@mi-and-core 
Please provide list of commands you executed.

mi-and-core, 2022-11-19:
@vijayank88 

### How to reproduce :

git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts.git
cd OpenROAD-flow-scripts
sudo ./tools/openroad/etc/DependencyInstaller -run 
sudo ./tools/openroad/etc/DependencyInstaller -dev 
./build_openroad.sh --nice --local **--latest**

( Note the "--latest" so that a recent commit "ord: add rpath for default or-tools install" is taken into account. Thats the one that adds "list(APPEND CMAKE_INSTALL_RPATH "/opt/or-tools/lib64")" to openroad/src/CMakeLists.txt )

source setup_env.sh
cd flow
make 

Everything went as expected - except for the last make

relevant message: 
OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad: 
error while loading shared libraries: libortools.so.9: 
cannot open shared object file: No such file or directory 

The reason for this is very clear: see my previous post

Proof: 

git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts.git
cd OpenROAD-flow-scripts
sudo ./tools/openroad/etc/DependencyInstaller -run 
sudo ./tools/openroad/etc/DependencyInstaller -dev 
./build_openroad.sh --nice --local  
source setup_env.sh
cd flow
make 

( Note: without --latest so that the mentioned or-commit does not take effect. This "solves" the problem )

rovinski, 2022-11-20:
You need to add `/opt/or-tools/bin` to your PATH. I think there is an open PR somewhere that adds a helper message to the dependency installer script to do so.

maliberty, 2022-11-20:
@rovinski if it is installed in /opt/or-tools then that shouldn't be necessary.

rovinski, 2022-11-20:
Are you sure? Packages in `/opt` are not in the PATH by default. It seems that openroad builds fine because cmake is set up to find it, it's just that during runtime the shared lib is not found.

vvbandeira, 2022-11-20:
@maliberty @rovinski this is a "odd" choice from upstream/or-tools. They have both `lib` and `lib64` for CentOS and openSUSE distributions and only `lib` for Ubuntu and macOS. Trying a fix with #2527

mi-and-core, 2022-11-20:
@rovinski @maliberty 
At present it is not yet necessary to have /opt/or-tools/bin in the PATH, as the or-tools are not yet used. Putting /opt/or-tools/bin in the PATH does not solve the problem.
@vvbandeira 
That solves the "Problem".
btw.: If the or-tools are compiled from source (e.g. see https://github.com/google/or-tools/blob/stable/cmake/README.md), instead of untar the release, then an or-tools/lib64 is also created for ubuntu.

vvbandeira, 2022-11-20:
> btw.: If the or-tools are compiled from source (e.g. see https://github.com/google/or-tools/blob/stable/cmake/README.md), instead of untar the release, then an or-tools/lib64 is also created for ubuntu.

Compiling from source could be an option to add to our installer script, but I rather not since it would add the build time and more important another point of possible failure.

Since the issue can be fixed with the proposed changes I would opt to merge the PR and close this issue. @maliberty @rovinski ?

maliberty, 2022-11-20:
Merged

closed 2022-11-20 by maliberty