The-OpenROAD-Project/OpenROAD #2526: Unable to accept ispd2015 DEF input

DonquixoteMzt, 2022-11-19:
### Describe the bug

Openroad was not able to accept ispd2015 DFF file and report that " error: Non-default rule (DWDS) has no rule for layer via1." However, it is OK in previous version (around 11/07). I find that recent commit add additional checks in  https://github.com/The-OpenROAD-Project/OpenROAD/blob/174739b179897aaf160580061053a213e5a5b07c/src/odb/src/defin/definNonDefaultRule.cpp#L219.

I ignored these codes and it is reported that
......
[INFO ODB-0130]     Created 374 pins.
[INFO ODB-0131]     Created 108292 components and 310036 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 0 connections.
[INFO ODB-0133]     Created 110281 nets and 310036 connections.
[INFO ODB-0134] Finished DEF file: /OpenROAD/data/ispd2015/results/mgc_des_perf_a/design/design.gp.def
[INFO GRT-0020] Min routing layer: metal1
[INFO GRT-0021] Max routing layer: metal5
[INFO GRT-0022] Global adjustment: 0%
[INFO GRT-0023] Grid origin: (0, 0)
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0088] Layer metal1  Track-Pitch = 0.2000  line-2-Via Pitch: 0.2000
[INFO GRT-0088] Layer metal2  Track-Pitch = 0.2000  line-2-Via Pitch: 0.2000
[INFO GRT-0088] Layer metal3  Track-Pitch = 0.2000  line-2-Via Pitch: 0.2000
[INFO GRT-0088] Layer metal4  Track-Pitch = 0.2000  line-2-Via Pitch: 0.2000
[INFO GRT-0088] Layer metal5  Track-Pitch = 0.2000  line-2-Via Pitch: 0.2000
[INFO GRT-0019] Found 0 clock nets.
/usr/include/c++/9/bits/stl_vector.h:1042: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::reference = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Signal 6 received
Stack trace:
 0# 0x0000560A43C696D8 in openroad
 1# 0x00007FC2ACCE7090 in /lib/x86_64-linux-gnu/libc.so.6
 2# gsignal in /lib/x86_64-linux-gnu/libc.so.6
 3# abort in /lib/x86_64-linux-gnu/libc.so.6
 4# 0x0000560A43C69838 in openroad
 5# grt::GlobalRouter::computeTrackConsumption(grt::Net const*, int&, std::vector<int, std::allocator<int> >*&) in openroad
 6# grt::GlobalRouter::makeFastrouteNet(grt::Net*) in openroad
 7# grt::GlobalRouter::initNets(std::vector<grt::Net*, std::allocator<grt::Net*> >&) in openroad
 8# grt::GlobalRouter::initFastRoute(int, int) in openroad
 9# grt::GlobalRouter::globalRoute(bool) in openroad
10# 0x0000560A4443EDFB in openroad
11# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
12# 0x00007FC2AE9D4924 in /lib/x86_64-linux-gnu/libtcl8.6.so
13# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
14# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
15# ord::tclAppInit(Tcl_Interp*) in openroad
16# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
17# main in openroad
18# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
19# _start in openroad
Aborted
 

### Expected Behavior

Accept ispd2015 benchmarks input (https://www.ispd.cc/contests/15/ispd2015_contest.html)

### Environment

```shell
kernel: Linux 5.15.74.2-microsoft-standard-WSL2
os: CentOS Linux 7 (Core)
cmake version 3.24.2
gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-44)
./etc/Env.sh: line 45: clang: command not found
```

### To Reproduce

source gr.tcl
[mgc_des_perf_a.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/10046859/mgc_des_perf_a.zip)

### Relevant log output

```shell
N/A
```

### Screenshots

[image]

### Additional Context

N/A

maliberty, 2022-11-19:
I believe this has already been fixed.  Please update and try again.

QuantamHD, 2022-11-19:
@maliberty so is this a reversal on your previous stance that NDRs need to be defined at every layer?

maliberty, 2022-11-19:
No it is a bug that it requires them on the first via layer - they are required on all routing layers.

DonquixoteMzt, 2022-11-20:
The issue is solved and thanks a lot.

closed 2022-11-20 by DonquixoteMzt

LJK314, 2023-03-25:
I meet the bug when accept ispd2015 for mpl2

"OpenROAD v2.0-7231-gff3b43866 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /home/data/ispd15/mgc_fft_b/tech.lef
[WARNING ODB-0220] WARNING (LEFPARS-2007): NAMESCASESENSITIVE statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later. See file /home/data/ispd15/mgc_fft_b/tech.lef at line 4.

[INFO ODB-0388] unsupported LEF57_MINSTEP property for layer metal1 :"MINSTEP 0.100 MAXEDGES 1 ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer metal1 :"SPACING 0.11 ENDOFLINE 0.12 WITHIN 0.045 PARALLELEDGE 0.11 WITHIN 0.11 ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer via1 :"SPACING 0.13 PARALLELOVERLAP ;"
[INFO ODB-0388] unsupported LEF57_MINSTEP property for layer metal2 :"MINSTEP 0.1 MAXEDGES 1 ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer metal2 :"SPACING 0.12 ENDOFLINE 0.14 WITHIN 0.045 PARALLELEDGE 0.12 WITHIN 0.12 ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer via2 :"SPACING 0.13 PARALLELOVERLAP ;"
[INFO ODB-0388] unsupported LEF57_MINSTEP property for layer metal3 :"MINSTEP 0.1 MAXEDGES 1 ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer metal3 :"SPACING 0.12 ENDOFLINE 0.14 WITHIN 0.045 PARALLELEDGE 0.12 WITHIN 0.12 ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer via3 :"SPACING 0.13 PARALLELOVERLAP ;"
[INFO ODB-0388] unsupported LEF57_MINSTEP property for layer metal4 :"MINSTEP 0.1 MAXEDGES 1 ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer metal4 :"SPACING 0.12 ENDOFLINE 0.14 WITHIN 0.045 PARALLELEDGE 0.12 WITHIN 0.12  ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer via4 :"SPACING 0.13 PARALLELOVERLAP ;"
[INFO ODB-0388] unsupported LEF57_MINSTEP property for layer metal5 :"MINSTEP 0.1 MAXEDGES 1 ;"
[INFO ODB-0388] unsupported LEF57_SPACING property for layer metal5 :"SPACING 0.12 ENDOFLINE 0.14 WITHIN 0.045 PARALLELEDGE 0.12 WITHIN 0.12  ;"
[WARNING ODB-0220] WARNING (LEFPARS-2019): TOPOFSTACKONLY statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later See file /home/data/ispd15/mgc_fft_b/tech.lef at line 472.

[WARNING ODB-0220] WARNING (LEFPARS-2019): TOPOFSTACKONLY statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later See file /home/data/ispd15/mgc_fft_b/tech.lef at line 482.

[WARNING ODB-0220] WARNING (LEFPARS-2019): TOPOFSTACKONLY statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later See file /home/data/ispd15/mgc_fft_b/tech.lef at line 638.

[WARNING ODB-0220] WARNING (LEFPARS-2019): TOPOFSTACKONLY statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later See file /home/data/ispd15/mgc_fft_b/tech.lef at line 648.

[WARNING ODB-0220] WARNING (LEFPARS-2019): TOPOFSTACKONLY statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later See file /home/data/ispd15/mgc_fft_b/tech.lef at line 804.

[WARNING ODB-0220] WARNING (LEFPARS-2019): TOPOFSTACKONLY statement is obsolete in version 5.6 and later.
The LEF parser will ignore this statement.
To avoid this warning in the future, remove this statement from the LEF file with version 5.6 or later See file /home/data/ispd15/mgc_fft_b/tech.lef at line 814.

[INFO ODB-0223]     Created 9 technology layers
[INFO ODB-0224]     Created 64 technology vias
[INFO ODB-0226] Finished LEF file:  /home/data/ispd15/mgc_fft_b/tech.lef
[INFO ODB-0222] Reading LEF file: /home/data/ispd15/mgc_fft_b/cells.lef
[INFO ODB-0225]     Created 16 library cells
[INFO ODB-0226] Finished LEF file:  /home/data/ispd15/mgc_fft_b/cells.lef
[WARNING ORD-1011] LEF master in01f01 has no liberty cell.
[WARNING ORD-1011] LEF master oa12f01 has no liberty cell.
[WARNING ORD-1011] LEF master oa22f01 has no liberty cell.
[WARNING ORD-1011] LEF master ao12f01 has no liberty cell.
[WARNING ORD-1011] LEF master no02f01 has no liberty cell.
[WARNING ORD-1011] LEF master ao22s01 has no liberty cell.
[WARNING ORD-1011] LEF master na03f01 has no liberty cell.
[WARNING ORD-1011] LEF master na02f01 has no liberty cell.
[WARNING ORD-1011] LEF master no03m01 has no liberty cell.
[WARNING ORD-1011] LEF master h0 has no liberty cell.
[WARNING ORD-1011] LEF master h1 has no liberty cell.
[WARNING ORD-1011] LEF master h2 has no liberty cell.
[WARNING ORD-1011] LEF master h3 has no liberty cell.
[WARNING ORD-1011] LEF master h4 has no liberty cell.
[WARNING ORD-1011] LEF master h5 has no liberty cell.
[WARNING ORD-1011] LEF master ms00f80 has no liberty cell.
[... 144 lines cut ...]
[WARNING ODB-0103] error: undefined NONDEFAULTRULE (DWDS) referenced
[... cut ...]
[WARNING ODB-0103] error: undefined NONDEFAULTRULE (DWDS) referenced
[WARNING ODB-0103] error: undefined NONDEFAULTRULE (DWDS) referenced
[WARNING ODB-0103] error: undefined NONDEFAULTRULE (DWDS) referenced
[WARNING ODB-0103] error: undefined NONDEFAULTRULE (DWDS) referenced
[WARNING ODB-0103] error: undefined NONDEFAULTRULE (DWDS) referenced
[WARNING ODB-0103] error: undefined NONDEFAULTRULE (DWDS) referenced
[WARNING ODB-0103] error: undefined NONDEFAULTRULE (DWDS) referenced
[WARNING ODB-0149] DEF parser returns an error!"

maliberty, 2023-03-25:
Please open a new issue and don't tack onto a closed one.

LJK314, 2023-03-25:
Dear Openroad Team,

Thank you for reaching out to me. I appreciate your prompt response and guidance regarding the issue I submitted. I apologize for not following proper protocol in my previous submission.

I have taken your advice and submitted a new issue as instructed. It is labeled as the #3082, and I hope that it addresses everything that needed attention.

Thank you once again for your help, and please let me know if there is anything else I need to do.

Best regards, Jiangkao Li

| |
李江考
|
|
***@***.***
|
签名由网易邮箱大师定制

maliberty, 2023-03-25:
No problem.