The-OpenROAD-Project/OpenROAD #2540: ‘d2’ in namespace ‘boost::geometry::model’ does not name a type

cvic1080, 2022-11-23:
### Describe the bug

Hi, guys
When compiling openroad, i found below issue that 'd2' couldnt be identified by compiler, so i made debugging, and done some trials, then it passed. so i thought we should add below header file for error files(search.h and ruler.h). 
"#include <boost/geometry/geometries/point_xy.hpp>"

ERROR LOG:
/home/centos/OpenROAD/build/src/gui/gui_autogen/UVLADIE3JM/../../../../../src/gui/src/search.h:65:28: error: ‘d2’ in namespace ‘boost::geometry::model’ does not name a type
   using Point = bg::model::d2::point_xy<int, bg::cs::cartesian>;

/home/centos/OpenROAD/src/gui/src/ruler.cpp:62:35: error: ‘d2’ in namespace ‘boost::geometry::model’ does not name a type
   typedef boost::geometry::model::d2::point_xy<int> point_t;

### Expected Behavior

add "#include <boost/geometry/geometries/point_xy.hpp>" for search.h and ruler.h

### Environment

```shell
centos8
```

### To Reproduce

i refer to all steps in guide, so if you want reproduce, just refer to guide to make test.

### Relevant log output

```shell
ERROR LOG:
/home/centos/OpenROAD/build/src/gui/gui_autogen/UVLADIE3JM/../../../../../src/gui/src/search.h:65:28: error: ‘d2’ in namespace ‘boost::geometry::model’ does not name a type
   using Point = bg::model::d2::point_xy<int, bg::cs::cartesian>;

/home/centos/OpenROAD/src/gui/src/ruler.cpp:62:35: error: ‘d2’ in namespace ‘boost::geometry::model’ does not name a type
   typedef boost::geometry::model::d2::point_xy<int> point_t;
```

### Screenshots

[image]

maliberty, 2022-11-23:
What version of boost are you using?

vijayank88, 2022-11-23:
@cvic1080 
Can you confirm dependency packages installed by running below steps?
```
./etc/DependencyInstaller.sh -run
./etc/DependencyInstaller.sh -dev
```

cvic1080, 2022-11-23:
> @cvic1080 Can you confirm dependency packages installed by running below steps?
> 
> ```
> ./etc/DependencyInstaller.sh -run
> ./etc/DependencyInstaller.sh -dev
> ```

my system is centos8, seems it couldnt be supported by this 2 scripts.

cvic1080, 2022-11-23:
> 

boost 1.66, actually, i solved this issue manually, but hope it better supports more version, it will  be convenient to users.

rovinski, 2022-11-23:
Boost 1.66 is very old. I think that openroad is supposed to require 1.78? Or maybe that requirement was removed.

rovinski, 2022-11-23:
I don't think it was ever put into the cmake file, but we had it in the readme, you can see evidence of it in #866 and #1530. As I recall, it was DRT that required it (looks like it was boost >= 1.68).

I think at some point, there was a PR that fixed the dependency on the specific boost version. But we never tested with older boost versions, only newer ones. I think it is worth a shot for @cvic1080 to try building with boost >= 1.68 to see if the issue is resolved.

vvbandeira, 2022-11-28:
> boost 1.66, actually, i solved this issue manually, but hope it better supports more version, it will be convenient to users.

Closing since the user reports that the issue is fixed.

closed 2022-11-28 by vvbandeira

maliberty, 2022-11-28:
@vvbandeira shall we put a minimum boost version in cmake to avoid this?

rovinski, 2022-11-28:
Did we actually get a clear response of what the issue was and what the fix is?

vvbandeira, 2022-11-29:
> @vvbandeira shall we put a minimum boost version in cmake to avoid this?

I think a range would be better. At least from other issues linked here, there was an issue with 1.75 and 1.80 works fine on macOS builds. Let me run some tests and propose a PR.

> Did we actually get a clear response of what the issue was and what the fix is?

@cvic1080 did you update your boost install or modify the code?