The-OpenROAD-Project/OpenROAD #2541: Standard cell placement is not correct which leads to LVS issues

lulu9312, 2022-11-23:
I am working on OpenFASOC' s ldo-gen which has a comparator (cmp1). The cmp1 is placed in seperate voltage domain.Now the issue for one of the setting is that the VDD rail is passing over cmp1's VSS pin and VSS rail is passing over cmp1's VDD pin this then further causes LVS issue.
Here is screenshots for incorrect placement 
Incorrect placement :
[image]

Here is a screenshot of correct placement
[image]

To reproduce the issue:
[detail_place_ldo_sky130hvl_base_2022-11-23_12-44.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/10073446/detail_place_ldo_sky130hvl_base_2022-11-23_12-44.tar.gz)

vijayank88, 2022-11-23:
@lulu9312 
Please follow issue template to fill `OpenROAD` version, OS etc.,

vijayank88, 2022-11-23:
with latest OR LDO placement as follows
[image]

lulu9312, 2022-11-23:
> with latest OR LDO placement as follows [image]

This is incorrect,this is same as the error image above.

rovinski, 2022-11-23:
This is a double-height standard cell legalization issue. The issue could be in either DPL or DPO.

maliberty, 2022-11-23:
OR doesn't support double height cells.  I suggest you manually place it and set its status to firm.

rovinski, 2022-11-23:
It used to support multi-height cells. Was that completely removed? Would it take significant effort to support it again?

lulu9312, 2022-11-23:
> OR doesn't support double height cells. I suggest you manually place it and set its status to firm.

@maliberty Yes. I thought of that but I have kept the status to PLACED as keeping it FIRM causes site aligned failed issue.
This is implemented here https://github.com/idea-fasoc/OpenFASOC/pull/140

rovinski, 2022-11-23:
@lulu9312 then you need to make sure you snap the cell to a cell site when you place it.

lulu9312, 2022-11-23:
@rovinski Sorry , but what do you mean by snap the cell ?

maliberty, 2022-11-23:
@rovinski it used to incorrectly attempt to support multi-height cells only in dpl.  Since that code was not in use and not at all correct it was removed.  Creating proper support is a project that would need non-trivial effort.

When manually placing the cell you must pick a location that aligns with the sites/rows.

rovinski, 2022-11-23:
@lulu9312 There is a cell site grid in each row. You need to manually align your cell to this grid, otherwise fill cells cannot be inserted properly.

lulu9312, 2022-11-24:
@maliberty @rovinski Understood.

maliberty, 2025-08-26:
There have been substantial improvments to mult-height cells since 2022.  Please reopen if still an issue.

closed 2025-08-26 by maliberty