The-OpenROAD-Project/OpenROAD #2547: Support for installing on debian

vinay-keshava, 2022-11-29:
### Describe the bug

Requesting to support the installation for Debian as well, as the script works only for Ubuntu and fails on debian 
Please update the https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/etc/DependencyInstaller.sh 
as there are some dependencies are different from debain and ubuntu and also the installation steps for debian as well.   

### Expected Behavior

.

### Environment

```shell
.
```

### To Reproduce

.

vinay-keshava, 2022-12-01:
https://shenki.github.io/Building-OpenROAD/
i found this script to build and install dependencies but the build failed due to the following error.

CMake Error at src/gpl/CMakeLists.txt:43 (find_package):
  By not providing "FindEigen3.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Eigen3", but
  CMake did not find one.

  Could not find a package configuration file provided by "Eigen3" with any
  of the following names:

    Eigen3Config.cmake
    eigen3-config.cmake

  Add the installation prefix of "Eigen3" to CMAKE_PREFIX_PATH or set
  "Eigen3_DIR" to a directory containing one of the above files.  If "Eigen3"
  provides a separate development package or SDK, be sure it has been
  installed.

rovinski, 2022-12-02:
You need to install [Eigen3](https://eigen.tuxfamily.org). It might be under the package name `libeigen3-dev` or similar.

vinay-keshava, 2022-12-03:
> You need to install [Eigen3](https://eigen.tuxfamily.org). It might be under the package name `libeigen3-dev` or similar.

Thanks that worked

vinay-keshava, 2022-12-03:
-- STA version: 2.3.2
-- STA git sha: b752018e27dc09c545a0226329581a3861cb0944
-- System name: Linux
-- Compiler: GNU 12.2.0
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- Install prefix: /home/vinay/tools/openroad
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- SSTA: 0
-- STA executable: /home/vinay/pkg/OpenRoad/OpenROAD/src/sta/app/sta
-- GPU is not enabled
CMake Error at /usr/share/cmake-3.25/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  Could not find a package configuration file provided by "Protobuf" with any
  of the following names:

    ProtobufConfig.cmake
    protobuf-config.cmake

  Add the installation prefix of "Protobuf" to CMAKE_PREFIX_PATH or set
  "Protobuf_DIR" to a directory containing one of the above files.  If
  "Protobuf" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/ortools/ortoolsConfig.cmake:45 (find_dependency)
  src/mpl2/CMakeLists.txt:42 (find_package)

but  i have libortools-dev and libprotobuf-dev installed on debian

vinay-keshava, 2022-12-05:
@vvbandeira i tried using the above method, got the following errors

1. Package qt5-default is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'qt5-default' has no installation candidate

Is there any other package provided. ?

2. After running cmake or the build script 
got the following error
https://gist.github.com/vinay-keshava/5a4fbaaea7b661e38bcbd4cdc1065845

habibayassin, 2022-12-05:
@vinay-keshava which Debian version are you using?

vinay-keshava, 2022-12-06:
> @vinay-keshava which Debian version are you using?
I am using Debian 12 (bookworm)

habibayassin, 2022-12-06:
@vinay-keshava I did some changes to the installer….can you try it again?

closed 2022-12-07 by vvbandeira

vvbandeira, 2022-12-07:
The issue got closed automatically with the PR merge. Discussion about Debian 12 can continue here; if there's a docker image available we can consider adding support. Otherwise, we would wait to add support as we would not have a way to implement and test the feature.