The-OpenROAD-Project/OpenROAD #2549: DPL: Cells can't be placed in seperate voltage domain

lulu9312, 2022-11-30:
### Describe the bug

I am trying to place some cells in a voltage domain created by using create_voltage_domain command. Earlier detail_placer was able  place cells in region but the region was inside core_area. Now in this case when I try to place the cells outside core_area it gives the error as shown in below screenshot. 

### Expected Behavior

Deatil placer should place the cells

### Environment

```shell
OpenROAD version : OpenROAD v2.0-5170-gd919ca2d4
```

### To Reproduce

[detail_place_ldo_sky130hvl_base_2022-11-30_10-11.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/10119591/detail_place_ldo_sky130hvl_base_2022-11-30_10-11.tar.gz)

### Screenshots

Here's the screenshot of error:
[image]
Here's the floorplan:
[image]

lulu9312, 2022-11-30:
@maliberty Is there a way to specify second core area inside same die area and then try to place cells there.

maliberty, 2022-11-30:
Your test case is incorrectly packaged with full paths
```
[ERROR STA-0511] cannot open '/home/pranav/OpenFASOC/openfasoc/generators/ldo-gen/flow/scripts/detail_place.tcl'.
```

I don't understand what "place the cells outside core_area" means.  The core area defines where cells can be placed so putting them outside is never legal.

There is no meaning to a second core area.  The core area encompasses all placeable area.

lulu9312, 2022-12-01:
@maliberty Understood. Actually I want to seperate the two voltage domains. How can I achieve that ?

maliberty, 2022-12-01:
OR has never had support for multiple voltage domains.  We have work in progress to do that in the future.

lulu9312, 2022-12-01:
@maliberty Got it. Looking forward to it!

maliberty, 2022-12-01:
This is part of the general UPF support work.  If there is interest I'm glad to figure out how to divide up the work.

rovinski, 2022-12-01:
@lulu9312 you might be able to approximate separate voltage domains by putting placement fences around your two regions and using `cutRow` to remove rows between the two regions. This should become easier with multivoltage support, but I don't think there is a timetable on that.

maliberty, 2022-12-01:
@rovinski we don't have placement fences... that's on the UPF support list.

lulu9312, 2022-12-02:
@rovinski @maliberty Is there a way to have seperate rings for the two domains as it can be seen in the above floorplan image??

maliberty, 2022-12-02:
@rovinski it has been talked about a long time but it has never been implemented.   There may be some support in dpl but without gpl it won't help too much if the cells are far off.   I don't know about dpo but doubt it.

@msaligane how do you handle this in FaSOC?

rovinski, 2022-12-02:
I see, that makes things very difficult.