The-OpenROAD-Project/OpenROAD #2562: About odb::dbDatabase::getChips()

dearmrlv, 2022-12-02:
Hi, I noticed that in odb/include/odb/db.h, getChips() is marked as “DEPRECATED”. Can a dbDatabase contain multiple dbChip still?

maliberty, 2022-12-02:
The database supports it but the rest of OR assumes a single chip.  What is motivating your question?

dearmrlv, 2022-12-02:
Thank you for replying.
Here is my motivation: A large chip always contains several parts, and each part can be implemented seperately. Since dbBlock supports only two-levels of hierarchy, if there exists one dbChip in dbDatabase, the parts has to be child of the top block and the child of the top dbBlock must be “flatten”, i.e. without submodules in it. This is not so friendly for each part implementation. So I’m thinking that make each part a single chip may be a good choice.

rovinski, 2022-12-02:
I would recommend against that. The correct solution would likely be a hierarchical `dbBlock` implementation (arbitrary hierarchy depth) but it is a large task to support it and it has been pending for a while. 

The current best flow is to use a hierarchical flow where the block is extracted as a macro using liberty extraction and LEF extraction, then incorporated as a macro into the next level of hierarchy.

dearmrlv, 2022-12-03:
Thanks, I've got it.

dearmrlv, 2022-12-03:
question solved.

closed 2022-12-03 by dearmrlv