The-OpenROAD-Project/OpenROAD #2564: Error in detail placement

vinayrayapati, 2022-12-02:
### Describe the bug

Error in Detail placement
[CRITICAL DPL-0026] legalPt called on fixed cell.

[image]

### Expected Behavior

Succesfull placement

### Environment

```shell
Please pull the latest changes and try again, if problem persists file a github issue with the re-producible test case.
```

### To Reproduce

[Issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/10141299/Issue_reproducible.zip)

vijayank88, 2022-12-02:
The attached test case is not valid. Please provide steps to reproduce.
Also you're using outdated OpenROAD version, update and try.

vijayank88, 2022-12-02:
`Please don't attach screenshot, always copy paste the logs in texts in given template`

As per the log trying to copy `3_5_place_dp.odb` and file not found and flow failing.

vinayrayapati, 2022-12-02:
@vijayank88 I am using OpenROAD in macOS. `3_5_place_dp.odb` file is the output of the detail placement. Since there is an error in the detail placement, the output file is not created and the next step is giving error as no file or directory. The actual error is `[CRITICAL DPL-0026] legalPt called on fixed cell`.

vijayank88, 2023-06-22:
Please re-open an issue with test case, if still there is an issue with latest OpenROAD commits.

closed 2023-06-22 by vijayank88