The-OpenROAD-Project/OpenROAD #2566: saving cts image maps the GUI window

maliberty, 2022-12-03:
### Describe the bug

I tried adding 
```
foreach clock [get_clocks *] {
  set clock_name [get_name $clock]
  gui::save_clocktree_image $::env(REPORTS_DIR)/cts_$clock_name.webp $clock_name
}
```
to flow/scripts/save_images.tcl.  If I run this in non-graphical mode I see the CTS window briefly maps to the screen and then disappears but I get the normal image on a white background.  If I run it with DISPLAY= then I get:
```
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
This plugin does not support raise()
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
```
and the image is saved but it is on a black background:
[image]

### Expected Behavior

It should be possible to save the correct image without needing to map the window or generating any warnings.  The other image save commands don't have this issue.

### Environment

```shell
master branch
```

### To Reproduce

any design (eg gcd/ng45)

gadfort, 2022-12-03:
@maliberty I tried this and I get the `This plugin does not support propagateSizeHints()` which appears to have always been there. But I get a white background on the clocks and no warning about the `raise()`. This was on WSL Ubuntu 20, what are you doing it on?

QuantamHD, 2022-12-03:
I suspect the image includes transparency

maliberty, 2022-12-03:
That's a good guess but the result looks different in Firefox on CentOS depending on whether DISPLAY is set or not.