The-OpenROAD-Project/OpenROAD #2577: grt creates guides where only a single layer of metal is available, causing shorts in drt

antonblanchard, 2022-12-06:
### Describe the bug

I think this is a similar issue to https://github.com/The-OpenROAD-Project/OpenROAD/issues/1518 however in this case it's the top layer of metal (Metal5) and there is an obstruction covering Metal4. This test is on GF180.

drt is finishing with one short, seen just to the right of the middle of this screenshot:

[image]

The macro has an obstruction on Metal4, and grt has created guides for two wires on Metal5:

`_027524_`
[image]

and

`microwatt_0.soc0.processor.execute1_0.divider_0_div_0.d_in_divisor\[22\] `
[image]

It would be a pity not to use all that free space on Metal5, but on the other hand it seems extremely easy to get stuck with unroutable situations in drt.

Test case:  [metal5-grt-drt.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/10161804/metal5-grt-drt.tar.gz)

### Expected Behavior

No DRCs from drt

### Environment

```shell
Git commit: bdb90cc3f51a1a866234817effb6ac129d9bc0c8
os: Fedora Linux 36 (Workstation Edition)
cmake version 3.24.2
gcc (GCC) 12.2.1 20221121 (Red Hat 12.2.1-4)
clang version 14.0.5 (Fedora 14.0.5-1.fc36)
```
```

### To Reproduce

See test case above.

maliberty, 2022-12-06:
The problem isn't the guides but that drt doesn't follow them as it sees a much shorter path.  One of the shorted nets:
[image]
and the other
[image]

Perhaps we should increase the out-of-guide cost when in a planar routing area.  Note that the guides are pretty ugly and suggest major detouring due to congestion.

maliberty, 2022-12-06:
Another option is to remove/cost the wrong way edges higher in the planar regions

antonblanchard, 2022-12-12:
> Perhaps we should increase the out-of-guide cost when in a planar routing area. Note that the guides are pretty ugly and suggest major detouring due to congestion.

I wonder if we should test an increase to `GUIDECOST`. If there is little cost for drt to stay on the grt guide, then drt could drastically change the wire length (as it has above) and create hold violations.

A change to `GUIDECOST` could negatively impact some designs, but I bet in a lot of cases the root cause is the huge derating we often apply to grt (eg 50% by default in ORFS)

closed 2025-08-26 by maliberty