The-OpenROAD-Project/OpenROAD #2580: Error message of : RePlAce divergence detected

david200340105, 2022-12-06:
### Describe the bug

I encountered some problems when I ran the stage of global placement (RePlAce).

When I ran the global placement(GP) with the benchmark of ISPD 2015 contest,
there were two benchmarks that could not process the GP stage smoothly.
( mgc_superblue16_a and mgc_superblue19 )
*benchmark of ISPD 2015 : https://www.ispd.cc/contests/15/web/downloads.html (Hidden benchmark designs )

And I process these two benchmark with the following commands in my tcl file:
read_lef  ${tech.lef}
read_lef ${cells.lef}
read_def  ${floorplan.def}
global_placement -density 1.0

After I processed benchmark of mgc_superblue19, the last 2 rows of error messages are: 
[NesterovSolve] Iter: 680 overflow: 0.18357 HPWL: 14598922301
[ERROR GPL-0307] RePlAce divergence detected. Re-run with a smaller max_phi_cof value.

When I try smaller max_phi_cof value ( 1.1,1.3) with following command:
global_placement -density 1.0 -max_phi_coef 1.03

There still had error message:
[NesterovSolve] Iter: 680 overflow: 0.18357 HPWL: 14598922301
[ERROR GPL-0307] RePlAce divergence detected. Re-run with a smaller max_phi_cof value.

And I further try smaller max_phi_cof
global_placement -density 1.0 -max_phi_coef 1.01

The error will be solved, but the overflow will not decrease less than 0.1 after 5000 iterations of NesterovSolve. The last 2 rows of messages are:
[NesterovSolve] Iter: 4990 overflow: 0.362241 HPWL: 12902711426
[NesterovSolve] Iter: 5000 overflow: 0.362278 HPWL: 12902395788

Another problem is that after I processing mgc_superblue16_a, the error message showed that :
[ERROR GPL-0305] RePlAce diverged at newStepLength.

### Expected Behavior

The overflow should be converged when iteration less than maybe 2000, without showing error message "RePlAce diverged at newStepLength".

### Environment

```shell
git commit: unknown, this is not a git repository.
(Please make sure that you have the latest code changes and add the commit hash in the description)
kernel: Linux 5.15.0-53-generic
os: Ubuntu 20.04.3 LTS (Focal Fossa)
cmake version 3.25.0
gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
clang version 10.0.0-4ubuntu1
```

### To Reproduce

The benchmarks are ISPD2015 contest' hidden cases ( mgc_superblue16_a, mgc_superblue19).
It can be downloaded by the following link : [https://www.ispd.cc/contests/15/web/benchmarks/hidden.tgz](url)

And the commands I used are showed below:

read_lef ${tech.lef}
read_lef ${cell.lef}
read_def ${floorplan.def}
global_placement -density 1

QuantamHD, 2022-12-06:
Your placement density is too high. 1.0 is equivalent to 100% density i.e. perfectly packed try something closer to .35 to start, and move upwards if you think there's room for improvement.

david200340105, 2022-12-07:
I had also tried density of 0.6,0.7,0.8 and 0.9, but it also showed the same error finally.

maliberty, 2022-12-08:
It does work with 0.55 but fails at higher densities.  I look at it briefly and it isn't trivial to resolve.  I'll leave this open to investigate but it may be a while until it gets more attention.

maliberty, 2025-08-25:
This is rather old and a lot has changed since it was filed.  Please reopen if still reproducible.