The-OpenROAD-Project/OpenROAD #2608: libortools.so.9: cannot open shared object file: No such file or directory

zerzerzerz, 2022-12-12:
### Describe the bug

I am installing OpenROAD with Docker. I have successfully the image and run the container. But when I type `openroad -h`, it failed with 
```
openroad: error while loading shared libraries: libortools.so.9: cannot open shared object file: No such file or directory
```

```
NAME="Ubuntu"
VERSION="20.04.2 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.2 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
```
My Docker is `Docker version 20.10.8, build 3967b7d`

### Expected Behavior

`openroad -h` should be executed correctly

### Environment

```shell
I fail to run this command, it tells me to pull the latest branch. But I have already done this and it did not work.
```

### To Reproduce

```bash
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
./build_openroad.sh
docker run -it -u $(id -u ${USER}):$(id -g ${USER}) -v $(pwd)/flow/platforms:/OpenROAD-flow-scripts/flow/platforms:ro openroad/flow-scripts
source ./setup_env.sh
yosys -help
openroad -help
cd flow
make
exit
```
When I run `openroad -help`, it failed with `openroad: error while loading shared libraries: libortools.so.9: cannot open shared object file: No such file or directory`

maliberty, 2022-12-12:
Where do you have or_tools installed?

zerzerzerz, 2022-12-12:
> Where do you have or_tools installed?

If I use Docker to build OpenROAD, is it need to install or_tools? Actually I try to build OpenROAD with Docker.

maliberty, 2022-12-12:
@habibayassin @vvbandeira in the openroad/centos7-builder-gcc container:
```
% ldd build/src/openroad|grep or
	libortools.so.9 => /opt/or-tools/lib64/libortools.so.9 (0x00007f2e88822000)
```
However /opt is empty in the final openroad/flow-scripts:latest container.  We need to copy or_tools over as well as the executable.

maliberty, 2022-12-12:
or add /opt/or-tools to openroad/flow-runtime

zerzerzerz, 2022-12-12:
So what do I need to do? Wait for the latest container or copy `/opt/or-tools` from `openroad/centos7-builder-gcc` to `openroad/flow-scripts:latest`? Thanks!

maliberty, 2022-12-12:
We will put out a new openroad/flow-runtime to resolve this.

vvbandeira, 2022-12-12:
@zerzerzerz can you try the build command again? `./build_openroad.sh`