The-OpenROAD-Project/OpenROAD #2617: Documentation search is broken

xobs, 2022-12-14:
### Describe the bug

The search function of https://openroad.readthedocs.io dies not work. For example, going to https://openroad.readthedocs.io/en/latest/search.html?q=flowtutorial&check_keywords=yes&area=default results in a never-ending "Searching..." prompt, and the error `Uncaught ReferenceError: Stemmer is not defined`

### Expected Behavior

Documentation search should function

### Environment

```shell
v:latest
```

### To Reproduce

[Go to Uncaught ReferenceError: Stemmer is not defined](https://openroad.readthedocs.io/en/latest/search.html?q=flowtutorial&check_keywords=yes&area=default)

### Relevant log output

```shell
Uncaught ReferenceError: Stemmer is not defined
    query https://openroad.readthedocs.io/en/latest/_static/searchtools.js:245
    e https://openroad.readthedocs.io/_/static/javascript/readthedocs-doc-embed.js:1
    jQuery 10
    e https://openroad.readthedocs.io/_/static/javascript/readthedocs-doc-embed.js:1
    promise callback*e https://openroad.readthedocs.io/_/static/javascript/readthedocs-doc-embed.js:1
    setIndex https://openroad.readthedocs.io/en/latest/_static/searchtools.js:182
    <anonymous> https://openroad.readthedocs.io/en/latest/searchindex.js:1
searchtools.js:245:21
Uncaught (in promise) Error: No ad placements found.
    b https://media.ethicalads.io/media/client/beta/ethicalads.min.js:1
    g https://media.ethicalads.io/media/client/beta/ethicalads.min.js:1
    promise callback*ethicalads</</g< https://media.ethicalads.io/media/client/beta/ethicalads.min.js:1
    ethicalads https://media.ethicalads.io/media/client/beta/ethicalads.min.js:1
    ethicalads https://media.ethicalads.io/media/client/beta/ethicalads.min.js:1
    a https://media.ethicalads.io/media/client/beta/ethicalads.min.js:1
    ethicalads https://media.ethicalads.io/media/client/beta/ethicalads.min.js:1
    <anonymous> https://media.ethicalads.io/media/client/beta/ethicalads.min.js:1
```

rovinski, 2022-12-14:
Try https://openroad.readthedocs.io/en/latest/search.html?q=flow+tutorial

If anything this is an upstream template issue.

maliberty, 2022-12-15:
This looks related to https://github.com/sphinx-doc/sphinx/issues/8623

@vvbandeira what version of sphinx are we building the docs with?

vvbandeira, 2022-12-24:
We use the latest of all packages, see [here](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/docs/requirements.txt).

maliberty, 2022-12-26:
It appears to be working now.  @vvbandeira @habibayassin did you fix anything?

vvbandeira, 2022-12-26:
I did push some unrelated changes that triggered a new build of the docs. But nothing specific for this issue.

maliberty, 2022-12-27:
Maybe that triggered picking up a new sphinx version with the fix?

xobs, 2022-12-27:
Still appears to fail for me.

I only just now noticed that the full error message is:

```
Read the Docs search failed. Falling back to Sphinx search.

Uncaught (in promise) ReferenceError: Stemmer is not defined
    query https://openroad.readthedocs.io/en/latest/_static/searchtools.js:245
    T https://openroad.readthedocs.io/_/static/javascript/readthedocs-doc-embed.js:1
    e https://openroad.readthedocs.io/_/static/javascript/readthedocs-doc-embed.js:1
    promise callback*e https://openroad.readthedocs.io/_/static/javascript/readthedocs-doc-embed.js:1
    setIndex https://openroad.readthedocs.io/en/latest/_static/searchtools.js:182
    <anonymous> https://openroad.readthedocs.io/en/latest/searchindex.js:1
```

vvbandeira, 2022-12-27:
> Maybe that triggered picking up a new sphinx version with the fix?

It is possible. @habibayassin mentioned to me on Slack that the theme we use now has a `-old` prefix on the GH repo (https://github.com/SymbiFlow/sphinx_symbiflow_theme-old). We should consider changing the theme to something more up-to-date, I have asked @habibayassin to look at alternatives.

> Still appears to fail for me.

I am not able to reproduce the error. When I open the link, the search indeed does not finish, but there's no error message even after 5+min waiting.

xobs, 2022-12-27:
I should have clarified -- the error message is in the javascript console:

[image]

And you're right, searching for `flow tutorial` works, but `flowtutorial` does not.

vvbandeira, 2022-12-27:
Oh, I see. Then I would consider this a limitation of the current search engine rather than a problem/issue since there's no error presented to the user in the browser. Only someone who knows where to look would find this error; for everyone else, I would assume that the neverending search is enough to try some other keyword.

@habibayassin can you also check if there are any "search engine" plugins to improve this behaviour?

xobs, 2022-12-27:
So I think what happens is if there are no search results, it never returns and presents the error.

It should be changed so that it displays an error if there are no results.

habibayassin, 2022-12-27:
@vvbandeira I have tested out a couple of themes that seem to be solving this issue. As for search plug-ins within this theme, I have tried one extension that didn’t do any difference but I can explore others.

closed 2023-01-05 by vvbandeira