The-OpenROAD-Project/OpenROAD #2621: Problem with global placement with large def file

Tepminamop, 2022-12-14:
### Describe the bug

Currently using Ubutnu 20.04 LTS. I am trying to make global placement using large def file (5.4 millions lines) and on Iter: 370 it is crashing with next error message: /usr/include/c++/11/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = gpl::Bin*; _Alloc = std::allocator<gpl::Bin*>; std::vector<_Tp, _Alloc>::reference = gpl::Bin*&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Stack trace:
0# 0x0000564E55BB20C5 in /home/valeshina/Work/OpenROAD/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
1# 0x00007F5E0B5B8520 in /lib/x86_64-linux-gnu/[libc.so](https://vk.com/away.php?to=http%3A%2F%2Flibc.so&cc_key=).6
2# pthread_kill in /lib/x86_64-linux-gnu/[libc.so](https://vk.com/away.php?to=http%3A%2F%2Flibc.so&cc_key=).6
3# raise in /lib/x86_64-linux-gnu/[libc.so](https://vk.com/away.php?to=http%3A%2F%2Flibc.so&cc_key=).6
4# abort in /lib/x86_64-linux-gnu/[libc.so](https://vk.com/away.php?to=http%3A%2F%2Flibc.so&cc_key=).6
5# 0x0000564E55BB08A8 in /home/valeshina/Work/OpenROAD/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
6# gpl::BinGrid::updateBinsGCellDensityArea(std::vector<gpl::GCell*, std::allocator<gpl::GCell*> > const&) in /home/valeshina/Work/OpenROAD/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
7# gpl::NesterovPlace::doNesterovPlace(int) in /home/valeshina/Work/OpenROAD/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
8# 0x0000564E55C08DB8 in /home/valeshina/Work/OpenROAD/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
9# TclNRRunCallbacks in /lib/x86_64-linux-gnu/[libtcl8.6.so](https://vk.com/away.php?to=http%3A%2F%2Flibtcl8.6.so&cc_key=)
10# 0x00007F5E0D453028 in /lib/x86_64-linux-gnu/[libtcl8.6.so](https://vk.com/away.php?to=http%3A%2F%2Flibtcl8.6.so&cc_key=)
11# Tcl_EvalEx in /lib/x86_64-linux-gnu/[libtcl8.6.so](https://vk.com/away.php?to=http%3A%2F%2Flibtcl8.6.so&cc_key=)
12# Tcl_Eval in /lib/x86_64-linux-gnu/[libtcl8.6.so](https://vk.com/away.php?to=http%3A%2F%2Flibtcl8.6.so&cc_key=)
13# gui::ScriptWidget::executeTclCommand(QString const&) in /home/valeshina/Work/OpenROAD/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
14# gui::ScriptWidget::executeCommand(QString const&, bool) in /home/valeshina/Work/OpenROAD/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
15# 0x00007F5E0C39A7C8 in /lib/x86_64-linux-gnu/[libQt5Core.so](https://vk.com/away.php?to=http%3A%2F%2FlibQt5Core.so&cc_key=).5
16# gui::TclCmdInputWidget::completeCommand(QString const&) in /home/valeshina/Work/OpenROAD/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
17# gui::TclCmdInputWidget::keyPressEvent(QKeyEvent*) in /home/valeshina/Work/OpenROAD/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
18# QWidget::event(QEvent*) in /lib/x86_64-linux-gnu/[libQt5Widgets.so](https://vk.com/away.php?to=http%3A%2F%2FlibQt5Widgets.so&cc_key=).5
19# QFrame::event(QEvent*) in /lib/x86_64-linux-gnu/[libQt5Widgets.so](https://vk.com/away.php?to=http%3A%2F%2FlibQt5Widgets.so&cc_key=).5
20# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/[libQt5Widgets.so](https://vk.com/away.php?to=http%3A%2F%2FlibQt5Widgets.so&cc_key=).5
21# QApplication::notify(QObject*, QEvent*) in /lib/x86_64-linux-gnu/[libQt5Widgets.so](https://vk.com/away.php?to=http%3A%2F%2FlibQt5Widgets.so&cc_key=).5
22# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/[libQt5Core.so](https://vk.com/away.php?to=http%3A%2F%2FlibQt5Core.so&cc_key=).5
23# 0x00007F5E0CEB402F in /lib/x86_64-linux-gnu/[libQt5Widgets.so](https://vk.com/away.php?to=http%3A%2F%2FlibQt5Widgets.so&cc_key=).5
24# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/[libQt5Widgets.so](https://vk.com/away.php?to=http%3A%2F%2FlibQt5Widgets.so&cc_key=).5
25# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/[libQt5Core.so](https://vk.com/away.php?to=http%3A%2F%2FlibQt5Core.so&cc_key=).5
26# QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) in /lib/x86_64-linux-gnu/[libQt5Gui.so](https://vk.com/away.php?to=http%3A%2F%2FlibQt5Gui.so&cc_key=).5
27# QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/[libQt5Gui.so](https://vk.com/away.php?to=http%3A%2F%2FlibQt5Gui.so&cc_key=).5
28# 0x00007F5E070C8D6E in /lib/x86_64-linux-gnu/[libQt5XcbQpa.so](https://vk.com/away.php?to=http%3A%2F%2FlibQt5XcbQpa.so&cc_key=).5
29# g_main_context_dispatch in /lib/x86_64-linux-gnu/[libglib-2.0.so](https://vk.com/away.php?to=http%3A%2F%2Flibglib-2.0.so&cc_key=).0
30# 0x00007F5E0AC536F8 in /lib/x86_64-linux-gnu/[libglib-2.0.so](https://vk.com/away.php?to=http%3A%2F%2Flibglib-2.0.so&cc_key=).0
31# g_main_context_iteration in /lib/x86_64-linux-gnu/[libglib-2.0.so](https://vk.com/away.php?to=http%3A%2F%2Flibglib-2.0.so&cc_key=).0
32# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/[libQt5Core.so](https://vk.com/away.php?to=http%3A%2F%2FlibQt5Core.so&cc_key=).5
33# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/[libQt5Core.so](https://vk.com/away.php?to=http%3A%2F%2FlibQt5Core.so&cc_key=).5
34# QCoreApplication::exec() in /lib/x86_64-linux-gnu/[libQt5Core.so](https://vk.com/away.php?to=http%3A%2F%2FlibQt5Core.so&cc_key=).5
35# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) in /home/valeshina/Work/OpenROAD/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
36# ord::tclAppInit(Tcl_Interp*) in /home/valeshina/Work/OpenROAD/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
37# Tcl_MainEx in /lib/x86_64-linux-gnu/[libtcl8.6.so](https://vk.com/away.php?to=http%3A%2F%2Flibtcl8.6.so&cc_key=)
38# main in
/home/valeshina/Work/OpenROAD/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
39# 0x00007F5E0B59FD90 in /lib/x86_64-linux-gnu/[libc.so](https://vk.com/away.php?to=http%3A%2F%2Flibc.so&cc_key=).6
40# __libc_start_main in /lib/x86_64-linux-gnu/[libc.so](https://vk.com/away.php?to=http%3A%2F%2Flibc.so&cc_key=).6
41# _start in /home/valeshina/Work/OpenROAD/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad. 

When I am using smaller def files, I have no errors.

### Expected Behavior

I expect to successfully make global placement with large def file

### Environment

```shell
Unfortunately, I dont have Env.sh.
```

### To Reproduce

I used read_lef and read_def, then I used global_placement -density 0.86. Usually it works fine with smaller def files.

### Relevant log output

```shell
/usr/include/c++/11/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = gpl::Bin*; _Alloc = std::allocator<gpl::Bin*>; std::vector<_Tp, _Alloc>::reference = gpl::Bin*&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Stack trace:
0# 0x0000564E55BB20C5 in /home/valeshina/Work/OpenROAD/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
[... 19 lines cut ...]
20# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/[libQt5Widgets.so](https://vk.com/away.php?to=http://libQt5Widgets.so&cc_key=).5
21# QApplication::notify(QObject*, QEvent*) in /lib/x86_64-linux-gnu/[libQt5Widgets.so](https://vk.com/away.php?to=http://libQt5Widgets.so&cc_key=).5
22# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/[libQt5Core.so](https://vk.com/away.php?to=http://libQt5Core.so&cc_key=).5
23# 0x00007F5E0CEB402F in /lib/x86_64-linux-gnu/[libQt5Widgets.so](https://vk.com/away.php?to=http://libQt5Widgets.so&cc_key=).5
24# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/[libQt5Widgets.so](https://vk.com/away.php?to=http://libQt5Widgets.so&cc_key=).5
25# QCoreApplication::notifyInternal2(QObject*, QEvent*) in
/lib/x86_64-linux-gnu/[libQt5Core.so](https://vk.com/away.php?to=http://libQt5Core.so&cc_key=).5
26# QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) in /lib/x86_64-linux-gnu/[libQt5Gui.so](https://vk.com/away.php?to=http://libQt5Gui.so&cc_key=).5
[... cut ...]
37# Tcl_MainEx in /lib/x86_64-linux-gnu/[libtcl8.6.so](https://vk.com/away.php?to=http://libtcl8.6.so&cc_key=)
38# main in
/home/valeshina/Work/OpenROAD/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
39# 0x00007F5E0B59FD90 in /lib/x86_64-linux-gnu/[libc.so](https://vk.com/away.php?to=http://libc.so&cc_key=).6
40# __libc_start_main in /lib/x86_64-linux-gnu/[libc.so](https://vk.com/away.php?to=http://libc.so&cc_key=).6
41# _start in /home/valeshina/Work/OpenROAD/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad.
```

maliberty, 2022-12-14:
You need to supply a test case

Tepminamop, 2022-12-14:
> You need to supply a test case

Added lef and def files
[test_case.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/10230849/test_case.zip)
[test_case_def.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/10230852/test_case_def.zip)

rovinski, 2022-12-14:
Seems like it could be an out-of-memory error? @Tepminamop how much memory is on this system? Are you running in Docker or locally?

You should have `Env.sh` at `/home/valeshina/Work/OpenROAD/OpenROAD-flow-scripts/tools/OpenROAD/etc/Env.sh`. Please update the description with this info if you can.

maliberty, 2022-12-14:
I have deleted your test case as I saw it contained a proprietary LEF.

maliberty, 2022-12-14:
I did run it before I noticed and got:
```
[INFO GPL-0019] Util(%): 143.11
[INFO GPL-0020] StdInstsArea: 248950135193600
[INFO GPL-0021] MacroInstsArea: 0
[ERROR GPL-0301] Utilization exceeds 100%.
[ERROR GUI-0070] GPL-0301
```

Please reproduce this on a non-proprietary PDK if possible.

Tepminamop, 2022-12-15:
> Seems like it could be an out-of-memory error? @Tepminamop how much memory is on this system? Are you running in Docker or locally?
> 
> You should have `Env.sh` at `/home/valeshina/Work/OpenROAD/OpenROAD-flow-scripts/tools/OpenROAD/etc/Env.sh`. Please update the description with this info if you can.

I tried to run locally on Ubuntu 20.04 with 8 GB RAM and on WSL2 with 32 GB RAM, but it crashes in both cases. I checked Task Manager and OpenROAD uses ~6 GB RAM when I am trying to make global palcement.

Tepminamop, 2022-12-15:
> I did run it before I noticed and got:
> 
> ```
> [INFO GPL-0019] Util(%): 143.11
> [INFO GPL-0020] StdInstsArea: 248950135193600
> [INFO GPL-0021] MacroInstsArea: 0
> [ERROR GPL-0301] Utilization exceeds 100%.
> [ERROR GUI-0070] GPL-0301
> ```
> 
> Please reproduce this on a non-proprietary PDK if possible.

I am sorry but I gave wrong def file where Util(%) is 143.11, I fixed it and now Util(%) is around 74, here is log:
[INFO ODB-0127] Reading DEF file: ./defs/MKZI_0large.def
[INFO ODB-0128] Design: MKZI_dev
[INFO ODB-0094] 		Created 100000 Insts
[INFO ODB-0094] 		Created 200000 Insts
[INFO ODB-0094] 		Created 300000 Insts
[INFO ODB-0094] 		Created 400000 Insts
[INFO ODB-0094] 		Created 500000 Insts
[INFO ODB-0094] 		Created 600000 Insts
[INFO ODB-0094] 		Created 700000 Insts
[INFO ODB-0094] 		Created 800000 Insts
[INFO ODB-0094] 		Created 900000 Insts
[INFO ODB-0094] 		Created 1000000 Insts
[INFO ODB-0097] 		Created 100000 Nets
[INFO ODB-0097] 		Created 200000 Nets
[INFO ODB-0097] 		Created 300000 Nets
[INFO ODB-0097] 		Created 400000 Nets
[INFO ODB-0097] 		Created 500000 Nets
[INFO ODB-0097] 		Created 600000 Nets
[INFO ODB-0097] 		Created 700000 Nets
[INFO ODB-0097] 		Created 800000 Nets
[INFO ODB-0097] 		Created 900000 Nets
[INFO ODB-0097] 		Created 1000000 Nets
[INFO ODB-0097] 		Created 1100000 Nets
[INFO ODB-0097] 		Created 1200000 Nets
[INFO ODB-0097] 		Created 1300000 Nets
[INFO ODB-0097] 		Created 1400000 Nets
[INFO ODB-0130]     Created 184 pins.
[INFO ODB-0131]     Created 1072262 components and 6615934 component-terminals.
[INFO ODB-0133]     Created 1434377 nets and 4471778 connections.
[INFO ODB-0134] Finished DEF file: ./defs/MKZI_0large.def

[INFO GPL-0006] NumInstances: 1072078
[INFO GPL-0007] NumPlaceInstances: 1072078
[INFO GPL-0008] NumFixedInstances: 0
[INFO GPL-0009] NumDummyInstances: 0
[INFO GPL-0010] NumNets: 1434377
[INFO GPL-0011] NumPins: 4471962
[INFO GPL-0012] DieAreaLxLy: 0 0
[INFO GPL-0013] DieAreaUxUy: 27197440 14786560
[INFO GPL-0014] CoreAreaLxLy: 798720 798720
[INFO GPL-0015] CoreAreaUxUy: 26398720 13987840
[INFO GPL-0016] CoreArea: 337641472000000
[INFO GPL-0017] NonPlaceInstsArea: 0
[INFO GPL-0018] PlaceInstsArea: 248950135193600
[INFO GPL-0019] Util(%): 73.73
[INFO GPL-0020] StdInstsArea: 248950135193600
[INFO GPL-0021] MacroInstsArea: 0

[InitialPlace]  Iter: 20 CG residual: 0.00002070 HPWL: 61448146729
[INFO GPL-0031] FillerInit: NumGCells: 1287037
[INFO GPL-0032] FillerInit: NumGNets: 1434377
[INFO GPL-0033] FillerInit: NumGPins: 4471962
[INFO GPL-0023] TargetDensity: 0.86
[INFO GPL-0024] AveragePlaceInstArea: 232212707
[INFO GPL-0025] IdealBinArea: 270014752
[INFO GPL-0026] IdealBinCnt: 1250455
[INFO GPL-0027] TotalBinArea: 337641472000000
[INFO GPL-0028] BinCnt: 1024 1024
[INFO GPL-0029] BinSize: 25000 12880
[INFO GPL-0030] NumBins: 1048576

maliberty, 2022-12-15:
I doubt 1M instances runs in  "~6 GB RAM".   Without a test case to examine I don't have more to suggest.

Tepminamop, 2022-12-15:
> I doubt 1M instances runs in "~6 GB RAM". Without a test case to examine I don't have more to suggest.

> I doubt 1M instances runs in "~6 GB RAM". Without a test case to examine I don't have more to suggest.
Task Manager told me that OpenROAD uses 5.7 GB RAM and my entire system usign 7 GB RAM (I applied screenshot), I applied fixed def file as a test case
[test_case.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/10235803/test_case.zip)
[image]
[image]

maliberty, 2022-12-15:
RSS is resident set size and only tells you what is currently swapped in, not total process size.

maliberty, 2022-12-15:
I can't do anything with your DEF as it uses cells from a proprietary LEF.

maliberty, 2023-02-10:
Closing with no actionable.  You can reopen if a test case becomes available.

closed 2023-02-10 by maliberty