The-OpenROAD-Project/OpenROAD #2637: cts viewer highlight cofusing for mulitple clocks

maliberty, 2022-12-15:
### Describe the bug

If I open sky130hd/microwatt I see:

[image]

It seems both clocks are being highlighted at once.  

### Expected Behavior

I think we should only highlight for the clock in the active tab.

### Environment

```shell
head of master
```

### To Reproduce

as above

gadfort, 2022-12-17:
@maliberty would you still want to be able to turn off the clock rendering all together? Ie. so you can view the tree without the rendering or do you think that would not be a useful control?

maliberty, 2022-12-17:
I think that would be useful too

gadfort, 2022-12-21:
@maliberty when I run the skyhd microwatt I get
```
[INFO][FLOW-xxxx] Using manual macro placement file ./designs/sky130hd/microwatt/microwatt.macro_placement.cfg
Error: macro_place.tcl, 125 Cannot find instance soc0.bram.bram0.ram_0.memory_0
```

maliberty, 2022-12-21:
It passed in last night's CI.  Something odd is happening in your setup.

gadfort, 2022-12-21:
> It passed in last night's CI.  Something odd is happening in your setup.

Yes, I had turned off flattening its working again

closed 2022-12-22 by gadfort