The-OpenROAD-Project/OpenROAD #2638: pdngen missing repair strap

maliberty, 2022-12-16:
### Describe the bug

With the update to hier-rtlmp & cts a problem has developed in ng45/bp_be_top where psm errors on a disconnected power grid.  
```
[WARNING PSM-0038] Unconnected PDN node on net VDD at location (10.070um, 505.400um), layer: 1.
[WARNING PSM-0039] Unconnected instance PHY_2260 at location (10.070um, 505.400um) layer: 1.
[WARNING PSM-0039] Unconnected instance PHY_2250 at location (10.070um, 505.400um) layer: 1.
...
```
It appears that pdn is not putting in a repair strap in a channel on the left:

[image]

You can see a repair strap in the lower portion of the channel but not in the upper portion.  

I have temporarily disabled psm on this design.  When this is fixed it should be re-enabled.

### Expected Behavior

A fully connected power grid.

### Environment

```shell
head of master
```

### To Reproduce

DESIGN_CONFIG=./designs/nangate45/bp_be_top/config.mk make

gadfort, 2022-12-20:
@maliberty I ran the command, but my output looks mirrored from the picture you uploaded. And I haven't been able to see the issue yet on it.

maliberty, 2022-12-20:
Maybe something else changed - I'll rerun it and attach a test case if I can reproduce it.

maliberty, 2022-12-20:
Something has changed and the problem disappeared due to change in macro locations.  That's teach me to not record the commit id.  I'll close this and reopen if the problem returns.

closed 2022-12-20 by maliberty