The-OpenROAD-Project/OpenROAD #2642: Error from place_pins command

nayiri-k, 2022-12-16:
### Describe the bug

OpenROAD segfaults from place_pins command.

`place_pins -random -hor_layers {met3} -ver_layers {met2 met4} `

(with or without `-random` argument)

### Expected Behavior

In my previous install version of OpenROAD (~Jan 2022), this error never occurred and pins were placed normally.

### Environment

```shell
git commit: unknown, this is not a git repository.
(Please make sure that you have the latest code changes and add the commit hash in the description)
kernel: Linux 3.10.0-1160.80.1.el7.x86_64
os: Red Hat Enterprise Linux Server 7.9 (Maipo)
cmake version 3.22.1
gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-44)
clang version 15.0.6 (https://github.com/conda-forge/clangdev-feedstock b6cf7124da893bcb57ab1f38c768ea825fabbdb1)

**NOTE: this OpenROAD install is located on a server where I don't have sudo permissions, so the git version may not have been located due to this
Running 'git log' returns:
commit 5144431cb9ad3a71208cf0a810a0674cfeced15b
Merge: f86efab 23abb26
Author: Peter <46405338+arlpetergadfort@users.noreply.github.com>
Date:   Fri Oct 28 10:56:28 2022 -0400

    Merge pull request #2430 from The-OpenROAD-Project-staging/pdn-clang
    
    pdn: clang format
```

### To Reproduce

[ChipTop_Linux-3.10.0-1160.80.1.el7.x86_64-x86_64-with-redhat-7.9-Maipo_2022-12-16_10-26.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/10248207/ChipTop_Linux-3.10.0-1160.80.1.el7.x86_64-x86_64-with-redhat-7.9-Maipo_2022-12-16_10-26.tar.gz)

```
tar -xvf ChipTop_Linux-3.10.0-1160.80.1.el7.x86_64-x86_64-with-redhat-7.9-Maipo_2022-12-16_10-26.tar.gz
cd ChipTop_Linux-3.10.0-1160.80.1.el7.x86_64-x86_64-with-redhat-7.9-Maipo_2022-12-16_10-26
./runme.sh
```

### Relevant log output

```shell
place_pins  -hor_layers {met3} -ver_layers {met2 met4} 
Found 3 macro blocks.
Using 1u default distance from corners.
Using 2 tracks default min distance between IO pins.
/opt/rh/devtoolset-8/root/usr/include/c++/8/bits/stl_vector.h:932: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int; _Alloc = std::allocator<int>; std::vector<_Tp, _Alloc>::reference = int&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Signal 6 received
Stack trace:
 0# 0x0000000000C9D887 in openroad
 1# 0x00007F5ABE03B400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# boost::stacktrace::detail::this_thread_frames::collect(void const**, unsigned long, unsigned long) in openroad
 5# 0x0000000000D44B8D in openroad
 6# ppl::IOPlacer::findSlots(std::set<int, std::less<int>, std::allocator<int> > const&, ppl::Edge) in openroad
 7# ppl::IOPlacer::defineSlots() in openroad
 8# ppl::IOPlacer::run(bool) in openroad
 9# 0x0000000000DFCDFD in openroad
10# 0x00007F5AC0D87EB2 in /lib64/libtcl8.5.so
11# 0x00007F5AC0DCC390 in /lib64/libtcl8.5.so
12# TclObjInterpProcCore in /lib64/libtcl8.5.so
13# 0x00007F5AC0D87EB2 in /lib64/libtcl8.5.so
14# 0x00007F5AC0DCC390 in /lib64/libtcl8.5.so
15# 0x00007F5AC0DD46E7 in /lib64/libtcl8.5.so
16# TclEvalObjEx in /lib64/libtcl8.5.so
17# 0x00007F5AC0E0F31F in /lib64/libtcl8.5.so
18# 0x00007F5AC0D87EB2 in /lib64/libtcl8.5.so
19# 0x00007F5AC0DCC390 in /lib64/libtcl8.5.so
20# 0x00007F5AC0DD46E7 in /lib64/libtcl8.5.so
21# TclEvalObjEx in /lib64/libtcl8.5.so
22# 0x00007F5AC0D8F1D0 in /lib64/libtcl8.5.so
23# 0x00007F5AC0D87EB2 in /lib64/libtcl8.5.so
24# 0x00007F5AC0DCC390 in /lib64/libtcl8.5.so
25# 0x00007F5AC0DD46E7 in /lib64/libtcl8.5.so
26# TclEvalObjEx in /lib64/libtcl8.5.so
27# 0x00007F5AC0D92F00 in /lib64/libtcl8.5.so
28# 0x00007F5AC0D87EB2 in /lib64/libtcl8.5.so
29# 0x00007F5AC0DCC390 in /lib64/libtcl8.5.so
30# TclObjInterpProcCore in /lib64/libtcl8.5.so
31# 0x00007F5AC0D87EB2 in /lib64/libtcl8.5.so
32# 0x00007F5AC0DCC390 in /lib64/libtcl8.5.so
33# TclObjInterpProcCore in /lib64/libtcl8.5.so
34# 0x00007F5AC0D87EB2 in /lib64/libtcl8.5.so
35# 0x00007F5AC0D88F1E in /lib64/libtcl8.5.so
36# Tcl_EvalEx in /lib64/libtcl8.5.so
37# Tcl_Eval in /lib64/libtcl8.5.so
38# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
39# ord::tclAppInit(Tcl_Interp*) in openroad
40# Tcl_Main in /lib64/libtcl8.5.so
41# main in openroad
42# __libc_start_main in /lib64/libc.so.6
43# 0x0000000000C98BC7 in openroad
```

maliberty, 2022-12-16:
@eder-matheus there appears to be an assumption that there will be an equal number of layers for H & V pins in IOPlacer::findSlots.  This case breaks that assumption.

maliberty, 2022-12-16:
@nayiri-k in your earlier run did you use the same layer specification?  It doesn't seem like a new issue.

eder-matheus, 2022-12-20:
@nayiri-k Could you try this branch: https://github.com/eder-matheus/OpenROAD/tree/ppl_fix_slots?

closed 2022-12-21 by eder-matheus