The-OpenROAD-Project/OpenROAD #2643: Invalid command name "partition_design"

Fia132, 2022-12-17:
### Describe the bug

I have downloaded MacroPlacement from TILOS to convert lef, def into netlist protocol buffer formats. 
[https://github.com/TILOS-AI-Institute/MacroPlacement](url)

The sample clustering code using `openroad` binary can be found in the directory, `MacroPlacement/CodeElements/Clustering/test`.

However, due to the compatibility issue, I got a following issue. 
`./openroad: symbol lookup error: ./openroad: undefined symbol: _ZdlPvm, version Qt_5`

I, therefore, compiled `openroad` binary from the current openroad github.
However, replacing the previous `openroad` binary to `OpenROAD/build/src/openroad` occurs a following error.

> python3 test_clustering.py 
OpenROAD v2.0-6114-gcf23c5680 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: ./lefs/NangateOpenCellLibrary.tech.lef
[INFO ODB-0223]     Created 22 technology layers
[INFO ODB-0224]     Created 27 technology vias
[INFO ODB-0226] Finished LEF file:  ./lefs/NangateOpenCellLibrary.tech.lef
[INFO ODB-0222] Reading LEF file: ./lefs/NangateOpenCellLibrary.macro.mod.lef
[INFO ODB-0225]     Created 135 library cells
[INFO ODB-0226] Finished LEF file:  ./lefs/NangateOpenCellLibrary.macro.mod.lef
[INFO ODB-0222] Reading LEF file: ./lefs/fakeram45_256x16.lef
[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./lefs/fakeram45_256x16.lef
[WARNING ORD-1002] Liberty cell LOGIC0_X1 pin Z missing from LEF macro.
[WARNING ORD-1002] Liberty cell LOGIC1_X1 pin Z missing from LEF macro.
[INFO ODB-0127] Reading DEF file: ./design/ariane.def
[INFO ODB-0128] Design: ariane
[INFO ODB-0097]                 Created 100000 Nets
[INFO ODB-0130]     Created 495 pins.
[INFO ODB-0131]     Created 89806 components and 587982 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 179612 connections.
[INFO ODB-0133]     Created 115333 nets and 408370 connections.
[INFO ODB-0134] Finished DEF file: ./design/ariane.def
Error: extract_hypergraph.tcl, 54 invalid command name "partition_design"

I searched for the commands of `openroad` by `./openroad; help`, but I was not able to find the command "partition_design".

> OpenROAD v2.0-6114-gcf23c5680 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
% help
add_global_connection -net
add_pad [-name name] [-type type] [-cell library_cell] [-signal signal_name]
   [-edge edge] [-location location] [-bump rowcol]
   [-padcell_to_rdl list_of_vias] [-rdl_to_bump list_of_vias] [-bondpad bondpad]
   [-inst_name inst_name]
add_pdn_connect [-grid grid_name] -layers list_of_2_layers
   [-cut_pitch pitch_value] [-fixed_vias list_of_vias]
   [-dont_use_vias list_of_vias] [-max_rows rows] [-max_columns columns]
   [-ongrid ongrid_layers] [-split_cuts split_cuts_mapping]
add_pdn_ring [-grid grid_name] -layers list_of_2_layer_names -widths
add_pdn_stripe [-grid grid_name] -layer layer_name [-width width_value]
   [-followpins] [-extend_to_core_ring] [-pitch pitch_value]
   [-spacing spacing_value] [-offset offset_value]
   [-starts_width (POWER|GROUND)] [-extend_to_boundary] [-snap_to_grid]
   [-number_of_straps count] [-nets list_of_nets]
add_to_physical_cluster
   [-modinst path | -inst inst_name | -physical_cluster cluster_name]
   cluster_name
add_worker_address [-host host] [-port port]
adjust_rc [-res_factor res] [-cc_factor cc] [-gndc_factor gndc]
all_clocks
all_inputs
all_outputs
all_registers [-clock clocks] [-rise_clock clocks] [-fall_clock clocks] [-cells]
   [-data_pins] [-clock_pins] [-async_pins] [-output_pins] [-level_sensitive]
[... 557 lines cut ...]

The command, `partitioning_design` works without committing any errors.
Successful conversion of Lef, def format into netlist protocol buffer format.

### Environment

```shell
Git commit: cf23c5680233290495b817a84611c5b222f4af4f
kernel: Linux 5.13.0-41-generic
os: Ubuntu 20.04.5 LTS (Focal Fossa)
cmake version 3.25.1
gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
```

### To Reproduce

```
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git
cd OpenROAD
./etc/Build.sh -cmake="-DCMAKE_INSTALL_PREFIX=<prefix_path>" -no-gui

cd bulid/src
./openroad
partitioning_design
```

ravi-varadarajan, 2022-12-17:
partition_design is a depricated command in OpenROAD.  Which script was using the command?

ravi-varadarajan, 2022-12-17:
Will investigate further with the MacroPlacement repo scripts and get back

Fia132, 2022-12-17:
I appreciate your effort. Originally, I want to run [https://github.com/google-research/circuit_training](url) with my custom designs.
I faced the error on the preprocess stage, the conversion of LEF/DEF and Bookshelf to the Netlist Protocol Buffer used as the input for circuit training.
The description for the conversion is below:
 [https://github.com/TILOS-AI-Institute/MacroPlacement/tree/main/CodeElements/FormatTranslators](url) 
And the test script running the above is represented as follow (which is almost equivalent to the aforementioned script from the issue `test_clustering.py`):
[https://github.com/TILOS-AI-Institute/MacroPlacement/blob/main/CodeElements/FormatTranslators/test/LefDef2ProtocolBufferFormat/test2.py](url)

maliberty, 2025-02-21:
stale

closed 2025-02-21 by maliberty