The-OpenROAD-Project/OpenROAD #2666: filler_placement $::env(FILL_CELLS) terminates script with exit code 0

oharboe, 2022-12-24:
### Describe the bug

When building a design with ASAP7, ```filler_placement $::env(FILL_CELLS)``` terminates the script with exit code zero. 

Since there's no file generated by this step in OpenROAD-flow-script the flow terminates as make doesn't see the dependency generated.

### Expected Behavior

Either an error or success.

### Environment

```shell
The example includes proprietary code. I'll work on debugging this to see if I can create a reproducible case.
```

### To Reproduce

The example includes proprietary code. I'll work on debugging this to see if I can create a reproducible case.

### Relevant log output

```shell
Here we can see that the step completed with exit code 0, but that no 4_2_cts_fillcell.odb was created.

Elapsed time: 1:44.97[h:]min:sec. CPU time: user 76.62 sys 24.61 (96%). Peak memory: 31761364KB.
Reaping losing child 0x907cc0 PID 64 
make: *** [results/asap7/foobar/base/4_2_cts_fillcell.odb] Error 9
Removing child 0x907cc0 PID 64 from chain.
```

maliberty, 2022-12-24:
4_2_cts_fillcell.odb is generated by fillcell.tcl so I'm not sure what you are referring to.

oharboe, 2022-12-24:
> 4_2_cts_fillcell.odb is generated by fillcell.tcl so I'm not sure what you are referring to.

I added puts calls before and after ```filler_placement $::env(FILL_CELLS)```. The puts before was printed, the one after was not, but the exit code of openroad was 0.

That is, ```  write_db $::env(RESULTS_DIR)/4_2_cts_fillcell.odb``` was ever reached, even if there was no error message.

```
puts "fillcell 2"

filler_placement $::env(FILL_CELLS)
puts "fillcell 3"
check_placement
puts "fillcell 4"

if {![info exists save_checkpoint] || $save_checkpoint} {
puts "fillcell 5"
  write_db $::env(RESULTS_DIR)/4_2_cts_fillcell.odb
}
puts "fillcell 6"
```

oharboe, 2022-12-24:
Trying to step through the code a bit to see what I see...

[image]

maliberty, 2022-12-24:
Does it throw any exceptions?  In gdb you can do "catch throw" to stop when one is thrown.

oharboe, 2022-12-25:
> 

No exceptions, it just terminates with exit code 0:

[image]

```
row=0
row=1000
row=2000
row=3000
$
```

oharboe, 2022-12-25:
Nice.... I've seen it terminate without error after 3265 rows and 3268 rows... So it would seem that the process gets "poisoned" and then dies with exit code 0 some time after...

oharboe, 2022-12-25:
Memory usage at:

- 1000 rows, 30gbyte ram
- 2000 rows, 39gbyte ram.

so... 10gbyte/1000 rows. I seem to recall 8151 rows. That's 80gbyte RAM or so... I think I'll try to reconfigure my system with virtual memory...

```
$ cat /proc/meminfo
MemTotal:       65699520 kB
MemFree:        48513920 kB
MemAvailable:   54923640 kB
[deleted]
```

oharboe, 2022-12-25:
Cleaned up machine, I now see virtual memory usage of this process peaks at about 140gbyte.

oharboe, 2022-12-25:
This stage in the flow completed after ca. 90 minutes by increasing available memory.

So, it would appear that Linux can terminate a process without an exit code if it runs out of memory. This is a little bit surprising, but not a complete shocker...

Looking at memory usage of tritonroute, it takes a long time to start, but I already see it using 109gbyte of memory, so I'm going to go out on a limb and to say that even if the CTS stage could reduce memory usage drastically, it would improve performance substantially, but not reduce the maximum amount of memory necessary to run the flow.

oharboe, 2022-12-25:
@maliberty Close this issue?

rovinski, 2022-12-26:
@oharboe can you specify roughly how many instances are in the design?

maliberty, 2022-12-26:
What are the core dimension?  

Its your issue so you are free to close it anytime you wish.

oharboe, 2022-12-26:
> What are the core dimension?
> 
> Its your issue so you are free to close it anytime you wish.

@maliberty @rovinski 

I didn't want to close it if you had some more questions for me w.r.t. investigations or tests to try.

I'll close it now, feel free to ask followup questions if you have them.

This was with ASAP7.

I was interested in knowing how OpenROAD scaled, how much physical memory is needed for a design and if it would crash. My conclusion is that virtual memory doesn't really help, because it slows down the execution by a factor of 10-20x.

It looks like 2TB of RAM, on the order of 10x more RAM than in all the C64s ever sold :-), would cost ca. $15000. That's not too bad if one knows that the OpenROAD tools don't fall over at those scales.

This is a top-level design, with macros having almost all the logic. I was curious if OpenROAD would save a lot of memory by having mainly macros at the top level, then I scaled up until OpenROAD showed signs of scaling problems. 

I only have 64gbyte of RAM on the machine I tested with and I ran out. The "time" command shows maximum physical memory, not maximum virtual memory, so it's not possible to when I ran out, but TritonRoute used up all my virtual memory(256gbyte) when I looked in "top" at some point.

To find out how OpenROAD scales, I'll need to run a substantially smaller design that maxes out at 64 gbyte and then my prior assumption, until I know more, is that memory usage scales linearly with area or instances.

From: 3_5_opendp.log:

Design area 987194 u^2 20% utilization.

From 3_4_resizer.log:

```
instance_count
--------------------------------------------------------------------------
350908
```

maliberty, 2022-12-26:
There is some old work to improve the memory usage around fill cells that could help (never quite finished as the developer left).  Is the peak during detailed routing?

oharboe, 2022-12-27:
@maliberty 
> There is some old work to improve the memory usage around fill cells that could help (never quite finished as the developer left). Is the peak during detailed routing?

I saw ca. 2x the memory usage of fill cell during 5_1_fastroute.log.

Does this work old work reduce memory usage as of fill cell, or only during fillcell?

If it only reduces it *during* fillcell, then it isn't going to remove the peak memory usage of the flow.

maliberty, 2022-12-27:
The work was on detailed route.  I don't know why fastroute would be the peak.  Can you make a generic reproducer of your issue?

oharboe, 2022-12-27:
> The work was on detailed route. I don't know why fastroute would be the peak. Can you make a generic reproducer of your issue?

I will try. I have some ideas on how to create ca. 1mm^2 ASAP7 designs that can be open sourced.

maliberty, 2022-12-27:
We have a large non public design which is
```
export DIE_AREA    = 0 0 1400 1400 
```
In the stats you generate I see:
```
3_3_place_gp                    2552
5_1_fastroute                   2204
5_2_TritonRoute                  147
6_report                        2266
```

So I can see something like what you report.

maliberty, 2022-12-27:
However your timings are not correct as I see 5_2 took: elapsed time = 02:10:41 which should be 7841s

oharboe, 2022-12-27:
> However your timings are not correct as I see 5_2 took: elapsed time = 02:10:41 which should be 7841s

Ah. That's what you get from having chatgpt write your code for you :-) Needs a tweak so that it counts hours too.

maliberty, 2022-12-27:
It would be nice to add "Peak memory" to your table too

oharboe, 2022-12-27:
> It would be nice to add "Peak memory" to your table too

Yes... however, we only get peak physical memory from "time", not peak virtual memory.

oharboe, 2022-12-30:
This is mainly a problem that occurs because there are too many ASAP7 filler cells: #732

closed 2022-12-30 by oharboe