The-OpenROAD-Project/OpenROAD #2667: Clock tree exceptions

Blebowski, 2022-12-26:
### Description

Hello,

I was wondering whether OpenROAD currently provides a way how to specify CTS exceptions/constraints
(I did not find any). By CTS constraints, I reffer to ability of specifying:
-  Leaf / Exclude / Ignore pins.
- Skew groups
- Float pins - Ability to achieve usefull skew.

If not, I am wondering whether there is an intent to add support for such features.

I hope this is adequate place to open this question (That I should not put it into discussion).

maliberty, 2022-12-26:
In short, no support exists.  Plans exist only the general sense that is something we should do.  How would you prioritize your list?

Blebowski, 2022-12-26:
Hi, @maliberty , thanks for such a quick response.

First let me say that I am no PnR expert here, I do front-end work, Synthesis, Constraining and Sign-off, but
PnR is something that I would eventually like to learn and handle. That being said, what I say below might
be completely inacurate.

As for the priorities, I think following makes sense (first item is most important)
1. Exclude pins - Allows completely excluding output pins to which clock propagates. That is important for chip-top level stuff, where it is common to bring-out internal clocks for debug / observability / functional testability at waver level, etc...
2. Float pins - Allows manually adding skew on some part of the design. Basically, if you add hand-written clock gate on RTL, e.g. to gate clock for whole "Block A", then setting pin of this clock gate as float pin, accounts for adding/removing skew on all flops within the block (which is effectively a skew group).
3. Rest of the stuff - Skew groups (with specified target usefull skew), Maybe "dont_touch" on sub-tree.

Btw. wish you a nice almost post-Christmas time :)

closed 2025-08-30 by maliberty