The-OpenROAD-Project/OpenROAD #2674: Global Routing crashes

vijayank88, 2022-12-29:
### Describe the bug

Trying to run this MPW design: https://github.com/The-OpenROAD-Project/caravel_user_project/blob/711cfb254cf7fb1595f9e08912dd091d9d383339/openlane/sha1_top/config.tcl
```
set ::env(PL_RESIZER_HOLD_MAX_BUFFER_PERCENT) 80
set ::env(GLB_RESIZER_HOLD_MAX_BUFFER_PERCENT) 80
set ::nev(PL_RESIZER_SETUP_MAX_BUFFER_PERCENT) 80
set ::nev(GLB_RESIZER_SETUP_MAX_BUFFER_PERCENT) 80
```
With above variables facing timing violations. So tried to use default configuration and facing global routing crashes.

### Expected Behavior

Pass global routing stage

### OpenROAD Environment

```shell
`issue_reproducible` tested with latest OpenROAD Head, still facing same error

Git commit: cf9e3b282573a32a7ae3678cd97074a5697276ae
kernel: Linux 3.10.0-1160.49.1.el7.x86_64
os: CentOS Linux 7 (Core)
cmake version 3.24.2
gcc (GCC) 8.3.1 20190311 (Red Hat 8.3.1-3)
clang version 7.0.1 (tags/RELEASE_701/final)
```

### OpenLane Environment

```shell
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: fix_22_12_28-dev
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

866f236 2022-12-28T20:09:47+05:30 mpw7 designs added (#124) - vijayan -  (HEAD -> fix_22_12_28, origin/main, origin/HEAD, main)
04cf3e3 2022-12-27T10:27:50+05:30 mpw7 designs added (#123) - vijayan -  ()
dfdfdb7 2022-12-25T13:43:50+00:00 [BOT] Update OpenLane tag. (#122) - github-actions[bot] -  ()
---
Git Remotes

origin	git@github.com:vijayank88/OpenLane-MPW-CI.git (fetch)
origin	git@github.com:vijayank88/OpenLane-MPW-CI.git (push)
```

### To Reproduce

Attached test case here: 
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/10318776/issue_reproducible.zip)

### Relevant log output

```shell
[INFO GRT-0101] Running extra iterations to remove overflow.
/opt/rh/devtoolset-8/root/usr/include/c++/8/bits/stl_vector.h:932: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = int*; _Alloc = std::allocator<int*>; std::vector<_Tp, _Alloc>::reference = int*&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Signal 6 received
Stack trace:
 0# 0x0000000000CEFEA7 in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 1# 0x00007FA447892400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# 0x0000000000CEFF82 in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 5# grt::FastRouteCore::mazeRouteMSMDOrder3D(int, int, int, int) in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 6# grt::FastRouteCore::run() in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 7# grt::GlobalRouter::findRouting(std::vector<grt::Net*, std::allocator<grt::Net*> >&, int, int) in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 8# grt::GlobalRouter::globalRoute(bool) in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 9# 0x000000000137D3B4 in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
10# 0x00007FA44C3E8EB2 in /lib64/libtcl8.5.so
11# 0x00007FA44C42D36C in /lib64/libtcl8.5.so
```

### Additional Context

If I change `SYNTH_STRATEGY` to `DEALY 4` not facing any crash.

arthurjolo, 2023-01-02:
Hi, In the latest version of OpenROAD this crash does not occur, and GRT finishes normally!

vijayank88, 2023-01-03:
Last crash faced with commit `cf9e3b282573a32a7ae3678cd97074a5697276ae` and with latest not facing any crash.

closed 2023-01-03 by vijayank88