The-OpenROAD-Project/OpenROAD #2677: Differences in generated liberty timing file between OpenROAD and OpenSTA

antonblanchard, 2022-12-30:
### Describe the bug

I generated liberty timing files using the same input liberty and spef inputs. For OpenSTA:

```
read_liberty sky130_fd_sc_hd__tt_025C_1v80.lib
read_verilog RAM32_1RW1R.v
link_design RAM32_1RW1R
read_spef RAM32_1RW1R.spef

read_sdc RAM32_1RW1R.sdc
set_propagated_clock [all_clocks]

write_timing_model sta.lib
```

and for OpenROAD:

```
read_liberty sky130_fd_sc_hd__tt_025C_1v80.lib
read_lef merged.nom.lef
read_def RAM32_1RW1R.def

read_spef RAM32_1RW1R.spef

read_sdc RAM32_1RW1R.sdc
set_propagated_clock [all_clocks]

write_timing_model or.lib
```

The liberty files have some quite large differences, eg:

```
     pin("EN0") {
       direction : input;
       capacitance : 0.2755;
       timing() {
         related_pin : "CLK";
         timing_type : hold_rising;
 	rise_constraint(scalar) {
-          values("0.72490");
+          values("0.87555");
 	}
 	fall_constraint(scalar) {
-          values("0.15198");
+          values("0.87103");
 	}
       }
       timing() {
         related_pin : "CLK";
         timing_type : setup_rising;
 	rise_constraint(scalar) {
-          values("4.74118");
+          values("2.11183");
 	}
 	fall_constraint(scalar) {
-          values("2.61736");
+          values("1.72349");
 	}
       }
       timing() {
         related_pin : "CLK";
         timing_type : hold_falling;
 	rise_constraint(scalar) {
           values("-0.92890");
 	}
 	fall_constraint(scalar) {
           values("-0.66057");
```

### Expected Behavior

The same liberty file (+/- small differences)

### Environment

```shell
Git commit: cf9e3b282573a32a7ae3678cd97074a5697276ae
os: Fedora Linux 36 (Workstation Edition)
cmake version 3.25.1
gcc (GCC) 12.2.1 20221121 (Red Hat 12.2.1-4)
clang version 14.0.5 (Fedora 14.0.5-2.fc36)
```

### To Reproduce

Test case: [liberty-or-vs-sta.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/10322574/liberty-or-vs-sta.tar.gz)

antonblanchard, 2023-01-01:
This might be transparent latch related (used in DFFRAM), because it didn't reproduce on a FF only macro.

jjcherry56, 2023-01-01:
no, it is tristate related. odb has no clue about tristates

jjcherry56, 2023-01-03:
fixed in 03d9aa35f

closed 2023-01-03 by jjcherry56

antonblanchard, 2023-01-03:
Thanks @jjcherry56