The-OpenROAD-Project/OpenROAD #2678: PDNGEN crashes

macd, 2022-12-30:
### Describe the bug

When running the following script in the OpenROAD/src/pdn/test directory, it segfaults (OOM)

### Expected Behavior

No crashes.

### Environment

```shell
macd@macd-NUC9:~/road/OpenROAD$ ./etc/Env.sh
Git commit: cf9e3b282573a32a7ae3678cd97074a5697276ae
kernel: Linux 5.15.0-56-generic
os: Ubuntu 22.04.1 LTS (Jammy Jellyfish)
cmake version 3.22.1
gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
Ubuntu clang version 14.0.0-1ubuntu1
```

### To Reproduce

run the following script in the OpenROAD/src/pdn/test directory
```
source "helpers.tcl"
read_lef Nangate45/Nangate45_tech.lef
read_lef Nangate45/Nangate45_stdcell.lef
read_def nangate_gcd/floorplan.def
add_global_connection -net VDD -pin_pattern VDD -power
add_global_connection -net VSS -pin_pattern VSS -ground
set pwr [[ord::get_db_block] findNet "VDD"]
set gnd [[ord::get_db_block] findNet "VSS"]
set region "NULL"
set name "NULL"
set secondary {}
set switched_power "NULL"
pdn::make_region_domain $name $pwr $switched_power $gnd $secondary $region
define_pdn_grid -name "Core"
add_pdn_stripe -followpins -layer metal1
```

### Relevant log output

```shell
openroad> source "helpers.tcl"
read_lef Nangate45/Nangate45_tech.lef
read_lef Nangate45/Nangate45_stdcell.lef
read_def nangate_gcd/floorplan.def
add_global_connection -net VDD -pin_pattern VDD -power
add_global_connection -net VSS -pin_pattern VSS -ground
set pwr [[ord::get_db_block] findNet "VDD"]
set gnd [[ord::get_db_block] findNet "VSS"]
set region "NULL"
set name "NULL"
set secondary {}
set switched_power "NULL"
pdn::make_region_domain $name $pwr $switched_power $gnd $secondary $region
define_pdn_grid -name "Core"
add_pdn_stripe -followpins -layer metal1
[INFO ODB-0222] Reading LEF file: Nangate45/Nangate45_tech.lef
[INFO ODB-0223]     Created 22 technology layers
[INFO ODB-0224]     Created 27 technology vias
[INFO ODB-0226] Finished LEF file:  Nangate45/Nangate45_tech.lef
[INFO ODB-0222] Reading LEF file: Nangate45/Nangate45_stdcell.lef
[INFO ODB-0225]     Created 135 library cells
[INFO ODB-0226] Finished LEF file:  Nangate45/Nangate45_stdcell.lef
[INFO ODB-0128] Design: gcd
[INFO ODB-0130]     Created 54 pins.
[INFO ODB-0131]     Created 482 components and 2074 component-terminals.
[INFO ODB-0133]     Created 385 nets and 1110 connections.
[INFO PDN-0101] Using VDD as power net for Core domain.
[INFO PDN-0102] Using VSS as ground net for Core domain.
Segmentation fault (core dumped)
```

### Additional Context

Exposure of this to end users is maybe minimal, but roughly what happens is that a voltage domain is defined before a core domain. Then when one of the commands does  a findDomain("Core") (which is in PdnGen.cc) the very first thing it does is to call "ensureCoreDomain()" which winds up adding an new superfluous domain (because core_domain_ is not set) Then when getDomainRows, from domain.cpp, is called it hits an finite recursion between getRows and getDomainRows. I guess the fix would be either to 1) disallow creating a voltage domain before the core domain, or 2) put in a call to ensureCoreDomain()  as the very first thing in makeRegionVoltageDomain, (in PdnGen.cc), but honestly, I don't know this code well enough to do much more.

gadfort, 2022-12-31:
@macd why are you directly calling the `make_region_domain`? I can add a check that the region is NULL, but you should not be calling them directly in TCL.

closed 2022-12-31 by gadfort

macd, 2022-12-31:
I get that, but a user could still do as I did.  I only found it by trying to recreate in Python some of the functionality in the tcl API.