The-OpenROAD-Project/OpenROAD #2680: Detail route ends with DRC violations

vijayank88, 2022-12-30:
### Describe the bug

With OpenROAD commit `7c85c140308f01b73f57ea1117f3e43f39abd437` attached design successfully completed routing without any violations.

Same design creates short violations with latest OpenROAD commit.

### Expected Behavior

Complete detail routing

### OpenROAD Environment

```shell
`4f1108b6f558718ed142cbb6c1f5ba20958195ca`
```

### OpenLane Environment

```shell
OpenLane Tag 2022.12.29
OpenROAD Commit `4f1108b6f558718ed142cbb6c1f5ba20958195ca`
```

### To Reproduce

Due to file size limit, test case uploaded here:

### Relevant log output

```shell
[INFO DRT-0195] Start 64th optimization iteration.
    Completing 10% with 1065 violations.
    elapsed time = 00:00:00, memory = 17330.63 (MB).
    Completing 20% with 1065 violations.
    elapsed time = 00:00:00, memory = 17331.92 (MB).
    Completing 30% with 1064 violations.
    elapsed time = 00:01:12, memory = 17570.39 (MB).
    Completing 40% with 1064 violations.
    elapsed time = 00:01:12, memory = 17570.77 (MB).
    Completing 50% with 1062 violations.
    elapsed time = 00:03:15, memory = 17872.86 (MB).
    Completing 60% with 1062 violations.
    elapsed time = 00:03:15, memory = 17872.86 (MB).
    Completing 70% with 1062 violations.
    elapsed time = 00:03:15, memory = 17872.86 (MB).
    Completing 80% with 1062 violations.
    elapsed time = 00:04:17, memory = 18168.96 (MB).
    Completing 90% with 1062 violations.
    elapsed time = 00:04:17, memory = 18169.19 (MB).
    Completing 100% with 1062 violations.
    elapsed time = 00:06:29, memory = 18387.95 (MB).
[INFO DRT-0199]   Number of violations = 1062.
Viol/Layer        met1   met2
Short              307    755
[INFO DRT-0267] cpu time = 00:25:23, elapsed time = 00:06:30, memory = 18382.40 (MB), peak = 18724.14 (MB)
```

maliberty, 2023-01-02:
When I run your test case:
```
Error: droute.tcl, 15 cannot read file /home/vijayan/FORK_OPENROAD/OL_MPW_CI/OpenLane-MPW-CI/OpenLane/pdks/sky130A/libs.ref/sky130_sram_macros/lib/sky130_sram_1kbyte_1rw1r_32x256_8_TT_1p8V_25C.lib.
```

Please upload a self-contained test

vijayank88, 2023-01-02:
Include missing library with test case again here: https://slack-files.com/T01699QAZBQ-F04GR7YN7U7-8803993c7d

maliberty, 2023-01-02:
All the violations are on top of a macro:
[image]

If I look at one of the nets the guide goes right through the macro on met2:
[image]

so the problem is in global routing.  Please prepare a test case for the global router.

vijayank88, 2023-01-03:
@maliberty Attached Global routing stage test case: https://slack-files.com/T01699QAZBQ-F04H82ASCBV-f5ad3a03f6

maliberty, 2023-01-03:
@luis201420 please look at why there are guides go through the macro that should be blocking them

vijayank88, 2023-01-03:
@maliberty @luis201420 
Please stop debugging this issue.
It was the problem of `GRT_OBS` not defined properly.
There was extra space during `\` definition and its caused violation.
After fixing that empty space now its passing routing stage locally without any issue.

closed 2023-01-03 by vijayank88